MarlinFirmware/Marlin #15683: [BUG] Laggy SD card files browsing

salacpavel, 2019-10-26:
### Description

Beeing on 32bit MPU board, entire interface is snappy and fun to use. Except one part of it. SD card contents browser is, compared to rest of UI, laggy. That is despite the fact I do not keep more then say 10 files on it. This often leads to wrong file selected for print, which is always painful experience. As far as I can tell the responsivness of UI when browsing through files does not depend on brand and/or size of the SD card, IMO its same on no-name 256MB as on Sandisk class-10 32GB card.
Also, file name sorting is not enabled on any of my builds.
As far as I can tell its consistently slow on any bugfix-2.0.X version I have ever tried. 

### Steps to Reproduce

Envoke main menu.
Browse it, observe its responsivness.
Then select Media browser, observe how laggy it is.

**Expected behavior:** 

Snappy

**Actual behavior:**

Laggy
#### Additional Information

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3775072/config.zip)

boelle, 2019-11-03:
@salacpavel many updates have been done in the last 8 days, is the issue still there in 2.0.x ?

salacpavel, 2019-11-03:
I will verify it with fresh build later, if that is all right.

boelle, 2019-11-03:
yep, a fresh build is always the best :-D

salacpavel, 2019-11-06:
Still about 4x slower browsing files, then browsing main menu items.

boelle, 2019-11-16:
could it be that it takes 4x time longer to read from an sd card?

boelle, 2019-12-01:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-01 by boelle