MarlinFirmware/Marlin #15684: Z-Offset change Z-Value

CGES19, 2019-10-26:
Marlin Version : Marlin-bugfix-2.0.x

In the menue where is to change the Z-offset value, the value changes the first time after four  encoder steps. It should change after one.

the encoder values in configuration.h are:
#define ENCODER_PULSES_PER_STEP 4  // 4 necessary for one full circle, that's the minimum
#define ENCODER_STEPS_PER_MENU_ITEM 1 

The changes in the menues works fine.

I expect that the Z-offset values changes also after one step of the Encoder (one full circle).

thinkyhead, 2019-10-26:
What happens if you use…
```cpp
#define ENCODER_PULSES_PER_STEP 1  // 4 necessary for one full circle, that's the minimum
#define ENCODER_STEPS_PER_MENU_ITEM 4
```
…?

CGES19, 2019-10-27:
Now I have changed the parameter ENCODER_PULSES_PER_STEP to 1, recompiled it and examined how Marlin behaves.

In any menu the selection jumps to the end or to the beginning. A selection in between is not possible. 

In the menu for the GCODE file selection the selection always jumps 7 files forwards or backwards.

If you could show me in which subroutines the processing of the encoder signals is done and in which subroutine the Z offset is changed, I could take a look myself.

From: Scott Lahteine 
Sent: Saturday, October 26, 2019 10:41 PM
To: MarlinFirmware/Marlin 
Cc: CGES19 ; Author 
Subject: Re: [MarlinFirmware/Marlin] Z-Offset change Z-Value (#15684)

What happens if you use…

#define ENCODER_PULSES_PER_STEP 1  // 4 necessary for one full circle, that's the minimum

#define ENCODER_STEPS_PER_MENU_ITEM 4…?

—

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

CGES19, 2019-10-28:
I have now made another attempt.
Menu control works.
Z-value works correctly but now the feed value in the info menu always jumps by 4.

CGES19, 2019-10-29:
I have now made another attempt.
#define ENCODER_PULSES_PER_STEP 1
#define ENCODER_STEPS_PER_MENU_ITEM 4

Menu control works.
Z-value works correctly but now the feed value in the info menu always jumps by  4. 

From: Scott Lahteine 
Sent: Saturday, October 26, 2019 10:41 PM
To: MarlinFirmware/Marlin 
Cc: CGES19 ; Author 
Subject: Re: [MarlinFirmware/Marlin] Z-Offset change Z-Value (#15684)

What happens if you use…

#define ENCODER_PULSES_PER_STEP 1  // 4 necessary for one full circle, that's the minimum

#define ENCODER_STEPS_PER_MENU_ITEM 4…?

—

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

CGES19, 2019-10-30:
Obviously there is a fixed divider factor for the values of Z-offset and feedrate.
Therefore my suggestion: add additional definitions in configure.h for Z-Offset and feedrate.

boelle, 2019-11-03:
@CGES19 still an issue?

CGES19, 2019-11-04:
For me, it's one problem.
The dividers for signals from the rotary encoder for the different functions are different and therefore there are different and undesirable effects.
It would be good to solve the problem comprehensively and not just on the Z-Offset setting.
You are welcome to change the heading of the issue.

boelle, 2019-11-16:
> Z-value works correctly but now the feed value in the info menu always jumps by 4.

change #define ENCODER_STEPS_PER_MENU_ITEM 4 to 1 
and change #define ENCODER_PULSES_PER_STEP 1 to 4

?

CGES19, 2019-11-17:
These are the defaults in Configuration.h .

All works well but not Z-adjust  –---    I need 3 or 4 steps per Z-count.

From: Bo Herrmannsen 
Sent: Saturday, November 16, 2019 11:42 AM
To: MarlinFirmware/Marlin 
Cc: CGES19 ; Mention 
Subject: Re: [MarlinFirmware/Marlin] [BUG] Z-Offset change Z-Value (#15684)

  Z-value works correctly but now the feed value in the info menu always jumps by 4.

change #define ENCODER_STEPS_PER_MENU_ITEM 4 to 1
and change #define ENCODER_PULSES_PER_STEP 1 to 4

?

boelle, 2019-11-24:
@CGES19 have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-12-07:
since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2019-12-09:
will close this one as its prob. a config error

play with encoder settings until you finds what works with your particular board/LCD setup

And if it's always all over the place its probably has a bad encoder on the lcd, Or it's picking up interference

also ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, we can reopen

closed 2019-12-09 by boelle

CGES19, 2019-12-10:
I have a question for you:
Do You have a 3D printer and have checked this issue?
Or are You just a theoretician?
From: Bo Herrmannsen 
Sent: Tuesday, December 10, 2019 12:01 AM
To: MarlinFirmware/Marlin 
Cc: CGES19 ; Mention 
Subject: Re: [MarlinFirmware/Marlin] [BUG] Z-Offset change Z-Value (#15684)

will close this one as its prob. a config error

play with encoder settings until you finds what works with your particular board/LCD setup

And if it's always all over the place its probably has a bad encoder on the lcd, Or it's picking up interference

also ask here:

RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, we can reopen

boelle, 2019-12-10:
> Do You have a 3D printer

i have 2, and 1 or 2 more comming in the future

> have checked this issue

Yes

> are You just a theoretician

No, more a practical person