MarlinFirmware/Marlin #15686: [FR] different driver current for sensorless_homing

danym21, 2019-10-26:
### Description
I use TMC5160 stepper drivers, since I use 0.9deg/step instead 1.8deg/step moters stallguard is not triggered  correctly with the the current that I use for printing.

as example for printing I use a current from 1100 mA and "STALL_SENSITIVITY  2" the machine is crashing, when I use "STALL_SENSITIVITY  1" Stallguard is trigger often in the middle of the way to the edge.  When I use a current from 462 for homing with "STALL_SENSITIVITY  2" all is working fine but on printing it is loosing steps.

when this is usefull for all, then I will share my code changes that I testing at the moment.
I add additional settings in the configuration_adv.h and changed the G28 source code.
at the moment it is woring fine for me.

thinkyhead, 2019-10-26:
When you have a PR submitted, we can close this issue and discuss the PR.

danym21, 2019-10-27:
I have tried to submitt, can you check please?

closed 2020-06-30 by github-actions[bot]