MarlinFirmware/Marlin #15687: [BUG] Setting the FW to factory settings or to Default Values the Bed Level is invalid

Powerprobot, 2019-10-26:
The many changes in Marlin FW requires the flashing FW often.
To start print and there is no mismatch in EPROM i set after every flash the factory defaults.

After many Prints i bewildered over the leveling. So i start to observe the Settings G-Code file make by "M503 S0" 

### Description

After the factory defaults is the level state "Invalid Mesh" - readout with Pronterface with "M420 V1"
Set the level state to "ON" or Change Points with "G29 W I0 J0 Z0.12000" are failed with ERROR:No bilinear grid.

the only way to enable the Mesh is leveling (log process) or activate "Debug_leveling" with following command:
"G29 C1"
"M500"

after this procedure i can use my settings File with correct Bed Level Values and turn on the Leveling with "M420 S1 Z5"

_____________________________

It is not necessary or essential to fill up the Mesh with Zeros after a Factory Reset?

It would be nice to setup the Bed level to Zero after a factory default or loading default settings, it can be used by a previous Setting Gcode file.

boelle, 2019-10-27:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Powerprobot, 2019-10-27:
Yes, right there.
Here are my files and additional Infos:
Board BTT SKR1.3 (NXP1768) with LCD12864 FG + TFT35-1.2 on modified Alfawise U30.
(Only the Stepper-Motors and skeleton are original...)
Clone BL-Touch 1.2 in Software Mode. (Works fine !)

I hope you can help to identify a BUG or a configuration issue.
[configuration-files.zip](https://github.com/MarlinFirmware/Marlin/files/3775710/configuration-files.zip)

boelle, 2019-11-03:
@Powerprobot many updates have been added in the last 7 days, is the issue still there in latest 2.0.x ?

Powerprobot, 2019-11-03:
@boelle 
Sorry to say... Yes. 
It is possible to make a fake grid with Zeros after factory reset in Marlin Code?

boelle, 2019-11-03:
i dont think so

@Roxy-3D might know how to do that

ManuelMcLure, 2019-11-03:
If you’re using UBL then ``G29 P0`` will zero out the current mesh.

Powerprobot, 2019-11-03:
@ManuelMcLure  UBL is too much for me an my Printer... I only need ABL Bilinear; it's easy and simple and works fine for me. If i have more Time, i will learn UBL to use.
I tested for a few minutes ago... nice, that firmware defaults work; no error "Invalid Mesh" on M420 command.

boelle, 2019-11-16:
what i do after marlin is flashed

M502
M500
G28
G29

afte G29 is complete 

M500

boelle, 2019-11-16:
maybe provide debug from during G29?
you can use a site like paste bin to provide us that

Vertabreak, 2019-11-16:
generally every time you flash new firmware you want to initialize the eeprom to wipe the old data and load the new data as the eeprom is not effected by the flash allowing old settings to be transferred in to new firmware but doing so often causes issues.

Powerprobot, 2019-11-18:
I think no every one understand my problem....
I know to make a "ABL", but it take a long time... heat up bed... G28... G29... Save Mesh.
It took 15 Minutes to finish the "ABL"

To shorten up this time, i make a G-Code FILE with information from "M503 S0" after a sucessfully "ABL".

After a FW flash, i clear EEPROM and print this "EEPROM.GCODE" file to recovery the "ABL" and all other settings from file in a fraction of second.

But it doesn't work without "G29 Debug-Option". Debug-Options a normaly disabled on a production machine.

here the EEPROM.GCODE file:
```
G21    ; Units in mm (mm)
M149 C ; Units in Celsius
M200 D1.75
M200 D0
M92 X160.00 Y160.00 Z800.00 E192.10
M203 X300.00 Y300.00 Z25.00 E100.00
M201 X700.00 Y700.00 Z100.00 E3000.00
M204 P600.00 R3000.00 T600.00
M205 B20000.00 S0.00 T0.00 J0.04
M206 X0.00 Y0.00 Z0.00
M420 S0 Z3.00
G29 W I0 J0 Z0.11250
G29 W I1 J0 Z0.03500
G29 W I2 J0 Z0.01375
G29 W I3 J0 Z0.00750
G29 W I4 J0 Z0.03750
G29 W I5 J0 Z0.04500
G29 W I6 J0 Z0.02125
G29 W I0 J1 Z0.26625
G29 W I1 J1 Z0.19125
G29 W I2 J1 Z0.18125
G29 W I3 J1 Z0.13250
G29 W I4 J1 Z0.21875
G29 W I5 J1 Z0.23750
G29 W I6 J1 Z0.19250
G29 W I0 J2 Z0.13625
G29 W I1 J2 Z0.02625
G29 W I2 J2 Z-0.01125
G29 W I3 J2 Z-0.03250
G29 W I4 J2 Z0.00000
G29 W I5 J2 Z0.02125
G29 W I6 J2 Z0.02125
G29 W I0 J3 Z0.18125
G29 W I1 J3 Z0.05875
G29 W I2 J3 Z0.03250
G29 W I3 J3 Z0.01625
G29 W I4 J3 Z0.01375
G29 W I5 J3 Z0.07250
G29 W I6 J3 Z0.01125
G29 W I0 J4 Z0.12375
G29 W I1 J4 Z0.02375
G29 W I2 J4 Z0.01125
G29 W I3 J4 Z-0.07000
G29 W I4 J4 Z-0.02125
G29 W I5 J4 Z0.01625
G29 W I6 J4 Z-0.00250
G29 W I0 J5 Z0.12250
G29 W I1 J5 Z0.00875
G29 W I2 J5 Z-0.01625
G29 W I3 J5 Z-0.05375
G29 W I4 J5 Z-0.02125
G29 W I5 J5 Z0.01375
G29 W I6 J5 Z0.02250
G29 W I0 J6 Z0.16750
G29 W I1 J6 Z0.02750
G29 W I2 J6 Z0.04625
G29 W I3 J6 Z0.03000
G29 W I4 J6 Z0.05625
G29 W I5 J6 Z0.09625
G29 W I6 J6 Z0.07750
M145 S0 H200 B60 F0
M145 S1 H230 B80 F0
M301 P36.61 I5.57 D60.09
M304 P105.53 I16.27 D456.33
M207 S3.00 W13.00 F2700.00 Z0.00
M208 S0.00 W0.00 F480.00
M209 S0
M851 X-33.00 Y-6.00 Z-1.26
M906 X800 Y800 Z800
M906 T0 E800
M913 X253 Y253 Z3
M913 T0 E30
M569 S1 X Y Z
M900 K1.60
M603 L0.00 U100.00
M425 F0.90 X0.05 Y0.10 Z0.00 S3.00
M412 S1
M500
```

and here the Pronterface Output:
```
echo:Now fresh file: eeprom~1.gco
File opened: eeprom~1.gco Size: 1737
File selected
echo:Unknown command: "M149 C "
echo:Bed Leveling OFF
echo:Fade Height 3.00
Error:No bilinear grid
[ERROR] Error:No bilinear grid

Error:No bilinear grid
[ERROR] Error:No bilinear grid

Error:No bilinear grid
[ERROR] Error:No bilinear grid

Error:No bilinear grid
[ERROR] Error:No bilinear grid

Error:No bilinear grid
[ERROR] Error:No bilin....  and so on.....
```
If i activate the G29 Debug-Options an add the Line "G29 C" before first G29. The next change ist to place the "M420" command to the end of "G29" commands and change the option "S0" to "S1".
This works fine!
But it is complicated for i simple User to make this. The better Way is -> Save EEPROM to SD and read EEPROM from SD.
The "M502 S0" is like that, but not perfect and didn't work without Debug options.
```
echo:Now fresh file: eeprom~1.gco
File opened: eeprom~1.gco Size: 1749
File selected
echo:Unknown command: "M149 C "
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 +0.064 -0.046 +0.079 -0.092 +0.068 -0.001 -0.018
 1 +0.079 +0.089 +0.007 -0.013 -0.093 +0.040 -0.025
 2 +0.020 -0.098 -0.076 +0.056 +0.061 -0.044 +0.081
 3 -0.058 -0.082 -0.099 -0.021 -0.044 -0.029 +0.001
 4 +0.077 -0.012 -0.094 -0.038 +0.075 +0.098 +0.053
 5 +0.017 +0.041 +0.010 -0.084 +0.034 -0.059 +0.019
 6 +0.095 +0.017 -0.051 -0.070 -0.085 +0.099 -0.003
echo:busy: processing
echo:Bed Leveling ON
echo:Fade Height 3.00
echo: p:36.61 i:5.57 d:60.09
echo: p:105.53 i:16.27 d:456.33
echo:Settings Stored (764 bytes; crc 65273)
```

boelle, 2019-11-25:
i assume you still have an issue here, but i'm curious, are you doing bed level before each print or just when you move the printer arround?

boelle, 2019-11-26:
@Powerprobot btw what version of marlin are you using?

Powerprobot, 2019-11-26:
@boelle 
Normaly i do bed level after change the print material from PLA to PETG or otherwise, depend on the different bed temperature. The Alfawise U30 is a dirty puzzle box and have many mechanical issues...
I solve this problems with stabilizes and modifications. I change the Mainboard to a SKR1.3 with TMC2208 V2 driver and works fine. But the Bed is evil... i screw the Y-Axis Aluminium pole to tight with the Base. So the pole bend a little bit and then the Bed swipe up and down in the range of the screws.
To the Marlin FW: i updated every two days the FW with the pull functiion from GITHUB Desktop.
After a cleaning enviroment i build a new firmware.bin with VSCode and PlatformIO.
After flashing the FW i clean up the EEPROM with Factory defaults. ( I know, that the Board has no real EEPROM. The EEPROM function ist realized with an snap from the FW memory).
Then i will execute (print...) the EEPROM.Gcode file for faster setup, also load fine tunings from different FW defaults. The Bed leveling values in the G-code help me to setup the printer fast without any time scraping movements.
In previous versions of Marlin (i think it was September beginning) works fine. I don`t know if come the issue. i remarked the issue by viewing print results, that the print differ in the first layer.
I think that was fine to have a "zero" ABL Mesh at start/factory defaults to work with G29 commands to programm the specific User Bed level.

By the Way... the I2C Bus can access the WiiNunchuck. I try to implement the DigiJoystick but i struggle with C++, declarations, type declarations and structure... Marlin is so Big!...It flash me totally as hobby programmer...

boelle, 2019-11-26:
oki, i only do bed level if i move the printer,  get same results if i print with abs, pla or petg

i assume you use marlin 2.0.x (just so i can put that in the title

boelle, 2019-12-07:
@Powerprobot since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle