MarlinFirmware/Marlin #15692: [BUG] Marlin 2.0 linear advence + K>0 = Line stair for the diagonal + vibration

Gerggg, 2019-10-27:
### Description

Since the last update 2.0, the Linear Advance with a K value greater than 0 makes
line stair for the diagonal that make the printer vibrate unnaturally.

Borad : SKR 1.3 (CPU : LPC1768)
Stepper driver : TMC 2209 (in UART)
Stepper NEMA17 0,9°

### Steps to Reproduce

1- Set the linear advence with a value of K greater than 0
2- Start printing with a diagonal (45 °)
3- The printer turns into a vibration system
4- If during printing you return the value of K to 0, the vibrations stop.

#### Additional Information

my config files : [Config.zip](https://github.com/MarlinFirmware/Marlin/files/3775579/Config.zip)
or : https://github.com/Gerggg/Marlin-for-my-printer

ManuelMcLure, 2019-10-27:
Do you know what version did not show the problem? It helps if we can narrow down the changes to a specific time period.

Gerggg, 2019-10-27:
@ManuelMcLure I think 1 or 2 weeks... sorry i can not be more specific

Gerggg, 2019-10-27:
Despite the transition from K to 0, even if the vibrations are less present, the quality of the prints is far below my usual impressions.
so that you can see the difference in quality, I add photos.

[Photos.zip](https://github.com/MarlinFirmware/Marlin/files/3775765/Photos.zip)

Gerggg, 2019-10-28:
@ManuelMcLure I saw that I have archives (on 8 days) of old version, I will recompile them to see from which I have the bug.

Gerggg, 2019-10-29:
@ManuelMcLure it's obviously more than a week that the problem is present.
I went back to 9 days and he is still present.
As I'm not good with git, I reset my fork 9 days ago that's why I do not have the history before ...
I think the bug appeared in week 42.

As it is more visible with the linear advance, I printed the attached gcode to test (be careful there is a Z offset in the file)

[Flash kfacto.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/3782013/Flash.kfacto.gcode.txt)

[image]

Gerggg, 2019-10-29:
another piece of information, trips are jerky only during impressions.
I did a test with the commands G1 X0 Y0 F7200 then G1 X240 Y240 F7200, no vibration.

Gerggg, 2019-10-30:
I will test the dev-2.1.x. which seems an old version of the 2.0, I keep you informed

Gerggg, 2019-10-30:
For linear feed, you can solve the problem by adding SQUARE_WAVE_STEPPING and leaving MINIMUM_STEPPER_PULSE disabled. for the "juddering" no solution for the moment ...

BastR, 2019-10-31:
A machine I'm working on have the same issue but with another board, config is:

-SKR PRO
-TMC 2130 SPI all axis (X, Y, Z, Z2, E)
-0.9° steppers

With **LA > 0** the printer seems to print "very fast" with sudden movements when printing in diagonal, infill with gyroid seems to print normally
With **LA = 0**, suddens movements are much lower but jerky (see picture)
With **LA disabled** in firmware, all problems have gone.

[image]

Gerggg, 2019-10-31:
I'm going crazy ... I tested the dev-2.1.x and I still have the problem. I even tested with the dev-2.1.x and TMCStepper v0.5.0 library. the problem is still present.

Gerggg, 2019-10-31:
a photo to show the result

[image]

BastR, 2019-10-31:
I see you have 0.9° stepper too, I will try to put  8 microsteps and 100 steps/mm instead of 16 microsteps and 200 steps/mm.

Gerggg, 2019-10-31:
the SRK is a 32 bit card, it should not have a problem for 16 microsteps.
I will test by completely disabling the linear advance to see if I arrive like you to remove the "tremors"

Gerggg, 2019-10-31:
@BastR Hmm ... you mean if I go from:
#define DEFAULT_AXIS_STEPS_PER_UNIT {160, 160, 800, 806}
at
#define DEFAULT_AXIS_STEPS_PER_UNIT {160.5, 160.5, 800, 806}
it should, as for you solve the problem?

hearnia2k, 2019-10-31:
@BastR so it sounds like now, you're back on settings you started with, which originally had the issue, yet now it's no longer presenting the problem, is that right?

Gerggg, 2019-10-31:
@hearnia2k it seems to me that it has gone from 200 steps/mm to 200.50 steps/mm. No ?

BastR, 2019-10-31:
We made a lot of test and yes, we back to 200steps/mm for X and Y with LA enabled and all work correctly

hearnia2k, 2019-10-31:
I just ran a test myself here with a settings that yesterday were juddering.  I have made 2 changes though:

`BLOCK_BUFFER_SIZE 64` - this was changed because for some reason my FYSETC mini12864 v2.1 is not updating properly during prints.  I tried setting 32, and it did not help, and nor does 64.  My LCD periodically updates, and at other times it is frozen.  The print continues as 'normal' while the LCD is frozen though.

`SQUARE_WAVE_STEPPING` - This was enabled, as suggested by @Gerggg 

My print failed, I think I got a missed step, but I don't think it was related; I got a missed step or something I think.

I am curious:  Are you guys using Delta printers?  I'm wondering if the issue only affects deltas or not.

Gerggg, 2019-10-31:
On my side I have a cartesian printer (a flash Tevo very modified)

I come back to the problem, how did you solve it?

hearnia2k, 2019-10-31:
Oh, I just realized, I also disabled this `MINIMUM_STEPPER_PULSE 2`  previously I enabled this to help resolve the issue, but for me it's better with this disabled, and only the `SQUARE_WAVE_STEPPING` enabled.

Gerggg, 2019-10-31:
I wonder if it is not an info stored by the firmware but is not reset to the firmware change. When you install a new version, do you do a default reset or an eeprom init?

hearnia2k, 2019-10-31:
I tried several things. Normally an eeprom init, then store, then load failsafe, then store, because I'm paranoid :)

hearnia2k, 2019-10-31:
I compile in Windows, suing PlatformIO in VSCode.  
I'm not too sure how to check which TMC libraries.....I barely know how to use VSCode :-)
If I check in Libraries in PIO Home, and then go to Installed it only shows me the neopixel Library, not any TMC libraries installed :-S
Do you know a better way to check?

hearnia2k, 2019-10-31:
I think I must not have cleared my settings every time, though I was generally clearing as I described if I do that, and try the again it is not behaving the same.  However, the juddering/jerky movement is not present, but my extrusion is not working again, which was an issue I faced when first enabling LA.

Extrusion works, if I use Pronterface to heat the nozzle, and then extrude, however, during prints the extrusion does not work.

BastR, 2019-10-31:
@hearnia2k have you try to disable stealthchop for the extruder?

Gerggg, 2019-10-31:
sorry I forgot but I use MINIMUM_STEPPER_PULSE 4 for my printer: p

you can look my config file (LA disabled)

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3795451/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3795452/Configuration.h.txt)

hearnia2k, 2019-10-31:
Thanks for sending them over.  I will take a look tomorrow or next week, and hopefully get things going.

Gerggg, 2019-10-31:
the same impression without the LA ... the LA is obviously not yet operational

[image]

boelle, 2019-11-03:
LA works fine for me

Gerggg, 2019-11-03:
can you post your configuration?

boelle, 2019-11-03:
configuration.h: https://pastebin.com/XRbJfA8G
configuration_adv: https://pastebin.com/35bQrF3v

Gerggg, 2019-11-03:
@boelle yes it's normal that it works for you, the problem seems to be when there are drivers TMC2208 and TMC2209 on the extruder (with the TMC2100 no problem)

hearnia2k, 2019-11-03:
@Gerggg I've not seen any suggestion it also affects TMC2209?  I understood in fact that 2209 would be OK?  I guess I'll find out at some point, as I ordered some.

Gerggg, 2019-11-04:
@hearnia2k I have TMC2209 and you can find the photos with LA and without the LA by looking at above. But I confirm that the LA worked for a time, I do not find the PR since the LA is working on with the TMC2209.

arminth, 2019-11-04:
I ran into probably the same issue today. I had downloaded a snapshot on September 15 2019. I am using it on an Anycubic Kossel Linear Plus with an SKR 1.3, TMC2208 UART on all steppers. Linear Advance k=0.27 and SQUARE_WAVE_STEPPING enabled. I had this version pretty well dialed in and received great prints with no issues on halted steppers anymore.
Today, I had the crazy idea, to get an up to date snapshot and transfer my September configuration in the November 4 snapshot. I went through the configurations line by line and put in the relevant values from September. It compiled and flashed perfectly on the machine. Everything looked great. Autocalibration and Bed Leveling went through no problems.
Until I started a print. I got terrible vibrations (Printer was humming like a bumblebee when printing). Changing speeds didn't change the vibrations. I stopped the print, reflashed the September 15 version and everything was back to normal. Then I found this issue here which might match the issue I have. So I reflashed the today version to check, if switching off Linear advance stops the vibrations. Did the full setup procedure again (including factory reset for sure, calibration and leveling), started printing, had the vibrations,Extruder stopped moving!!,  I switched of Linear advance and the vibrations were much less but the movement (a calibration circle) was very stuttery and unsmooth. Not nice...

I attach the configurations with this post. Hope it helps finding the issue!
Cheers
Armin

[Marlin_conf.zip](https://github.com/MarlinFirmware/Marlin/files/3806011/Marlin_conf.zip)

Gerggg, 2019-11-05:
I have a test with the version of this morning, the problem of LA seems solved ... I post my conf times it is used to someone.

Borad : SKR 1.3 (CPU : LPC1768)
Stepper driver : TMC 2209 (in UART)
Stepper NEMA17 0,9°

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/3808476/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3808477/Configuration.h.txt)

epicurean70, 2019-11-09:
Hi, 
I am having this same problem. is this only specific to TMC2208 , TMC2209 drivers for delta printers?

arminth, 2019-11-09:
@epicurean70 That could be the case. Maybe, it's better to open a seperate new issue explicitly pointing to Delta, SKRV1.3 and TMC220X. What do you think @thinkyhead ?

hearnia2k, 2019-11-09:
@arminth I don't think that makes sense.  @Gerggg is using a cartesian and also experienced the issue, so it's not unique to deltas.  Also @BastR has an SKR Pro, not an SKR 1.3, although they are similar, it's not as simple as the SKR 1.3, either.

arminth, 2019-11-09:
You may be right. Still, this issue has been closed by @Gerggg. So there is no open issue visible...

Gerggg, 2019-11-15:
Hello,

I just made an update and I have the same problem again ... one of you know if there may be a cache memory problem that is not properly initality during the firmware flash ?

In any case, while waiting, I turned off the linear advance and as before the problem is gone with.

and I said that I did not touch the configuration but just made an update.

swilkens, 2019-11-15:
With K0.35 I can't replicate this on the most recent Marlin 2.0.x-bugifx.

Board: SKR Mini E3 V1.2 (STMF1)
Drivers: TMC 2209
K: 0.35 (first picture) - stable speed for the lines
K: 0, 1, 2 (second picture) - speed 20 - 50 - 20 using LA patern gen

[image]

[image]

Gerggg, 2019-11-16:
well, I do not know where the problem is but it seems solved ... I flashed with a firmware without LA and then I flashed with a firmware with LA, I made an M502 and a M500.

arminth, 2019-11-16:
I will try today on the delta.
Will keep you posted!

arminth, 2019-11-16:
Tried it and the result is:
There is still a hige vibration on the moves both xy/motion and extruder. Vibration on xy doesn't stop by switching off LA.
I think this is a delta issue so I will open a new one. That might be supported by @epicurean70 with some infos on his machine.
Cheers
Armin

arminth, 2019-11-16:
No change for me. I will open a new issue! I hope, @epicurean70 supports me there.

Gerggg, 2019-11-16:
@arminth Did you activate the SQUARE_WAVE_STEPPING option? and / or MINIMUM_STEPPER_PULSE 2? I only have the SQUARE_WAVE_STEPPING option enabled.

arminth, 2019-11-16:
I have minimum stepper pulse 3 and square stepping.

Gerggg, 2019-11-18:
try to keep only the SQUARE_WAVE_STEPPING and leave the MINIMUM_STEPPER_PULSE commented

hearnia2k, 2019-11-18:
@Gerggg for me without the MINIMUM_STEPPER_PULSE my extruder does not work.

Gerggg, 2019-11-20:
@arminth It's strange ... I have a SKR 1.3 card with TMC2209 in UART.
what do you have as material?

BastR, 2019-11-23:
Ok now I have this problem too, before I never had it personally. I didn't change anything on my printer, it just stopped for a long time, the problem happened suddenly...

The printing of the first layer was done normally, but from the second layer (which is faster), the printer acted as if my accelerations were very very high. I then tried to set the linear advance to 0 and the printer started jerking as you can see in the video:

https://youtu.be/VicN88H4jf4

BastR, 2019-11-23:
Could this be related to Ejerk?

If I do M501 with LA enabled I don't have an Ejerk:
```
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.30
```

On the other hand if I do M501 just after flashing with LA disabled I do have an Ejerk but it is equal to 0
```
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B50000.00 S0.00 T0.00 X8.00 Y8.00 Z0.00 E0.00
```

Even after several M502 M500, I still have Ejerk equal to 0, I have to manually enter the value by M205 E2.5 for it to work.

Maybe @Sebastianv650 knows the printer behavior with Ejerk set to 0 or missing?

arminth, 2019-11-23:
I think it's all related to https://github.com/MarlinFirmware/Marlin/issues/15908

There is va PR tackling the issue.

boelle, 2019-11-24:
@arminth what is the pr number?

BastR, 2019-11-24:
@boelle this PR https://github.com/MarlinFirmware/Marlin/pull/15968.

I’ve applied the changes before it was merged and all works perfectly, no more issues for my case.

boelle, 2019-11-24:
will close this one then

we can always reopen if the issue it not gone completely

closed 2019-11-24 by boelle

arminth, 2019-11-24:
PR https://github.com/MarlinFirmware/Marlin/pull/15968  was merged today

boelle, 2019-11-24:
hence why the issue was closed 2 hours ago

arminth, 2019-11-24:
Sorry, posted in wrong thread.