MarlinFirmware/Marlin #15693: [May be BUG] ENDSTOP Don't work

hn0550, 2019-10-27:
### Description
Hardware:  Ramps FD V2.1
Firmware:   marlin bugfix-2.0x Download version today
fault:  endstop don't work , Open and close does not shut down the motor

[config.zip](https://github.com/MarlinFirmware/Marlin/files/3775795/config.zip)

boelle, 2019-10-27:
test with M119

put finger on endstop and send M119
remove finger and send M119 again

hn0550, 2019-10-27:
> test with M119
> 
> put finger on endstop and send M119
> remove finger and send M119 again
-----------------------------------------------
M119 display  open all min is open.  put finger   &  remove finger.
I confirm  endstop is normal
ramps1.4+marlin_bugfix_1.1.x+endstop is normal

boelle, 2019-10-27:
vil close this one as fixed

closed 2019-10-27 by boelle