MarlinFirmware/Marlin #15697: [BUG] MKS SBase USB-Pronterface connection fails (Win10)

J-PN, 2019-10-27:
Version 26-Oct-19 and previous (but I cannot pinpoint how far)

This has worked some months ago but now I have issues. Connection to Pronterface stays 'Connection...' for ever with all following cases.

1) After connection the cable, USB is recognized as Standard Port Com3 (manually changed the speed 115200)
2) Installed the Smoothieboard signed driver
3) Installed the ..\Marlin\src\HAL\HAL_LPC1768\win_usb_driver, using UnSigned driver installation mode

boelle, 2019-10-28:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

J-PN, 2019-10-29:
Here they are, thanks.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/3785177/Configurations.zip)

boelle, 2019-11-03:
@J-PN many updates have been done in last 5 days, is this issue still present in 2.0.x ?

J-PN, 2019-11-04:
Thanx. Though, I'm struggling to compile now.
/SanityCheck.h:28:4: error: #error "framework-arduino-lpc176x package is out of date
I've removed & re-load/install PlatformIO IDE 1.9.3 but ...

p3p, 2019-11-04:
To fix your initial issue you need to enable the USB serial port, 
`#define SERIAL_PORT -1`

If you are on Windows 10 you didn't need the driver .. hopefully that doesn't hurt anything.

The out of date error shouldn't happen, are you using the old `platformio.ini` after downloading the Marlin update? If not then try removing the `.platformio` folder from your user directory, ans `.pio` from the Marlin directory and try again.. PlatformIO should be downloading the new packages.

J-PN, 2019-11-05:
Tnx, had all those issues :-$ 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3809904/Configuration.zip)

Though, still have a problem :-( Connecting the USB cable, recognized as Standard USB serial (Com3) Pronterface-Connect just gives Connecting... till USB cable is disconnected. The same happens if I change the driver as Smoothieware one.

Start to wonder if the motherboard's USB port is damaged???
(wouldn't make any difference that I have both TFT32 & LCD_FOR_MELZI connected ?)

p3p, 2019-11-05:
Assuming you havn't messed things up by installing multiple drivers (Are you on windows 10? you didn't need any drivers in that case)

If the usb serial is showing up I doubt there is a problem with the port, its more than likely an issue with pronterface not handling 32bit boards properly (they don't restart on serial connection). I'd recommend testing with another host.

J-PN, 2019-11-06:
I deleted the additional drivers and Yes Win10/x64 selects USB Serial Device (Com3) as ..\usbser.sys 10.0.1.18362

I installed the Repetier:
Connect Com3/115200: No start signal detected - force starting
-and on the bottom line '1 Command Waiting'

:-(

p3p, 2019-11-06:
In the last config you posted `SERIAL_PORT` is still set to `0`, it should be `-1`, have you fixed that?

if you have could you try a different host program?

J-PN, 2019-11-06:
Got it:
#define RX_BUFFER_SIZE 1024 // by default this is commented out

p3p, 2019-11-06:
I can't say I have ever needed to enable that, and have no idea why it would be necessary for you, The USB Serial device does not use that value.

J-PN, 2019-11-07:
Well, don't know but here's the files working.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3820591/Configuration.zip)

p3p, 2019-11-07:
you're sure that you needed to set `RX_BUFFER_SIZE`  along with enabling the second serial port? If you are not using the onboard UART as well as the USB serial you should set `#define SERIAL_PORT -1` rather than enabling port 2.

J-PN, 2019-11-07:
>you should set   rather than enabling port 2.
If I set 
   #define SERIAL_PORT -1
   //#define SERIAL_PORT_2 0
The TFT32 display doesn't respond that's why I didn't notice your comment

Though the TFT32 works when both are enabled
   #define SERIAL_PORT -1
   #define SERIAL_PORT_2 0

AND with
  //#define RX_BUFFER_SIZE 1024 
USB works as  well

p3p, 2019-11-07:
> AND with
//#define RX_BUFFER_SIZE 1024
USB works as well

I'm glad you have it working but at no point does `RX_BUFFER_SIZE` interact with or is used by the usb serial, all this does it increase the hardware uart rx buffer from the default 256 to 1024, so it's a bit confusing.

J-PN, 2019-11-08:
Yes, sorry being unclear. I meant that now I could comment out the buffer size line. Why I needed to get USB working is a bit mystery.

closed 2019-11-08 by J-PN