MarlinFirmware/Marlin #15698: [BUG] UBL Manually probed mesh not interpolated

comps, 2019-10-27:
### Description
I have a Lulzbot TAZ6 printer and would like to use manual (paper-based) mesh leveling on it.

Since `MESH_BED_LEVELING` has broken LCD support (reported in many other issues, always closed without fixing), I went for UBL.

1. `G28`
1. `G0 X50`  ; homing button (Z=0) significantly above the actual bed (cca 4.5mm) and `G29 P2` would crash the nozzle
1. `G29 I999`   ; undocumented but required, extracted from LCD code
1. `G29 P2 H-4`   ; 4mm gives ~0.5mm safety margin
1. `G29 P3`
1. `G29 S`

This works as expected and works well. The mesh values are approx. the same as with manual (non-UBL) bed leveling.

However when I do `M420 S1`, the initial position/value is correct, but any `G0` or `G1` move moves Z back to the un-leveled position (while the displayed Z coordinate remains the same). **The bed level seems corrected only above the actual probed points and not interpolated in between.**

Full configs attached below, the mesh-related changes are:
```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 7f842c48e..bcc5ff7f3 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -998,9 +998,9 @@
  * Example: `M851 Z-5` with a CLEARANCE of 4  =>  9mm from bed to nozzle.
  *     But: `M851 Z+1` with a CLEARANCE of 2  =>  2mm from bed to nozzle.
  */
-#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
-#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
-#define Z_CLEARANCE_MULTI_PROBE     5 // Z Clearance between multiple probes
+#define Z_CLEARANCE_DEPLOY_PROBE    2 // Z Clearance for Deploy/Stow
+#define Z_CLEARANCE_BETWEEN_PROBES  2 // Z Clearance between probe points
+#define Z_CLEARANCE_MULTI_PROBE     2 // Z Clearance between multiple probes
 //#define Z_AFTER_PROBING           5 // Z position after probing is done
 
 #define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping
@@ -1202,14 +1202,14 @@
 //#define AUTO_BED_LEVELING_3POINT
 //#define AUTO_BED_LEVELING_LINEAR
 //#define AUTO_BED_LEVELING_BILINEAR
-//#define AUTO_BED_LEVELING_UBL
-#define MESH_BED_LEVELING
+#define AUTO_BED_LEVELING_UBL
+//#define MESH_BED_LEVELING
 
 /**
  * Normally G28 leaves leveling disabled on completion. Enable
  * this option to have G28 restore the prior leveling state.
  */
-//#define RESTORE_LEVELING_AFTER_G28
+#define RESTORE_LEVELING_AFTER_G28
 
 /**
  * Enable detailed logging of G28, G29, M48, etc.
@@ -1279,8 +1279,8 @@
 
   //#define MESH_EDIT_GFX_OVERLAY   // Display a graphics overlay while editing the mesh
 
-  #define MESH_INSET 1              // Set Mesh bounds as an inset region of the bed
-  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
+  #define MESH_INSET 30             // Set Mesh bounds as an inset region of the bed
+  #define GRID_MAX_POINTS_X 3       // Don't use more than 15 points per axis, implementation limited.
   #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
 
   #define UBL_MESH_EDIT_MOVES_Z     // Sophisticated users prefer no movement of nozzle
```

#### Additional Information

* [configs.zip](https://github.com/MarlinFirmware/Marlin/files/3776072/configs.zip)

boelle, 2019-11-03:
@comps many updates have passed in the last 7 days, is the issue still present in 2.0.x

comps, 2019-11-03:
Still present on b9116d4050c.

boelle, 2019-11-11:
present on https://github.com/MarlinFirmware/Marlin/commit/af61f37b6595edeb43ec880529880e75dd1649be also?

comps, 2019-11-12:
Yes.

boelle, 2019-11-21:
@comps still an issue if you download a new copy?

Roxy-3D, 2019-11-21:
> G29 I999 ; undocumented but required, extracted from LCD code

It is documented...   

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp#L117-L125

But more complete user documentation would be appreciated.   If you update the documentation, I'll merge it.

> G29 P3
> G29 S
> This works as expected and works well. The mesh values are approx. the same as with manual (non-UBL) bed leveling.
>
> However when I do M420 S1, the initial position/value is correct, but any G0 or G1 move moves Z back to the un-leveled position (while the displayed Z coordinate remains the same). The bed level seems corrected only above the actual probed points and not interpolated in between.

Did you do a G29 A to activate the UBL Bed Leveling system?       And...  as for the mesh points being 'approximate'...    I think you will find G26 very helpful.   You will be able to see where your mesh points are too high or too low.    And you can use G29 P4 to edit those points to be very finely tuned.

boelle, 2019-12-01:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-01 by boelle