MarlinFirmware/Marlin #15700: endstop interrupt issue

ruggb, 2019-10-27:
ver bugfix-2.0.x
build 9e8021d
I have **#define ENDSTOP_INTERRUPTS_FEATURE

when I compile I get this error
Arduino: 1.8.10 Hourly Build 2019/05/07 05:33 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\module\endstops.cpp:36:0:

c:\users\bill\appdata\local\temp\arduino_build_168374\sketch\src\hal\hal_avr\endstop_interrupts.h: In function 'void setup_endstop_interrupts()':

c:\users\bill\appdata\local\temp\arduino_build_168374\sketch\src\hal\hal_avr\endstop_interrupts.h:135:7: error: non-constant condition for static assertion

       static_assert(digitalPinToPCICR(Y_MIN_PIN), "Y_MIN_PIN is not interrupt-capable");

       ^~~~~~~~~~~~~

c:\users\bill\appdata\local\temp\arduino_build_168374\sketch\src\hal\hal_avr\endstop_interrupts.h:57:57: error: reinterpret_cast from integer to pointer

                                     WITHIN(p, 62, 69) ? &PCICR : nullptr )

                                                         ^

c:\users\bill\appdata\local\temp\arduino_build_168374\sketch\src\hal\hal_avr\endstop_interrupts.h:135:21: note: in expansion of macro 'digitalPinToPCICR'

       static_assert(digitalPinToPCICR(Y_MIN_PIN), "Y_MIN_PIN is not interrupt-capable");

                     ^~~~~~~~~~~~~~~~~

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

With interrupts disabled it compiles OK.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3776229/configs.zip)
** enabled

shitcreek, 2019-10-29:
Got same error messages with Arduino.
No issue compiling with PlatformIO.
So maybe try an older version of Arduino or use PlatformIO

ruggb, 2019-10-29:
What was the change in Marlin that is now causing this error? It appears that maybe the Y interrupt was not working in Marlin and the fix is now showing an issue with Arduino if it compiles with PlatformIO.

ruggb, 2019-10-29:
this is what I got from Arduino = it actually does not make total sense, because v1.8.10 worked, then it did not. Their indication is that older versions worked, newer are more complaint.......+

In 1.8.10, the compiler (gcc) version was upgraded. The new version is a lot more strict (that is, more standards-compliant) regarding what can be used as a constant expression (i.e. in a static_assert statement). Apparently the Marlin firmware uses an expression that is not a valid constant expression. Looking at the code, this is something that should be fixed either in Marlin (by changing the assert to use something that is constant), or the library that exposes digitalPinToPCICR (which I think is not the Arduino IDE or core itself).

In any case, this is not something that the Arduino IDE can fix - the problem is in code that has always been illegal, it's just that in older versions gcc did not produce an error and made it work but no longer does so.

Thus I'm closing this issue, I would suggest you report this at the Marlin firmware (pointing to this issue). They might want to take this up with the PCINT library used, but it is probably good to track this at Marlin too.

—

ruggb, 2019-10-29:
I re-unzipped the build completely. It appears that I accidentally deleted something that I should not have, because it now compiles. However, I do get these two errors in verbose mode:

Error while detecting libraries included by C:\Users\Bill\AppData\Local\Temp\arduino_build_387254\sketch\src\Marlin.cpp

Error while detecting libraries included by C:\Users\Bill\AppData\Local\Temp\arduino_build_387254\sketch\src\feature\leds\tempstat.cpp

any ideas?

ruggb, 2019-10-29:
when viewed in verbose mode, I also get these 2 errors early on:

Error while detecting libraries included by C:\Users\Bill\AppData\Local\Temp\arduino_build_13146\sketch\src\Marlin.cpp

Error while detecting libraries included by C:\Users\Bill\AppData\Local\Temp\arduino_build_13146\sketch\src\feature\leds\tempstat.cpp

What is the significance of that?

ruggb, 2019-10-29:
OK, whatever it is, it has to do with pin 14. If I change X_MIN_PIN to 14 and Y_MIN_PIN to 3, the error changes to X_MIN not an interrupt.

shitcreek, 2019-10-29:
Your original hypothesis is correct. It is a code compliance issue. We should be using something else instead of static_assert or create a special condition.

It isn't the issue with the pin - all that did was switch the uninterruptible and interrupterable pins.

Your two errors above just indicate that the libraries being included cannot be found, has a different name, or they contain errors.

ruggb, 2019-10-29:
everything else is defined exactly the same with the static_asserts. Why doesn't that cause errors on the other pins also?

The diff between the previous build that worked and the last one in the endstop.h file appears to be the use of shortcuts or maybe macros. One part of the error was a failure in expanding a macro.

pin 14 was not a problem in the previous build so why is it a problem now?

The previous build compiled with 1.8.10, which Arduino says was the version that changed the compliance. What is happening does not follow that logic......the previous build should have also failed.

Obviously, pin 14 was capable of being an interrupt pin, so why is it now not capable?

As far as I can tell all the libraries are there. If there is an error in one, I have no clue how to find which one.

ruggb, 2019-10-30:
I found the source of the non-fatal errors and wrote it up in #15729 as a separate bug.

ruggb, 2019-10-30:
AND THE ANSWER IS..............You can't run the endstops interrupts on a RAMPS board - at least not Y.
I found this:

The Mega 2560 is capable of 6 external interrupts, which are 0-5 on pins 2, 3, 21, 20, 19, 18 respectively.

Then I traced the schematics. The problem is that on Ramps boards the Y endstops are connected to 14 & 15, which are not interrupt capable pins.

It compiles using 20 & 21 with interrupts enabled, but it won't work that way because they are not connected that way.

Therefore, the BUG is >> interrupts cannot be an option for RAMPS.

closed 2019-10-31 by ruggb