MarlinFirmware/Marlin #15701: SKR PRO linking fails with BLTouch enabled (Servo library)

PKav89, 2019-10-27:
### Description

SKR PRO Linking fails with BLTouch Enabled (Servo library)

### Steps to Reproduce

1. Download latest Marlin 2.0 sources
2. Configure it for BigTreeTech SKR PRO Board (STM32F407) with BLTouch enabled
3. Try to build firmware, get type errors in Servo library
4. Replace all "uint16" line types to "uint16_t" (add "_t") in Servo.h and ServoTimers.h
5. Try to build firmware again

**Expected behavior:** Marlin builds successfully

**Actual behavior:** All sources compiles, but linking fails:

```
Linking .pio\build\BIGTREE_SKR_PRO\firmware.elf
.pio/build/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/Servo.cpp.o: In function `libServo::attach(int)':
c:\Marlin_Ultimaker/Marlin\src\HAL\HAL_STM32/Servo.cpp:36: undefined reference to `Servo::attach(unsigned char, unsigned short, unsigned short, short, short)'
.pio/build/BIGTREE_SKR_PRO/src/src/HAL/HAL_STM32/Servo.cpp.o: In function `libServo::move(int)':
c:\Marlin_Ultimaker/Marlin\src\HAL\HAL_STM32/Servo.cpp:48: undefined reference to `Servo::write(int)'
.pio/build/BIGTREE_SKR_PRO/src/src/gcode/control/M280.cpp.o: In function `GcodeSuite::M280()':
c:\Marlin_Ultimaker/Marlin\src\gcode\control/M280.cpp:46: undefined reference to `Servo::read() const'
c:\Marlin_Ultimaker/Marlin\src\gcode\control/M280.cpp:40: undefined reference to `Servo::detach()'
.pio/build/BIGTREE_SKR_PRO/src/src/module/servo.cpp.o: In function `servo_init()':
c:\Marlin_Ultimaker/Marlin\src\module/servo.cpp:42: undefined reference to `Servo::detach()'
.pio/build/BIGTREE_SKR_PRO/src/src/module/servo.cpp.o: In function `_GLOBAL__sub_I_servo':
c:\Marlin_Ultimaker/Marlin\src\module\../HAL/shared/../HAL_STM32/Servo.h:28: undefined reference to `Servo::Servo()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1
```

I am unable to figure out, if it's a Marlin bug, or Servo Library bug. Seems to be some inconsistency between library function parameters and Marlin calls for that functions. If I disable BLTouch in Configuration.h it builds successfully. Request assistance with building Marlin for SKR PRO with BLTouch.

Configuration files and modified Servo STM32F4 library are included.
[Conf_and_ServoLib.zip](https://github.com/MarlinFirmware/Marlin/files/3776246/Conf_and_ServoLib.zip)

DroneMang, 2019-10-27:
This works for the servos, but something is wrong with Conditionals_LCD.h

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3776429/Configs.zip)

PKav89, 2019-10-29:
I tried to build with these configs and files, but got same errors.

carl1961, 2019-10-30:
If Your using the Pro 1.1 default BLTouch header the set Zmin to PA2      #define Z_MIN_PIN    PA2     // PG8
If Your Using the Zmin pin for the (White and Black wires) from the bltouch the you change  PA2 to PG8 

// Z Probe must be this pins
//
#ifndef Z_MIN_PROBE_PIN
  #define Z_MIN_PROBE_PIN  PG8   //PA2
#endif

PKav89, 2019-10-30:
Yes, i understand it. But it won't help to resolve build issue.

boelle, 2019-11-03:
@PKav89 do you still have issues when using todays code?

PKav89, 2019-11-03:
Yes, just checked, still same errors.

sjasonsmith, 2019-11-08:
Your Configuration has a line in it that should not be there, which causes problems for that servo library.
`  #define ARDUINO_ARCH_STM32F4`

There are a couple other minor issues that need updated to be compatible with the latest bf2 commits, but they are simple enough to sort out.

Here are your configs with the modifications I made for it to compile.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3823238/Configuration.zip)

PKav89, 2019-11-08:
Nope, without this line it says:
```
In file included from Marlin\src\HAL\HAL_STM32\Servo.h:25:0,
                 from Marlin\src\HAL\HAL_STM32\Servo.cpp:29:
C:\users\pkav\.platformio\lib\Servo_ID883\src/Servo.h:75:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
```

sjasonsmith, 2019-11-08:
Ok, that looks like a globally installed Servo library. Try removing it. There is a different servo library inside the Arduino STM32 framework.

PKav89, 2019-11-08:
Now it builds, at last! Thanks.
When I created this issue PlatformIO always downloaded that external library before building.
Hope Arduino will die some day and there will be no more that crappy libraries, only clean and optimal C code. Closing issue.

closed 2019-11-08 by PKav89