MarlinFirmware/Marlin #15702: Marlin 2.0 TMC5160 axis doesnt move board skr v1.3

novabartos, 2019-10-27:
Hi,

I got TMC5160 v1.2 from BIGTECHTREE.
Board: SKR v1.3
Marlin 2.0
Im trying to apply tmc5160 on asix y configure all and still it doesnt work.
I figure it out that comend M122 there are some errors like msteps value 0
See below:
[image]
Driver register are allright.
Connection is ok. #define TMC_USE_SW_SPI, also change Y_RSENSE to 0.075. configure E0_AUTO_FAN_PIN to P2_04. Jumpers are okey.
When i give comends in pronterface to move y and thnt check it by M114 marlin thinks the movement was done. Motor and wires are okey and work fine with TMC2130
I spend whole weekend at this issue. I dont know what im missing
My config files attached

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3776275/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3776276/Configuration_adv.txt)

swilkens, 2019-10-27:
Any warnings during compilation? Afaik ststm32 5.7.0 just came out and caused some compiler warnings that did not appear with 5.6.0

thisiskeithb, 2019-10-28:
> Any warnings during compilation? Afaik ststm32 5.7.0 just came out and caused some compiler warnings that did not appear with 5.6.0

The SKR 1.3 is LPC1768-based, so it's unaffected.

I just rebased [my Ender-3/SKR 1.3/Watterortt TMC 5160 config](https://github.com/thisiskeithb/Marlin/tree/d42c23e51812b562f0153e9d62bf255a8c509366) to the latest commit and ran a print and it works fine.

novabartos, 2019-10-28:
> Any warnings during compilation? Afaik ststm32 5.7.0 just came out and caused some compiler warnings that did not appear with 5.6.0

Nope there are no warnings during compilation:(

novabartos, 2019-10-28:
There has to be something with this msteps, why value is 0 not 16 like i define in configuration? Someone can help me please?

novabartos, 2019-10-28:
> > Any warnings during compilation? Afaik ststm32 5.7.0 just came out and caused some compiler warnings that did not appear with 5.6.0
> 
> The SKR 1.3 is LPC1768-based, so it's unaffected.
> 
> I just rebased [my Ender-3/SKR 1.3/Watterott TMC 5160 config](https://github.com/thisiskeithb/Marlin/tree/d42c23e51812b562f0153e9d62bf255a8c509366) to the latest commit and ran a print and it works fine. [Here's a video](https://twitter.com/thisiskeithb/status/1188636693043142656) testing the new "Show Total E during print" feature with the steppers working as intended.

I download your firmware from github. change it for my settings and still M122 look like this:
[image]
Could you please check on your pronterface how your M122 look?
you need to define #define TMC_DEBUG in configuration_adv

thisiskeithb, 2019-10-28:
> Could you please check on your pronterface how your M122 look?

`M122` works fine when enabled (I’ve been printing with it enabled since the beginning of the year).

Did you run `M502` and `M500` after flashing?

Evg33, 2019-10-28:
My TMC5160 work on skr13.
#define X_DRIVER_TYPE  TMC5160
#define Y_DRIVER_TYPE  TMC5160
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
```
SENDING:M122
		X	Y	Z	E
Address				0	0
Enabled		false	false	false	false
Set current	900	900	150	700
RMS current	886	886	122	673
MAX current	1249	1249	172	949
Run current	17/31	17/31	3/31	21/31
Hold current	8/31	8/31	1/31	10/31
Global scaler	133/256	133/256
CS actual	8/31	8/31	1/31	10/31
PWM scale	19	18	133	33
vsense				1=.18	1=.18
stealthChop	true	true	true	true
msteps		32	32	8	8
tstep		max	max	max	max
pwm
threshold
[mm/s]
OT prewarn	false	false	false	false
OT prewarn has
been triggered	false	false	false	false
off time	3	3	3	3
blank time	24	24	24	24
hysteresis
-end		-1	-1	-1	-1
-start		1	1	1	1
Stallguard thrs	0	0	0	0
DRVSTATUS	X	Y	Z	E
stallguard
sg_result	0	0
fsactive
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0x80:08:40:00
		Y	0x80:08:40:00
		Z	0xC0:01:00:00
		E	0xC0:0A:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
```

novabartos, 2019-10-29:
"Do you power your board from the main power supply before connecting up your USB, or is the board connected to USB first and then the main power applied? You need to make sure that the main power is applied before USB is connected. Otherwise the drivers will not be correctly initialized."
Yes i do it correctryl but thanks for advise is good to remember that
After flashing i run M502 and M500 
The effect now is diffrent the mstep and tstep are all the time changing:
[image]
Motor still doesnt move;/ When i configure for tmc2130 it work with no problem but i really need more current thats why im trying to use TMC5160
How i can set tstep to max? someone know where i can configure that?

novabartos, 2019-10-29:
when i plug new tmc5160 new one it moves for like few second, and after that is some issue. It can be board skr v1.3 problem?

Patag, 2019-10-29:
On my conf SKR1.3/TMC5160, I had to disconnect CLK pin from board then ground it on drivers (Watterott)

Evg33, 2019-10-29:
Yes ! Cut CLK pin.
[image]

boelle, 2019-11-03:
@novabartos is this still an issue if using latest snapshot from today?

novabartos, 2019-11-04:
Hello guys.

I buy new skr v1.3 still doesnt work.
I cut CLK pin and it halped.
Thank you for TIP:)
Issue solved

novabartos, 2019-11-04:
Thank you marlin community:)

closed 2019-11-04 by novabartos