MarlinFirmware/Marlin #15705: #include HAL_PATH  for SKR_PRO

DroneMang, 2019-10-27:
[image]
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3776468/Configs.zip)

This is for an FT-6.  I have EEPROM turned on in configuration and also in in the pins file.  I have BLTOUCH turned on.

I am using Atom with platform io. I have python 3.8 installed.

I downloaded Marlin 2.0 this afternoon.

I am pulling TMCstepper from bigtree, this is shown in the platformio.ini file

You will need all 4 files in the zip to replicate my configuration.

This is the first error:
Marlin\src\lcd\extensible_ui\lib\lulzbot\archim2-flash\../../../../../inc/MarlinConfigPre.h:53:49: fatal error: ../HAL/HAL_STM32/inc/Conditionals_LCD.h: No such file or directory

 #include HAL_PATH(../HAL, inc/Conditionals_LCD.h)

Another user on the Bigtree SKR pro facebook said he did this:

another is a SoftwareSerial error that I had to do a work around by add a folder I named SoftwareSerial in the libdeps folder , looks like this
C:\BIGTREETECH\Marlin-bugfix-2.0.x\.pio\libdeps\BIGTREE_SKR_PRO\SoftwareSerial in that folder I got the SoftwareSerial.h and SoftwareSerial.cpp for the folder C:\BIGTREETECH\Marlin-bugfix-2.0.x\Marlin\src\HAL\HAL_STM32F1

I tried this as well but I still got all the same errors.

Haschtl, 2019-10-27:
I encountered some problems when using the latest tmcstepper v0.5.2 and/or the latest platformio stm32 library (i think 5.7) 
Using tmcstepper 0.5.1 with stm32 v5.6 is fine in my case. Dont know if that helps.
 The failure with the latest tmcstepper version is because this version "thinks" softwareSerial is available on the stm32. So I updated the stm32 library to the latest version (which should add softwareSerial support). But then I got even more errors, so I reverted to the older versions

Underp4ntz, 2019-10-27:
Same here, for some reason i cannot compile anymore and getting HAL errors I think it's a PlatformIO Issue.

TMCStepper 0.5.1 with STM 5.6 works but the later versions only give errors.

DroneMang, 2019-10-28:
So when I run platformio, it uses tmcstepper 0.5.2 even though I say TMCStepper@>=0.5.1,<1.0.0 in the platformioini file. I tried deleting the folder out of libdeps, but it just redownloads it.

ManuelMcLure, 2019-10-28:
0.5.2 satisfies the “TMCStepper@>=0.5.1,<1.0.0“ requirement so that’s not a bug. That requirement says “at least 0.5.2 but less than 1.0.0”. If you want to force 0.5.1 use simply  “TMCStepper@0.5.1l

DroneMang, 2019-10-29:
[Marlin-bugfix-2.0.x-3.zip](https://github.com/MarlinFirmware/Marlin/files/3781174/Marlin-bugfix-2.0.x-3.zip)

This compiles for SKR Pro with Bigtree TMCStepper and STM32 5.6 downloaded from this afternoon.

You have to have git installed, I used Atom to compile.

It's for an FT-6, maybe this can help someone.

rflulling, 2019-10-29:
As of a couple nights ago, Friday night, Sat morning. SoftwareSerial.h was required in order to get past whatever changes had happened and I think the Marlin team has been on damage control since then. As of Sunday the latest zip included a few fixed, not limited to restricting the STSTM32@5.6.0 as 5.7.0 doesn't work, least not with the SKR PRO. They also added a filter flag for TMCstepper as it has been replaced with folder TMCStepper_ID5513. 

I know the devs have their hands full merging in what BTT has done so that every one is on the same page, and BTT users aren't stuck on that one branch. Also so the full functions of the SKR and the newer TMC drivers can be unlocked. 

I am also  FT-6. Black Friday ordered. However is it still an FT-6 when all that remains of the FT-6 is the nuts bolts, bearings and framework? True our source was the FT-6 kit. But can it be argued argued that it is still an FT-6? As of the SKR PRO I am nearly ready to begin converting this thing to something more useful. Less wires in weird places. Added stability, etc. It is slowly becoming, something else. -I still respect John for making the kits and I will still refer DIY interested users to the kit.

closed 2019-10-30 by DroneMang