MarlinFirmware/Marlin #15707: LCD-Manu- Mixer submenu all info on top of eachother

3Dangel, 2019-10-28:
the Mixer menu after code changes, the toggle % information are all jammed on the left of the screen.
using Reprap full graphic discount LCD.
found the fix by changing the chr length and a value of x for the LCD-Length
menu_mixer.cpp
~~~~~~~~~~~~~~~~~~

#if DUAL_MIXING_EXTRUDER
 void _lcd_draw_mix(const uint8_t y) {
   char tmp**[21]**; // "100%_100%"
   SETCURSOR(2, y);
   lcd_put_u8str_P(GET_TEXT(MSG_MIX));
   SETCURSOR(LCD_WIDTH **+ 5**, y);
   sprintf_P(tmp, PSTR("%3d%% %3d%%"), int(mixer.mix[0]), int(mixer.mix[1]));
   lcd_put_u8str(tmp);

boelle, 2019-10-28:
maybe send a PR so that all can benefit from your fix

3Dangel, 2019-10-28:
Oh! Ok.
Will do there.
Cheers
Thanks

3Dangel, 2019-10-28:
PR made
👍 
LCD/Menu/menu_mixer.cpp Lcd rpint wrong coordinate #15715
closing this issue report

boelle, 2019-10-28:
no lets keep it open until pr is merged

3Dangel, 2019-10-28:
Okay! Thanks !!

thinkyhead, 2019-10-28:
I've merged a patch that may do the trick. Check the latest.

closed 2019-10-28 by thinkyhead

3Dangel, 2019-10-28:
Thanks a lot! Will try tonight.
Many thanks to all of you who work on this mega project!!! Kudos!!!!