MarlinFirmware/Marlin #15708: [BUG] M43 doesn't list FANMUX pins.

ellensp, 2019-10-28:
### Description

M43 Does not show any defined FANMUX pins. Applies to both 1.1.x and 2.0 

### Steps to Reproduce

1. Enable M43 in Configuration_adv.h
2. #define FANMUX0_PIN  in Configuration_adv.h  As and example I used  P1_26 
3. Compile and upload
4. Check with M43 gcode

**Expected behavior:** 

M43 P126 in this example should reply: "PIN: 1.26  FANMUX0_PIN Output = 0"

**Actual behavior:**

M43 P126 responds "PIN: 1.26  <unused/unknown>  Output = 0" 

#### Additional Information

Code is simply missing from pinsDebug_list.h

#### Solution

After 
```
#if PIN_EXISTS(FAN3)
  REPORT_NAME_DIGITAL(__LINE__, FAN3_PIN)
#endif

```
add the lines
```
#if PIN_EXISTS(FANMUX0)
  REPORT_NAME_DIGITAL(__LINE__, FANMUX0_PIN)
#endif
#if PIN_EXISTS(FANMUX1)
  REPORT_NAME_DIGITAL(__LINE__, FANMUX1_PIN)
#endif
#if PIN_EXISTS(FANMUX2)
  REPORT_NAME_DIGITAL(__LINE__, FANMUX2_PIN)
#endif
```

boelle, 2019-10-28:
maybe send a PR to 2.0.x so all can benefit

boelle, 2019-11-03:
closing this as its not a bug but i fix
this should have been send in as a PR

closed 2019-11-03 by boelle