MarlinFirmware/Marlin #15710: [BUG] MKS SGen-L - Z Dual Stepper Driver error "E1_MS1_PIN was not declared is this scope"

Ccleorina87, 2019-10-28:
My compile error is "E1_MS1_PIN was not declared is this scope" at pins.h on this line "#define __EPIN(p,q) E##p##_##q##_PIN"

i have free E1 driver never used and wanted to set it up for dual z driver... i was working before July update after i update my firmware latest version this error stop ne from update... im using DRV8825 stepper driver... i have same error on my MKS Gen-L and my new MKS SGen-L.

boelle, 2019-10-28:
please read the issue template and provide the missing information

Ccleorina87, 2019-11-01:
@boelle 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/3798122/Configuration.txt)

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3798124/Configuration_adv.txt)

randellhodges, 2019-11-01:
Comment out `PINS_DEBUGGING` in configuration_adv.h and try. I had a similar (if not the same issue) and a few searches lead me to having to comment that out.

See https://github.com/MarlinFirmware/Marlin/issues/15399

Sharkys80, 2019-11-01:
Try this in : pins_MKS_SGEN_L

#define Z2_STEP_PIN        P0_01
#define Z2_DIR_PIN         P0_00
#define Z2_ENABLE_PIN      P0_10
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN        P1_17
#endif

/*#define E1_STEP_PIN        P0_01
#define E1_DIR_PIN         P0_00
#define E1_ENABLE_PIN      P0_10
#ifndef E1_CS_PIN
  #define E1_CS_PIN        P1_17
#endif
*/

And have fun

Ccleorina87, 2019-11-01:
> Comment out `PINS_DEBUGGING` in configuration_adv.h and try. I had a similar (if not the same issue) and a few searches lead me to having to comment that out.
> 
> See #15399

Wow thanks it works...

closed 2019-11-02 by Ccleorina87