MarlinFirmware/Marlin #15711: [BUG] when SINGLENOZZLE has one fan (always!?) the LCD menu assumes there are more fans

iviar, 2019-10-28:
_(opening for push request)_

In the intent to configured Marlin for 2-in-1-out printing head (which has one fan), I noticed a couple of warnings of access outside array bound, it was only in the LCD menu about the number of fans controlling temperature
Note: I initially wonder for a while due to a compilation error mentioning only HOTENDS, why the calculated define HOTENDS was not 1 because I didn't assign a value, I searched HOTEND in Configuration.h without a direct clarification that SINGLENOZZLE needed to be defined, so I also add the mention "so HOTENDS remains 1" in the comment before SINGLENOZZLE.

### Steps to Reproduce

1. configure with SINGLENOZZLE in Configuration.h and
E0_AUTO_FAN_PIN and E1_AUTO_FAN_PIN  (in Configuration_adv..h)
pointing to the same pin in respect to their preceding documentation:
/*
...
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
*/
2. build
3. ignore... the warning

**Expected behavior:**
compile and work fine

**Actual behavior:**
compile with warnings while accessing out of bound
 thermalManager.fan_speed[1]
 and experience the unchecked (read) access out of array bound from the LCD menu (possibly showing more fans then reality where only the first is right)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2019-10-28:
remember to put link to this issue in the PR, i cant see any PR in 2.0.x yet

and put link to PR when its made in here

el-quique, 2019-10-29:
Hi read this:
https://github.com/MarlinFirmware/Marlin/issues/13928

boelle, 2019-10-29:
why read that? you stopped replying and did not even answer the question i posted 24 september

boelle, 2019-11-11:
@iviar any news on this one? are we still waiting for the PR to be merged?

iviar, 2019-11-12:
On Mon, 11 Nov 2019 04:19:42 -0800, Bo Herrmannsen <notifications@github.com> said:

>@iviar any news on this one? are we still waiting for the PR to be merged?

I changed in my PR (for FAN1 and FAN2)
-    #if HAS_FAN1 || (ENABLED(SINGLENOZZLE) && EXTRUDERS > 1)
+    #if HAS_FAN1 || (DISABLED(SINGLENOZZLE) && EXTRUDERS > 1)

but studiodyne changed over (fine for me, at least doesn't do either
the opposite than sense about SINGLENOZZLE)
-    #if HAS_FAN1 || (ENABLED(SINGLENOZZLE) && EXTRUDERS > 1)
+    #if HAS_FAN1

line was changed from

Remained one thing very marginal,
I recommended the possibility to lookup HOTENDS and
find SINGLENOZZLE because an error saying HOTENDS>1 doesn't say anything
about SINGLENOZZLE which need to be configured not to have HOTENDS>1)
it was just a comment change in the configuration, but not having picked
the PR is not there

-// For Cyclops or any "multi-extruder" that shares a single nozzle.
+// For Cyclops, 2-in-1-out or any "multi-extruder" that shares a single nozzle,
+// so HOTENDS remains 1
 //#define SINGLENOZZLE

Cheers,
MARco
-- 
x(t),y(t) = th(3t-34.5)*e^[-(3t-34.5)^2]/2-4.3+e^(-1.8/t^2)/(.8*atg(t-
3)+2)(t-1.8)-.3th(5t-42.5),(1.4e^[-(3t-34.5)^2]+1-sgn[|t-8.5|-.5]*1.5*
|sin(pi*t)|^[2e^(-(t-11.5)^2)+.5+e^(-(.6t-3.3)^2)])/(.5+t)+1  ; 0<t<14

InsanityAutomation, 2019-11-17:
I put a new PR up changing the menu text to clearly state what it is. I dont like wasting the progmem on another string but oh well. It is the stored fan speed that will be used when Extruder 2 is active. Fan 1 goes to extruder 1, fan 2 goes to extruder 2.

With that, there should no longer be any confusion.

boelle, 2019-11-26:
@InsanityAutomation with the PR merged should this not have been fixed?

InsanityAutomation, 2019-11-26:
yeah, description update should stop the confusion.

boelle, 2019-11-26:
oki doke, swinging the axe then

closed 2019-11-26 by boelle