MarlinFirmware/Marlin #15716: [BUG] compile failure related to extensible-ui changes

fiveangle, 2019-10-28:
Compiling for AT90USB using config:
https://github.com/fiveangle/Marlin/blob/bf2test-2019-10-28/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/bf2test-2019-10-28/Marlin/Configuration_adv.h

Results in compile failure error:

> Marlin/src/lcd/extensible_ui/lib/lulzbot/language/language_en.h:170:24: error: expected unqualified-id before numeric constant
> PROGMEM Language_Str SPEED                    = u8"Speed";

Which seems similar issue to #15368, but @marcio-ao changes in #15453 did not appear to resolve.

Changing language string in Marlin src from "SPEED" (conflicts with Teensy libraries I believe) to "ADJUSTED_SPEED" simply resulted in a hoard of similar downstream errors like:

> In file included from Marlin/src/lcd/menu/menu_advanced.cpp:31:0:
> Marlin/src/lcd/menu/menu_advanced.cpp: In function 'void menu_dac()':
> Marlin/src/lcd/menu/../../inc/../core/language.h:319:15: error: expected unqualified-id before string constant
>  #define MSG_X "X"

I suspect the namespace isolation in  is not functioning as expected for some reason :man_shrugging: 

Compile log:
```
Processing at90usb1286_dfu (platform: teensy; framework: arduino; board: at90usb1286; build_flags: -fmax-errors=5 -g -D__MARLIN_FIRMWARE__; lib_deps: U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip, LiquidCrystal@1.3.4, TMCStepper@>=0.5.0,<1.0.0, Adafruit NeoPixel@1.2.5, LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip, Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/dev.zip, SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip, SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip, SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip; src_filter: +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared> +<src/HAL/HAL_AVR>; monitor_speed: 250000)
--------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/at90usb1286.html
PLATFORM: Teensy 3.7.0 > Atmel AT90USB1286 based
HARDWARE: AT90USB1286 16MHz, 8KB RAM, 120KB Flash
PACKAGES: framework-arduinoteensy 1.145.0 (1.45), toolchain-atmelavr 1.50400.0 (5.4.0)
avr-g++ -o "/home/speedster/dev/git/Marlin/Marlin/Marlin.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpPO1PQW"
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 103 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1 [https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/U8glib-HAL_ID1932)
|   |-- <Wire> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire)
|   |-- <SPI> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI)
|-- <LiquidCrystal> 1.3.4 (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/LiquidCrystal_ID136)
|   |-- <Wire> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire)
|-- <TMCStepper> 0.5.2 (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/TMCStepper_ID5513)
|   |-- <SoftwareSerial> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial)
|   |-- <SPI> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI)
|-- <Adafruit NeoPixel> 1.2.5 (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/Adafruit NeoPixel_ID28)
|-- <LiquidTWI2> [https://github.com/lincomatic/LiquidTWI2/archive/master.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/LiquidTWI2)
|   |-- <Wire> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire)
|-- <Arduino-L6470> 0.7.0 [https://github.com/ameyer/Arduino-L6470/archive/dev.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/Arduino-L6470)
|-- <SailfishLCD> [https://github.com/mikeshub/SailfishLCD/archive/master.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/SailfishLCD)
|-- <SailfishRGB_LED> [https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/SailfishRGB_LED)
|   |-- <Wire> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire)
|-- <SlowSoftI2CMaster> [https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/SlowSoftI2CMaster)
|-- <TMC26XStepper> [https://github.com/trinamic/TMC26XStepper/archive/master.zip] (/home/speedster/dev/git/Marlin/.pio/libdeps/at90usb1286_dfu/TMC26XStepper)
|   |-- <SPI> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI)
|-- <Wire> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire)
|-- <SPI> 1.0 (/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI)
Warning! Unknown upload protocol 
avr-g++ -o .pio/build/at90usb1286_dfu/src/src/lcd/extensible_ui/lib/lulzbot/language/language.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40003 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=145 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -IMarlin -I.pio/libdeps/at90usb1286_dfu/TMC26XStepper -I.pio/libdeps/at90usb1286_dfu/SlowSoftI2CMaster -I.pio/libdeps/at90usb1286_dfu/SailfishRGB_LED -I.pio/libdeps/at90usb1286_dfu/SailfishLCD -I.pio/libdeps/at90usb1286_dfu/Arduino-L6470/src -I.pio/libdeps/at90usb1286_dfu/LiquidTWI2 "-I.pio/libdeps/at90usb1286_dfu/Adafruit NeoPixel_ID28" -I.pio/libdeps/at90usb1286_dfu/TMCStepper_ID5513/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial -I.pio/libdeps/at90usb1286_dfu/LiquidCrystal_ID136 -I.pio/libdeps/at90usb1286_dfu/U8glib-HAL_ID1932/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire/utility -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores Marlin/src/lcd/extensible_ui/lib/lulzbot/language/language.cpp
avr-g++ -o .pio/build/at90usb1286_dfu/src/src/lcd/extensible_ui/lib/lulzbot/marlin_events.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40003 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=145 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -IMarlin -I.pio/libdeps/at90usb1286_dfu/TMC26XStepper -I.pio/libdeps/at90usb1286_dfu/SlowSoftI2CMaster -I.pio/libdeps/at90usb1286_dfu/SailfishRGB_LED -I.pio/libdeps/at90usb1286_dfu/SailfishLCD -I.pio/libdeps/at90usb1286_dfu/Arduino-L6470/src -I.pio/libdeps/at90usb1286_dfu/LiquidTWI2 "-I.pio/libdeps/at90usb1286_dfu/Adafruit NeoPixel_ID28" -I.pio/libdeps/at90usb1286_dfu/TMCStepper_ID5513/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial -I.pio/libdeps/at90usb1286_dfu/LiquidCrystal_ID136 -I.pio/libdeps/at90usb1286_dfu/U8glib-HAL_ID1932/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire/utility -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores Marlin/src/lcd/extensible_ui/lib/lulzbot/marlin_events.cpp
avr-g++ -o .pio/build/at90usb1286_dfu/src/src/lcd/extensible_ui/lib/lulzbot/screens/about_screen.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40003 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=145 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -IMarlin -I.pio/libdeps/at90usb1286_dfu/TMC26XStepper -I.pio/libdeps/at90usb1286_dfu/SlowSoftI2CMaster -I.pio/libdeps/at90usb1286_dfu/SailfishRGB_LED -I.pio/libdeps/at90usb1286_dfu/SailfishLCD -I.pio/libdeps/at90usb1286_dfu/Arduino-L647 [... cut ...]
                 from Marlin/src/lcd/extensible_ui/lib/lulzbot/language/../../../../../Marlin.h:24,
                 from Marlin/src/lcd/extensible_ui/lib/lulzbot/language/language.cpp:23:
Marlin/src/lcd/extensible_ui/lib/lulzbot/language/language_en.h:170:24: error: expected unqualified-id before numeric constant
   PROGMEM Language_Str SPEED                    = u8"Speed";
                        ^
*** [.pio/build/at90usb1286_dfu/src/src/lcd/extensible_ui/lib/lulzbot/language/language.cpp.o] Error 1
========================== [FAILED] Took 2.80 seconds ==========================
```

boelle, 2019-11-03:
Likely to be a Duplicate of #15368

@fiveangle many updates have passed in last 6 days, is the issue still present in 2.0.x ?

fiveangle, 2019-11-04:
As I mentioned above, it is not an exact duplicate of #15368, and the resolution proposed by @marcio-ao did not appear to resolve.  No change.

fiveangle, 2019-11-09:
Same behavior as of 8061836e740920648aad71bb6ffd47de181e61b015844

fiveangle, 2019-11-13:
Latest compile error log of config https://github.com/fiveangle/Marlin/blob/bf2-extensible_ui_failures/ as of now (227951a4ec4f9c5f0d594253465e988918d7d33c):

```
avr-g++ -o .pio/build/at90usb1286_dfu/src/src/lcd/menu/menu_led.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fsingle-precision-constant -fmax-errors=5 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40100 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=145 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -IMarlin -I.pio/libdeps/at90usb1286_dfu/TMC26XStepper -I.pio/libdeps/at90usb1286_dfu/SlowSoftI2CMaster -I.pio/libdeps/at90usb1286_dfu/SailfishRGB_LED -I.pio/libdeps/at90usb1286_dfu/SailfishLCD -I.pio/libdeps/at90usb1286_dfu/Arduino-L6470/src -I.pio/libdeps/at90usb1286_dfu/LiquidTWI2 "-I.pio/libdeps/at90usb1286_dfu/Adafruit NeoPixel_ID28" -I.pio/libdeps/at90usb1286_dfu/TMCStepper_ID5513/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SoftwareSerial -I.pio/libdeps/at90usb1286_dfu/LiquidCrystal_ID136 -I.pio/libdeps/at90usb1286_dfu/U8glib-HAL_ID1932/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire/utility -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores Marlin/src/lcd/menu/menu_led.cpp
In file included from Marlin/src/lcd/menu/menu_advanced.cpp:31:0:
Marlin/src/lcd/menu/menu_advanced.cpp: In function 'void menu_dac()':
Marlin/src/lcd/menu/../../inc/../core/language.h:319:15: error: expected unqualified-id before string constant
 #define MSG_X "X"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:48: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:66: note: in expansion of macro 'GET_TEXT'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_X'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:69:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(X);
     ^
Marlin/src/lcd/menu/../../inc/../core/language.h:320:15: error: expected unqualified-id before string constant
 #define MSG_Y "Y"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:48: note: in expansion of macro 'EDIT_ITEM_P'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                ^
Marlin/src/lcd/menu/menu.h:381:66: note: in expansion of macro 'GET_TEXT'
 #define EDIT_ITEM(TYPE, LABEL, V...)           EDIT_ITEM_P(TYPE, GET_TEXT(LABEL), ##V)
                                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:33: note: in expansion of macro 'EDIT_ITEM'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                 ^
Marlin/src/lcd/menu/menu_advanced.cpp:68:50: note: in expansion of macro 'MSG_Y'
     #define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })
                                                  ^
Marlin/src/lcd/menu/menu_advanced.cpp:70:5: note: in expansion of macro 'EDIT_DAC_PERCENT'
     EDIT_DAC_PERCENT(Y);
     ^
Marlin/src/lcd/menu/../../inc/../core/language.h:321:15: error: expected unqualified-id before string constant
 #define MSG_Z "Z"
               ^
Marlin/src/lcd/menu/menu.h:348:26: note: in definition of macro '_MENU_ITEM_P'
     PGM_P const plabel = PLABEL;                                \
                          ^
Marlin/src/lcd/menu/menu.h:380:48: note: in expansion of macro 'MENU_ITEM_P'
 #define EDIT_ITEM_P(TYPE, PLABEL, V...)        MENU_ITEM_P(TYPE, PLABEL, ##V)
[... cut ...] me/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire/utility -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores Marlin/src/lcd/menu/menu_media.cpp
========================== [FAILED] Took 3.51 seconds ==========================
```

sjasonsmith, 2019-11-13:
The root-case is right here:
`#define EDIT_DAC_PERCENT(A) EDIT_ITEM(uint8, MSG_##A " " MSG_DAC_PERCENT, &driverPercent[_AXIS(A)], 0, 100, []{ dac_current_set_percents(driverPercent); })`

There may be an issue with the macro usage, but the more significant issue is that  MSG_DAC_PERCENT is no longer a string literal, so the build-time concatenation used here doesn't work.

I spent some time looking at this the other day, but couldn't figure out a way to concatenate these at build time. Perhaps someone else will have a recommendation, but it might need to be done at runtime.

boelle, 2019-11-26:
@fiveangle btw what version of marlin are you using?

boelle, 2019-12-01:
@sjasonsmith any ideas to how this one could be solved?

sjasonsmith, 2019-12-03:
No, I seem to recall a recent changelist from @thinkyhead regarding concatenating labels, I don't know whether that could be utilized here or is something entirely different.

boelle, 2019-12-07:
@fiveangle since 2.0 was just released a few days ago has this changed this issue at all?

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle

fiveangle, 2020-02-07:
> btw, who can confirm there is an issue? use the same configs as OP and the same hardware if possible

There is no requirement to have the hardware to validate a toolchain compile problem. Compile fails with the config files provided and there is no board-specific runtime libraries required.  @boelle - if you want to provide even more excellent value to the project than you currently do, any of us can walk you through the process of setting up a generic PlatformIO build environment (if you find you can't get it working when following the existing Marlin documentation).  With the power to confirm/deny yourself, you will know yourself immediately if an issue is replicatable for any issues where someone has provided the config files (and no board-specific runtime libraries are required - for those instances, you would handle similar to how you do now, although I would suggest to wait more than a mere 3 weeks before closing, as the wheels of open source grind slowly ;).

I'm sure you would prefer not to close issues that are still valid, right ?   This is why we get down on users who do not provide the configs when they open issues: it's nigh impossible sometimes to determine how the heck to replicate their toolchain problem without the configs.  But on the flip side, it is precisely why we should never close an issue where someone has provided configs, unless someone has confirmed that when using the configs that they cannot replicate the toolchain failure.

Make sense ?

-=dave

thinkyhead, 2020-02-07:
Indeed. Attempting to compile with your configs is what everyone aiming to move issues forward should endeavor to do. And those who can solve the problem should solve it and not move on from the issue until all avenues are exhausted.

I ask you to please reserve technical requests for the coder types. Bo is graciously helping us to figure out which issues still need attention, but he is not a coder or a technician. You will note that some of these replies are canned, and they are just part of the suite that gets thrown at issues he is not sure about to see what sticks.

So, like, let me see what's up with these configs of yours and what the malfunction is…

thinkyhead, 2020-02-07:
Success! I dropped your configurations into a copy of `bugfix-2.0.x` and ran `pio run -e at90usb1286_dfu` and it compiled perfectly. I think we can call this one solved!

fiveangle, 2020-02-07:
Thanks for checking @thinkyhead ! (I am travelling so could not check myself when I saw this earlier today)

fiveangle, 2020-02-07:
Digging around a bit, it looks like @drphil3d did confirm in https://github.com/MarlinFirmware/Marlin/issues/15302#issuecomment-569128416 that with PRs https://github.com/MarlinFirmware/Marlin/pull/15347 and https://github.com/MarlinFirmware/Marlin/pull/16346 applied, the compile failure I reported in https://github.com/MarlinFirmware/Marlin/issues/15302 (which was inadvertently closed, prompting me to open this one) was in fact resolved.  I wasn't aware of it as the issue had long been closed without resolution and so my notification settings hadn't alert me.  Again, illustrating how closing without clear resolution can often lead to confusion down the road :shrug:

thinkyhead, 2020-02-08:
My things to do list is astonishing and I've got about 1600 unread issues in my hopper at the moment so it is definitely easy for some to get lost and forgotten as new ones arrive. We encourage submitters to become the keepers of their issues, to follow up and nag us (gently) if needed. We definitely want to resolve extant issues, especially the bothersome ones that are hard to figure out.