MarlinFirmware/Marlin #15717: 5 independently homing axis with Endstops

twall7407, 2019-10-28:
### Description

I am trying to control a 5-axis system with Marlin and Ramps 1.4. I am currently controlling 5 stepper motor axis with Marlin/Ramps but have not been able to add homing endstops to two of the axis. I am controlling 3 of the motors with the X, Y and Z motor controller locations on Ramps and the other two as extruders. I am only using 1 endstop for each axis so I would like to re-purpose one of the open endstop ramps pins (ex Xmax or Ymax) for the 2 extra axis.

I have tried to follow the instructions from this [earlier marlin Issue post](https://github.com/MarlinFirmware/Marlin/issues/8519), but cannot get it to compile.

Is it possible to add this functionality to Marlin? Can anyone help provide any guidance or resources for modifying the firmware to add two endstops for 5 axis homing?

DerAndere1, 2019-10-30:
In case you tried my approach, I do not know if that works for more than one e stepper. If you only move either E0 or E1 (one at a time),  you could wire the limit switches for the E0 and E1 axis in parallel to one shared e endstop pin. https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot/Marlin  compiles for ONE E-stepper if you have changed your config file so it matches the default config files at https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot/Marlin defined the pin for the e endstop in the RAMPS pins file. If you need support with that specific approach, I suggest to open an issue at https://github.com/DerAndere1/Marlin/issues and attach your failing config files + exact compile errors.

DerAndere1, 2020-04-07:
Update: https://github.com/DerAndere/Marlin/tree/bf2_6axis_dev adds basic support for -in principle- up to 6 axes XYZIJK plus extruders (E). Tested on some 4-5 axis test rigs. Only G codes G1, G28, G90, G91, G92 (partially), M43, M114, M400, M500, M501, M502 work so far. Only single A4988 drivers per axis and limited support for dogm display so far. Updating the code to recent Marlin bugfix-2.0.x (see https://github.com/DerAndere1/Marlin/tree/bf2_6axis_dev13), gives a remaining issue with the feedrate of additional axes IJK (DerAndere1#7) that should be solved first. Everyone is welcome to test and improve the code or add support for additional G-codes and kinematics.

closed 2020-06-30 by github-actions[bot]