MarlinFirmware/Marlin #15718: 1.1.9 & 1.1.x & 1.1.x bugfix & 2.0 - Kossel Linear Plus Leveling issue

brosskgm, 2019-10-29:
Having tried 1.1.9 from https://www.thingiverse.com/thing:3612533 and 1.1.9 & 2.0 from http://www.lpomykal.cz/3d-printers/kossel/kossel-marlin-firmware/ and config files from https://www.thingiverse.com/thing:3071086 - Every one and every version does not bilinear level all the bed. Level bed does not go to the left side of the bed beyond the white line in any of the three versions with multiple installs of each. These are the config files I remember. I've been trying for about a week to figure this out. But these three have the same issue.

The Marlin download I tried to configure and only get a lot of bed crashes.

[image]

[Maviles798_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3781355/Maviles798_Configuration.zip)
[Configuration_LPomykal.zip](https://github.com/MarlinFirmware/Marlin/files/3781358/Configuration_LPomykal.zip)
[Anycubic_Linear_Plus_cfg.zip](https://github.com/MarlinFirmware/Marlin/files/3781361/Anycubic_Linear_Plus_cfg.zip)

thisiskeithb, 2019-10-29:
Have you tried the Linear Plus config from this repo? We can only help with code from here and not other sources.

brosskgm, 2019-10-29:
I tried to set the config in the download but ground the end into the bed.

Only one place LPomykal provided the full code download. All the other full code downloads was from yours and the others only provided configuration.h files. I could have tried yours but now not sure. Do you have a link to make sure I use the correct one and I can run it again?

Thanks for your reply.
Bob

brosskgm, 2019-10-29:
Is it ok to use 2.0 since this printer only has the TriGorilla boards. The one I have doesn't have a version number on it.

brosskgm, 2019-10-29:
I guess I should say the examples didn't say linear plus.

brosskgm, 2019-10-29:
I did notice there is dups in the DELTA_PRINTABLE_RADIUS one has 116 and the other under the 90 and under DELTA_DIAGONAL_ROD one has 267 and the other has 218 and none are commented out. Is this the way they should be?

#if ENABLED(ANYCUBIC_KOSSEL_PLUS)
    // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
    #define DELTA_PRINTABLE_RADIUS 116.0 // mm
    // Center-to-center distance of the holes in the diagonal push rods.
    #define DELTA_DIAGONAL_ROD 267 // mm
      // Horizontal offset from middle of printer to smooth rod center.
    #define DELTA_SMOOTH_ROD_OFFSET 186 // mm
    // Horizontal offset of the universal joints on the end effector.
    #define DELTA_EFFECTOR_OFFSET 31 // mm
    // Horizontal offset of the universal joints on the carriages.
    #define DELTA_CARRIAGE_OFFSET 20.6 // mm
    // Horizontal distance bridged by diagonal push rods when effector is centered.
    #define DELTA_RADIUS (DELTA_SMOOTH_ROD_OFFSET-(DELTA_EFFECTOR_OFFSET)-(DELTA_CARRIAGE_OFFSET))  //mm  Get this value from auto calibrate
  #else
    // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
    #define DELTA_PRINTABLE_RADIUS 90.0 // mm
    // Center-to-center distance of the holes in the diagonal push rods.
    #define DELTA_DIAGONAL_ROD 218.0 // mm
    // Horizontal distance bridged by diagonal push rods when effector is centered.
    #define DELTA_RADIUS 97.0 //mm  Get this value from auto calibrate
  #endif

brosskgm, 2019-10-29:
1st try. Auto Calibration. Goes to center and test a few areas, and then moves 1/2 way to outer edge and tests in that circle. Never makes it to the 10-20mm at the other edge. Iteration 3 so far. Seems to be 40 or so mm away from edge. My bed is 240 with the Ultrabase.

brosskgm, 2019-10-29:
I will check in tomorrow. Shutting down for the night.

Thanks for your help.

Bob

boelle, 2019-10-29:
tried to download marlin direct?

brosskgm, 2019-10-29:
Thank you for your reply.

Yes sir.
Was yours direct 1.1.x not the bug fix yet because my Trigorilla board is RAMPS 1.4 Protocol 1.0, Thermisters ATC 104GT-2 so I'm not sure 2.0 will work. Have read it needs to be 32bit.

It is also your config file CONFIGURATION_H_VERSION 010109 that I changed to 010110 with the changes I was making for the linear probe, ultrabase bed and REVERSE_MENU_DIRECTION and REVERSE_ENCODER_DIRECTION. Nothing major just the little things. Not sure if I should change the number with minimal changes, can always put it back.

The note I posted above, the DELTA_PRINTABLE_RADIUS and DELTA_DIAGONAL_ROD settings having duplicate etc was in your config.

boelle, 2019-10-29:
no only use bugfix 2.0.x, that is the one that is updated, the others are old stuff

boelle, 2019-10-29:
you can use a tool like winmerge to copy over old setting to a fresh download

brosskgm, 2019-10-29:
Recommendation to use with Ubuntu 18.04? I don't have any windows machines for some time. I've only used the two configuration files. I could go through and just change the new ones if the number is different.

boelle, 2019-10-29:
i only have windows

brosskgm, 2019-10-29:
Ok. Not a problem. I have a virtual of Win 10 if I need it.

Yea, a lot of changes to the configuration. The examples only have the i3 in it this time. Will 2.0 bugfix it work with Kossel Linear Plus?

I guess it's a good thing I don't have Linux Slackware only. Lol. Been using slack for 24 years. Ubuntu since v11.0

brosskgm, 2019-10-29:
The 1.1.x bugfix has kossel in it with same configuration file. 2.0 says Proceed with caution. I'll wait on 2.0 until is a release, not beta.

thisiskeithb, 2019-10-29:
> The examples only have the i3 in it this time. Will 2.0 bugfix it work with Kossel Linear Plus?

There are `bugfix-2.0.x` delta configs and [specifically one for the Anycubic Kossel series (read the readme)](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/delta/Anycubic/Kossel).

brosskgm, 2019-10-29:
ok

brosskgm, 2019-10-29:
It still has the duplicate declarations in it with different values.

#if ENABLED(ANYCUBIC_KOSSEL_PLUS)
    // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
    #define DELTA_PRINTABLE_RADIUS 116.0  // (mm)
    // Center-to-center distance of the holes in the diagonal push rods.
    #define DELTA_DIAGONAL_ROD 267        // (mm)
      // Horizontal offset from middle of printer to smooth rod center.
    #define DELTA_SMOOTH_ROD_OFFSET 186   // (mm)
    // Horizontal offset of the universal joints on the end effector.
    #define DELTA_EFFECTOR_OFFSET 31      // (mm)
    // Horizontal offset of the universal joints on the carriages.
    #define DELTA_CARRIAGE_OFFSET 20.6    // (mm)
    // Horizontal distance bridged by diagonal push rods when effector is centered.
    #define DELTA_RADIUS (DELTA_SMOOTH_ROD_OFFSET-(DELTA_EFFECTOR_OFFSET)-(DELTA_CARRIAGE_OFFSET))          // (mm) Get this value from G33 auto calibrate
  #else
    // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
    #define DELTA_PRINTABLE_RADIUS 90.0   // (mm)
    // Center-to-center distance of the holes in the diagonal push rods.
    #define DELTA_DIAGONAL_ROD 218.0      // (mm)
    // Horizontal distance bridged by diagonal push rods when effector is centered.
    #define DELTA_RADIUS 97.0             // (mm) Get this value from G33 auto calibrate
  #endif

brosskgm, 2019-10-29:
I'll work on getting 2.0 installed. I'll have plenty of time to work it if I don't have to work in a couple hours. But will work in it until then.

Vertabreak, 2019-10-29:
you can install Arduino or vscode on Ubuntu natively for compiling as well as flashing.
i have been compiling marlin 2 in Ubuntu 19.10 for over a year and have both IDE's installed. 
best way to get the latest Arduino is with "umake ide arduino" run from terminal.
vscode has a .deb installer https://code.visualstudio.com/download

brosskgm, 2019-10-29:
Thanks... I already have Arduino mentioned above 1.8.10. Been only using Marlin this year when I installed in on my Mega S when I got it in Jan. But that install went so easy I never really had to go in to it as deep. I've programed several with Arduino boards, but much smaller for an ESR meter I built and other little things. 3D has been a learning experience. This Kossel upgrade has been all week so far. I've had it since Jan also. I will get diffuse. But it won't find much. I haven't made that many changes to the configs. Just minor that I really might not need it.

brosskgm, 2019-10-29:
I know of Visual Studio. I might have used it on some Python or C code code some time ago thou. Forgot about it. I'll have to get it so I can look at code a little easier than text editor. Lol.

brosskgm, 2019-10-29:
To thisiskeithb

> 
> There are `bugfix-2.0.x` delta configs and [specifically one for the Anycubic Kossel series (read the readme)](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/delta/Anycubic/Kossel).

To let you know. The Kossel config in your link has most missing, and refers to the examples that only has i3 in it no delta.

// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

brosskgm, 2019-10-29:
Found it.

brosskgm, 2019-10-29:
The file in the Kossel folder referenced says the same thing as the one in the Marlin folder.

// For a Delta printer start with one of the configuration files in the // config/examples/delta directory and customize for your machine.
//

brosskgm, 2019-10-29:
Using diffuse now. I like it.

brosskgm, 2019-10-29:
It shows way to many differences and I don't know what most should even be used or for what.

brosskgm, 2019-10-29:
I went through the one in the example folder and made the minor changes for the bed, Linear etc. And will run it shortly.

brosskgm, 2019-10-29:
Compiled/uploaded no errors.

In your readme: Then you should execute Configuration > Delta Calibration > Set Delta Height 
It's under Configuration. but no selection Set Delta Height.

Then:  and also run Configuration > Delta Configuration > Probe Z-offset to verify the Probe offset. 
No such selection. There is only Configuration and Delta Calibration.

brosskgm, 2019-10-29:
Running Auto Calibrate now. Sure slams into the endstops between Iterations. Might need to set height in the config. It had 320. Linear is 280 - 300

brosskgm, 2019-10-29:
Well, after bed leveling it still does not go to the left side of bed to the left of white line submitted above.

brosskgm, 2019-10-29:
Also does not even touch the word Ultrabase at all during level. Goes far enough on the opposite side and far enough on the right.

brosskgm, 2019-10-29:
This is your 2.0.x bugfix and config files downloaded today.

brosskgm, 2019-10-29:
Not sure this will help, under delta settings it shows the Radius +133.18

brosskgm, 2019-10-29:
Other than that I'll wait to see if you find a setting to make it level the whole bed.

I'll look through the Anycubic Marlin code and see if I can find what they did and if I find something I'll send it. If you can let me know the file that covers that could help. They say it's Marlin but no .ino files. All hard coded.

boelle, 2019-10-29:
could you please attach your current configuration.h and configration_adv.h as a zip file

brosskgm, 2019-10-29:
Found their .ino, it was named different.

brosskgm, 2019-10-29:
Ok

brosskgm, 2019-10-29:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3784624/Configuration.zip)
Here you go.

brosskgm, 2019-10-29:
I did find their delta settings. They are not to different from yours.

#define DELTA

#if ENABLED(DELTA)

  // Make delta curves from many straight lines (linear interpolation).
  // This is a trade-off between visible corners (not enough segments)
  // and processor overload (too many expensive sqrt calls).
  #define DELTA_SEGMENTS_PER_SECOND 80

  // NOTE NB all values for DELTA_* values MUST be floating point, so always have a decimal point in them

  // Center-to-center distance of the holes in the diagonal push rods.
  #define DELTA_DIAGONAL_ROD 271.5//mm

  // Horizontal offset from middle of printer to smooth rod center.
//  #define DELTA_SMOOTH_ROD_OFFSET 182// 

  // Horizontal offset of the universal joints on the end effector.
//  #define DELTA_EFFECTOR_OFFSET 33 // mm

  // Horizontal offset of the universal joints on the carriages.
//  #define DELTA_CARRIAGE_OFFSET 15.5 // mm

  // Horizontal distance bridged by diagonal push rods when effector is centered.
  #define DELTA_RADIUS 135.4 //(DELTA_SMOOTH_ROD_OFFSET-(DELTA_EFFECTOR_OFFSET)-(DELTA_CARRIAGE_OFFSET))

  // Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
//  #define DELTA_PRINTABLE_RADIUS 120
#define DELTA_PRINTABLE_RADIUS 105//118
  // Delta calibration menu
  // uncomment to add three points calibration menu option.
  // See http://minow.blogspot.com/index.html#4918805519571907051
  // If needed, adjust the X, Y, Z calibration coordinates
  // in ultralcd.cpp@lcd_delta_calibrate_menu()
 // #define DELTA_CALIBRATION_MENU

  // After homing move down to a height where XY movement is unconstrained
  #define DELTA_HOME_TO_SAFE_ZONE

  //#define DELTA_ENDSTOP_ADJ { 0, 0, 0 }

#endif

brosskgm, 2019-10-29:
To let you know in case it might help. The Auto Calibrate hits all the other edges just fine. Auto Level Bilinear misses the top and left.

boelle, 2019-10-29:
you might also be able to get help at one of these places

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

brosskgm, 2019-10-29:
To get it to hit all spots of the disk during leveling?

brosskgm, 2019-10-29:
Ok. I'll keep looking.

closed 2019-10-29 by brosskgm