MarlinFirmware/Marlin #15720: Thermal Runaway (ACTUAL) Marlin 2.0 bugfix

VinceW31, 2019-10-29:
Just had a thermal runaway on my extruder, smoke everywhere!  
Ive just upgraded to Marlin 2.0 bugfix because Im upgrading from my old RAMPS 1.4 board to a new 32 bit Bigtreetech SKE v1.3 board.  I updated the firmware as per Bigtreetech instructions etc and  copied across all my old settings from configuration.h and adv.h that worked fine for my printer.  Everything worked fine until I got to testing my extruder heater on E0.  I decided to do a PID autotune first to get the correct settings for my printer with M303 E0 S200 C8 (I normally print at 215 deg).  I used Repetier (but Pronterface had the same results) to monitor the extruder temp and see what was going on.  The extruder temp climbed quickly in a steady line up towards the 200 deg point and then immediately flat lined without any smoothing or damping, I was really impressed at first, fantastic  PID control I thought.  But after about 5 seconds or so smoke started to come from the extruder head and it was getting worse by the second even though the display still showed a perfectly flat line at 200 deg.  It was obvious the head was still on full power and still heating even though the Repetier display showed otherwise!  I immediately unplugged everything and managed to stop it before any real physical damage.

I normally run my head at 215 deg, so 200 should be no problem for it.  After it all cooled down I heated the extruder once again manually this time in steps of 10 deg starting at 100 to see if it controlled ok at lower temps which it did, but when I got to 200 again the Repetier graph display flat lined once more and I quickly unplugged it all again as I had some doubts that it was in thermal runaway mode again.  

After a lot of investigation I discovered that in my old Marlin 1.1.x settings the extruder thermistor was set to index type 75 and HEATER_0_MAXTEMP was set to 250, these settings all worked fine for me before.  So in Marlin 2.0 Id set the thermistor to type 75 again and lowered the HEATER_0_MAXTEMP to 220 as good practice because I never intend to go above 215 in daily use.  So I altered the thermistor type to index 1 (100k thermistor) and tried again, this time setting the extruder temp in steps of 1 deg going upwards from 190 deg, but this time the firmware wouldn't let me select a temp higher then 205 deg in Repetier, if I tried to select 206 deg the display immediately reset itself to 205 deg effectively limiting me even though it wasnt at the Max point on the slider display and the HEATER_0_MAXTEMP was set to 220.  Then I went back into Marlin and changed the HEATER_0_MAXTEMP back to its default 275 and it seems to work fine now, I can increase the temp selection above 205 all the way up to 215 (I didnt try any further).

The problem is that the firmware did not detect that the extruder temp was still increasing after it flat lined on the Repetier display at 200 deg and it didnt control the temp above that value with those safe settings I used.  The thermistor is working fine and has done throughout with no glitches, spikes, open or short circuit indications and still gives a nice smooth display even now with the revised settings Im using.  Im convinced its a Marlin 2.0 issue thinking the temp is still constant at 200 deg for some reason, so it thinks it still needs to heat up because a 200 deg flat line is the info its getting from the thermistor code and this is whats displayed back to the Repetier software.

Patag, 2019-10-29:
There is a kind of safety margin on HEATER_0_MAXTEMP : you can't heat over (HEATER_0_MAXTEMP - 15degC ). If you need 215, set HEATER_0_MAXTEMP to 230

VinceW31, 2019-10-29:
Thanks for the info re HEATER_0_MAXTEMP - 15degC .
Thermal runaway issue still unresolved.

Patag, 2019-10-29:
Did you try to launch PID Auto tune using LCD menu ?  (#define PID_AUTOTUNE_MENU needed)
This may help to visualise actual temperature on LCD screen.
BTW, Config.h and Config_adv.h attached would be helpful...

Haschtl, 2019-10-29:
I wonder, why you are using thermistor 75 for the hotend?
The lookup table reaches only from 0 to 200°C, which could be the cause of your clipping temperature at 200°C. Anyway the LUT for thermistor 1 reaches from 0-300°C, so this should be fine.

VinceW31, 2019-10-30:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/3788263/Configuration_adv.zip)
I dont have an LCD fitted to my printer, Im using my PC as the interface with Repetier.
Im now using hotend thermistor index 1 as suggested and even trying to cal my thermistor to use with index 1000 and the settings in adv.h
With thermistor index 1 selected it seems to work fine because as you pointed out it never gets to the max temp defined in the LUT, this is where the temp reading just flat lines and the runaway occurs.  
I just copied the settings over from my Marlin 1.1.x files and thats where the thermistor 75 setting originally came from; I should have checked it but it was working fine in there.
Config files attached as requested, you can see Ive been messing with the thermistor 1000 settings in adv.h as I try to cal my hotend thermistor.

thisiskeithb, 2019-10-30:
Do you have a different/generic 100k thermistor to test with to rule that out?

LinFor, 2019-10-30:
Very strange issue.
Recently I got into a similar situation (adc reading outside the thermistor table) and immediately got a halt error (by these checks): https://github.com/MarlinFirmware/Marlin/blob/61225e685d80b34af582d833646a2498eacec577/Marlin/src/module/temperature.cpp#L2255

boelle, 2019-11-03:
@VinceW31 a lot of updates have been added to 2.0.x in the last 4 days, do you still have an issue?

VinceW31, 2019-11-03:
@boelle Ill check with the latest

boelle, 2019-11-11:
@VinceW31 what did thest show?

VinceW31, 2019-11-11:
@boelle 
Just tried it and the problem is still there in the latest version of Marlin 2.0 I downloaded 1 hr ago.
With thermistor 75 set as the Extruder thermistor (yes, I know this is the wrong one) then when I select an extruder temp of 215 deg (my normal PLA temp) the extruder heats up as normal initially but then the temp display in Repetier flat lines solid at 200 deg but the heater stays on full power leading to a thermal runaway.  No warnings or attempts to control the runaway from the firmware.

I know thermistor 75 is not correct for a hotend (it works ok when I select number 1), but a simple mistake like this can cause an undetected thermal runaway.  I checked my old 1.1.9 files and I also had thermistor 75 mistakenly selected in those too (thats where I copied my setting from) and never had a thermal runaway with 1.1.9 with this setting.

Marlin 2.0 works fine with thermistor 1 selected, so Im happy that it working ok when configured properly.  Maybe it just needs some logic check to make sure the temp cannot be selected higher than the thermistor LUT max value or something.

I will upload a vid of it running away later on if I can.

Steps to reproduce:
I uploaded a clean version of Marlin 2.0 to my Bigtreetech SKR V1.3 board just now with the min changes required to make it compatible with the board, these are 
Configuration.h
#define SERIAL_PORT -1

#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3

#define TEMP_SENSOR_0 75
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_5 0
#define TEMP_SENSOR_BED 0
#define TEMP_SENSOR_CHAMBER 0

Platformio.ini
default_envs = LPC1768

Ive attached the config files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3832939/Marlin.zip)

VinceW31, 2019-11-11:
@boelle 
attached is a photo of the Runaway display in Repetier, clearly shows thermistor working ok.
[image]

AnHardt, 2019-11-11:
Table 75 ends at 200°C. You will never measure a higher temperature with it.

Set MAXTEMP not higher than 200°C with it. That's your responsibility.
MAXTEMP is **NOT** the maximum temperature you want to print with. It's ought to be the maximum temperature your hardware can stand for a unlimited time. If the table has no values above 200°C that's a strong indicator for the thermistor is not ought to be used at that high temperatures.

```cpp
const short temptable_75[][2] PROGMEM = { // Generic Silicon Heat Pad with NTC 100K MGB18-104F39050L32 thermistor
  { OV(111.06), 200 }, // v=0.542 r=571.747 res=0.501 degC/count

  #ifdef HIGH_TEMP_RANGE_75
    { OV(174.87), 175 }, // v=0.854 r=967.950 res=0.311 degC/count  These values are valid.  But they serve no
    { OV(191.64), 170 }, // v=0.936 r=1082.139 res=0.284 degC/count  purpose.  It is better to delete them so
    { OV(209.99), 165 }, // v=1.025 r=1212.472 res=0.260 degC/count  the search is quicker and get to the meaningful
    { OV(230.02), 160 }, // v=1.123 r=1361.590 res=0.239 degC/count  part of the table sooner.
    { OV(251.80), 155 }, // v=1.230 r=1532.621 res=0.220 degC/count
  #endif

  { OV(275.43), 150 }, // v=1.345 r=1729.283 res=0.203 degC/count

  #ifdef HIGH_TEMP_RANGE_75
    { OV(300.92), 145 }, // v=1.469 r=1956.004 res=0.189 degC/coun
  #endif

  { OV( 328.32), 140 }, // v=1.603 r=2218.081 res=0.176 degC/count
  { OV( 388.65), 130 }, // v=1.898 r=2874.980 res=0.156 degC/count
```

AnHardt, 2019-11-11:
Likely the thermistor itselves can stand more than 200°C.
For the 'silicone heat pads', this table is made for, usually the M3-glue, holding the pad below the bed, is the limiting factor. 
From https://3m.citrination.com/pif/000286?locale=en-US
```
149 °C | 300 °F | Short Term (minutes, hours)
121 °C | 250 °F | Long Term (days, weeks)
```

VinceW31, 2019-11-12:
@AnHardt 
As I said, I know setting thermistor 75 was the wrong one, Ive now set it to number 1 and its all ok. The point is that all I did (as a generic 3D printer user) was to move over from 1.1.9 to Marlin 2.0 with the same Config settings (I incorrectly had thermistor 75 set in 1.1.9 also) and I got an undetected thermal runaway when doing a PID tune with Marlin 2.0.

boelle, 2019-11-12:
so with correct settings everything was fine?

VinceW31, 2019-11-12:
With the correct extruder thermistor setting the temp display never goes out of range defined in the relevant LUT, so the thermal runaway will not happen because everything is within the limits monitored by the firmware.  However, the firmware does not detect when the extruder temp is close to approaching or exceeding the Max limit (defined by the LUT) and provide a warning or switch off the heating element.  This would be easy to implement if its not a Bug.

boelle, 2019-11-12:
and the more strange thing is that you seem to be the only one having this issue

where did you get marlin from?

VinceW31, 2019-11-12:
downloaded straight from Github (this site)

Grogyan, 2019-11-12:
Any chance you can try a different thermistor? @VinceW31

boelle, 2019-11-12:
> downloaded straight from Github (this site)

had to ask since some people download from elsewhere and ask for help here

boelle, 2019-11-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-11-24 by boelle