MarlinFirmware/Marlin #15723: [FR Support/help] (Need Idea/expertise to solve/find a solution) For skilled devs

studiodyne, 2019-10-29:
Hi, I want to have a smart runout system by using a 'extrusion segmentation' 

Why ? 
Runout sensors are read after each block completed. Then Extrusion segmentation is the only solution for all problems , but cost computation and feedrate , but give serious protection by reducing the extrusion by the ' encoder wheel pulse distance ' , see my PR , all is explained

Problems:
- If i slice the G1 command  or the planner block: 
   - M600 script used to start runout will never start while my sliced command finish , runout will have no effects
   - Power loss recovery will loose my sliced command , 'now it loose also , but , i want to give power loss , the possibility to use my sliced command , to have a better resumed print
  
My ideas : 
   -A menu to convert sdfile ' all G1>Sensor pulse distance are sliced ' , easy to integrate in marlin
      -but for user on computers .... need , a windows/mac/linux app to slice it , while slicers give this option 😄 
   -Slicing G1 command/planner block in real time(with disabling of Linear advance , because no direction change or brakes) , with M600 death and definitive end of M600 script for runout & no benefits for Power loss 😢 
 
Someone with a better idea ?

boelle, 2019-10-29:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2019-10-29 by boelle

studiodyne, 2019-10-29:
@boelle 
I'm ok , but where i can find a better help for coding except here ....
I just need ideas , and sometimes i need to be help in writing , or debugging my own
What do i do , when i need help to build my PR ?
thks

boelle, 2019-10-29:
ask here:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

studiodyne, 2019-10-29:
Commodore 64 ? 👍 
I was so young .... and this computer have made discover me , the ' peek and poke ' commands
To set directly the screen memory . The ' circle command ' existed , but not in amstrad , i did plot each point by sin and cos maths to draw a circle
Thks for help

shitcreek, 2019-10-29:
Join the Discord channel: https://discord.gg/n5NJ59y