MarlinFirmware/Marlin #15724: error: missing binary operator before token "4013" when compiling

iLLDrDope, 2019-10-29:
Here is a copy of my error log. Trying to compile firmware for an SKR Mini E3 board and get the following errors. I've tried downloading Marlin 2.0 from the official git page as well as from BTT with the same results.

> Executing task in folder Marlin2: C:\Users\Rick\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5 #58c47ee
|-- <Adafruit NeoPixel> 1.2.4 #b72627a
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <STM32ADC> 1.0
|-- <USBComposite for STM32F1> 0.91
|-- <EEPROM>
|-- <Wire> 1.0
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\i2c_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\iwdg.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\nvic.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\pwr.c.o
[... 188 lines cut ...]
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
======================================================================================== 1 failed, 0 succeeded in 00:00:48.687 ======================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

boelle, 2019-10-29:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

and a small tip is to use a site like paste bin for logs etc

iLLDrDope, 2019-10-29:
Thank you for fixing the title and moving the log to pastebin. I'm new here.

As requested; 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3786137/Configuration.zip)

reloxx13, 2019-10-30:
the error is selfexplaining

```
#ifndef MOTHERBOARD
  #define MOTHERBOARD BMOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2
#endif
```

```
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_STM32F103RE
#endif
```

use the example config to start up
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/config/examples/STM32/STM32F103RE/Configuration.h

reloxx13, 2019-11-02:
@iLLDrDope close this if its fixed!

closed 2019-11-02 by iLLDrDope