MarlinFirmware/Marlin #15725: [BUG] "#define NOZZLE_TO_PROBE_OFFSET" does not work when using ABL

yudi-jpn, 2019-10-30:
[Bug content]
Even if #define NOZZLE_TO_PROBE_OFFSET {x, y, z} is defined, the definition is not reflected and the Z-axis stepping motor strokes at the position of X0, Y0.

The stepping motor slips because the head interferes with the bed outside the operating range.

[Problem confirmation procedure]
Marlin-bugfix-2.0.x / CONFIGURATION_H_VERSION 020000
Ener-3, BL-touch
Definition
#define AUTO_BED_LEVELING_3POINT
#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
#define BLTOUCH
#define NOZZLE_TO_PROBE_OFFSET {-43, -5, 0}
#define LCD_BED_LEVELING

[Remarks]
In Marlin-1.1.x (latest past release),
[#define X_PROBE_OFFSET_FROM_EXTRUDER x]
[#define Y_PROBE_OFFSET_FROM_EXTRUDER y]
[#define Z_PROBE_OFFSET_FROM_EXTRUDER z]
I have confirmed that these are valid.

This is my first post.
I am sorry if my confirmation was insufficient.
Thank you in advance.

boelle, 2019-10-30:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

yudi-jpn, 2019-10-31:
Thank you, boelle.
Thank you for teaching me the procedure.

"Configuration.h" and "Configuration_adv.h"
I will upload the archive.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3791732/Configuration.zip)

boelle, 2019-11-03:
@yudi-jpn a lot of updates have been added to 2.0.x, is the issue still there?

yudi-jpn, 2019-11-04:
@boelle Thank you.
In fact, some of the other bugs that have continued since 1.9 have not improved.

After leveling with "BL Touch", when trying to set "Temperature" and "Menu items" during printing, the "BL Touch" probe suddenly drops.

Since the stepping motor moves with the probe dropped, magnetic field slip occurs.
The X and Y axes are misaligned, and the support tree and the object being printed are also broken.

Maybe, and drop the probe at the timing at which the sound of the buzzer.
(By definition, the buzzer is commented out.)

I think it is necessary to check the probe status in units of several hundreds of milliseconds and reset the probe according to the read from the sensor.

I confirmed some other bugs, but I may not be able to check the current information.
Thank you politely.

boelle, 2019-11-11:
but the original bug you reported here is that solved?

"#define NOZZLE_TO_PROBE_OFFSET" does not work when using ABL

shitcreek, 2019-11-12:
> After leveling with "BL Touch", when trying to set "Temperature" and "Menu items" during printing, the "BL Touch" probe suddenly drops.
Have you seen this thread https://github.com/MarlinFirmware/Marlin/issues/15159 regarding the issue between BLTouch and the buzzer/speaker?

boelle, 2019-11-12:
@shitcreek so the issue is more hardware related and not marlin?

yudi-jpn, 2019-11-14:
@shitcreek Thank you for guiding me.

I had overlooked this point. [#15159](https://github.com/MarlinFirmware/Marlin/issues/15159)
このうち`#define BEEPER_PIN -1`に関しては、私は未確認です。
> #undef BEEPER_PIN
#define BEEPER_PIN -1

@boelle Thank you.

Currently, ABL does not work as intended in Marlin 2.x in an "Ender-3" environment.
Therefore, we use the unfortunately 1.9.x.

My confirmed bug is still unresolved.
Since there is no time for personal use, I will report again as soon as I can confirm the status of Marlin2.x.

There is a proposal such as the improvement of function.
Also, if you can help me, I will cooperate.

I'm not used to GitHub habits and responses yet.
Carefully me to support, thank you.

yudi-jpn, 2019-11-20:
@boelle
Thank you. Sorry for the late confirmation.

Currently, the following is defined, but the probe descends at the X0 and Y0 positions.
`#define NOZZLE_TO_PROBE_OFFSET {-43, -5, 0}`
`#define Z_SAFE_HOMING`

So far, the first problem has not been solved.
Mere, there is also the possibility of my lack of understanding.

Thank you for your advice in advance.

KaiGrassnick, 2019-11-23:
@yudi-jpn i had the same problem. Changed the Offset after having a different one before.

Solved by reseting the eeprom. (I issued: M502 ) Now the Offset is correct. It seems to be cached / saved to eeprom.

InsanityAutomation, 2019-11-23:
It is now saved to eeprom. You can change it at runtime with M851 XY now so it needs a storage point.

thinkyhead, 2019-11-23:
> Solved by reseting the EEPROM.

Always use `M502`, `M500` after flashing Marlin with new settings to ensure that the new values are saved to EEPROM. Whenever you see a configuration failing to take effect, this is probably the reason.

closed 2019-11-23 by thinkyhead

yudi-jpn, 2019-11-30:
@thinkyhead
Thank you.
My previous problem was solved.

It was confirmed that the following definitions are valid by the modification of "bugfix-2.0.x / 2019-11-18".
`#define NOZZLE_TO_PROBE_OFFSET {-43, -5, -0.4}`

I verified the process without resetting the EEPROM.
Even if the `#define NOZZLE_TO_PROBE_OFFSET` constant is changed, it is retained in the EEPROM correctly.

Thank you carefully.

MardellK, 2020-01-05:
I just finished pulling my hair out on this issue and found that putting in a positive value from sensor to nozzle fixed the problem.

 By my understanding; I have a BL touch directly left of the nozzle which gave me (-35,0,-.5). This made my BL touch drop off the hotbed on the far left in auto cal. The other thing that was explained in line 915 (Nozzle 10,10) confused me. So by trial and error; I just added 10 to my offset measurement.  This gets me (-45, 0,.5). 

I then found out that when changing the -45 to a 45 fixed my issue.  Now I'm not dropping my BL touch off the left side of the hotbed.

Wierd, but I thought you should know.