MarlinFirmware/Marlin #15735: [BUG] [delta] 20x4 LCD unresponsive during prints

eltoozero, 2019-10-30:
### Description

Interaction with LCD is normal until print starts, once print is started, info screen does not update regularly and pressing the knob to access menu causes blank screen until timeout occurs, then slow-update info screen re-appears.

### Steps to Reproduce

1. Enable LCD and SD options:
#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define SDSUPPORT
2. Start print
3. Observe LCD updates
4. Attempt interaction with menu

**Expected behavior:** 

LCD should update normally, menu should be accessible to stop/pause/tune print.

**Actual behavior:** 

Screen does not update regularly, menu is inaccessible.

#### Additional Information (fix?)

Decreasing the DELTA_SEGMENTS_PER_SECOND from 200 to 120 had no effect.
Increasing the BLOCK_BUFFER_SIZE from 16 to 32 made the LCD responsive again.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3791396/configs.zip)

thisiskeithb, 2019-10-30:
> Increasing the BLOCK_BUFFER_SIZE from 16 to 32 made the LCD responsive again.

This seems to be a common fix (or bumping it to 64) for many people, even with 32-bit boards.

karabas2011, 2019-11-01:
Partially works on sbase with 64. My display is responsive but suddenly may become unresponsive and blank,

boelle, 2019-11-08:
@eltoozero i assume this is still an issue?

thisiskeithb, 2019-11-08:
@boelle: https://github.com/MarlinFirmware/Marlin/pull/15837

boelle, 2019-11-08:
will close this one as pr 15837 should have fixed this

closed 2019-11-08 by boelle