MarlinFirmware/Marlin #15737: [FR] unconditional runout triggering

karabas2011, 2019-10-31:
I send gcodes to marlin 2.0 from octoprint.
I send "M118 A1 action:pause(resume)" from octoprint's button to trigger pause manually.
Not instantly but works good.Ok.

Now I want the same with filament runout sensor.
I set filament runout script to "M118 A1 action:pause"
But it does not trigger probably because marlin thinks there is no sd-printing and does not send anything.
M119 reports sersor works.
I tried to hack runout.h but no success.
```
static inline void run() {
      //if (enabled && !filament_ran_out && (IS_SD_PRINTING() || print_job_timer.isRunning() || did_pause_print)) 
      if (enabled && !filament_ran_out )  
      {
```

boelle, 2019-10-31:
sure this is not a bug?

karabas2011, 2019-10-31:
No, I am not sure. I am trying to log anyhow the triggering of sensor, but no any messages on console.
M119 reports all is ok.  Maybe strange config combination. I dowloaded todays zip and copied my  conf/adv/ pins_skr1.3 to this new folder. Still no success.

closed 2020-06-30 by github-actions[bot]