MarlinFirmware/Marlin #15738: [FR] Marlin 2.0 for ANET A2 (Plus) with Bigtreetech SKR v1.3

valfar999, 2019-10-31:
Hi
Does anyone have a functional Marlin 2 configuration on Anet A2? I searched in different forums, I tried different configurations, but none is fully functional, the print freezes, takes various illogical steps outside the printed object, etc. (see video https://youtu.be/ePeeJN2Cy6o).
Can anyone help me with the configuration or just edit and send the entire configuration file / code?
I'm just a beginner.

**Printer:** Anet A2 plus
**Board:** BTT SKR 1.3
**Display:** BTT TFT24 (in touch mode)
**Drivers:** TMC2208 (no UART)
**FW:** Marlin 2

Thanks for help

boelle, 2019-10-31:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/config/examples/Anet/A2plus

i bet that this is a issue with the SKR board, many people have issue with it

Powerprobot, 2019-10-31:
I think this is not a problem with Marlin. I have the SKR1.3 and runs fine.
Your informations are very rare to validate configurations. Steps to check:
- Print over USB -> check the "Raspberry" or Printserver, Check your Cable, If runs from PC -> don't play at the same time.
- Print from Display Slot -> change SD-Card, check connections, make a shield over the cable
- Check your slicer. -> Don't use Cura under version 4.3 or use PrusaSlicer2
-  check for heat problems from the TMC Driver -> add a Fan
- self Build Marlin -> check your configuration

It is significant to activate following lines in configuration_adv.h for TMC driver
```
#define CHOPPER_TIMING CHOPPER_DEFAULT_24V // If your Powerline has 24 Volts. Otherwise enter your Powerline Volts
#define SQUARE_WAVE_STEPPING
#define MINIMUM_STEPPER_POST_DIR_DELAY 20 // 20 : Minimum for TMC2xxx drivers
#define MINIMUM_STEPPER_PRE_DIR_DELAY 20
#define MINIMUM_STEPPER_PULSE 0 // 0 : Smallest possible width the MCU can produce, compatible with TMC2xxx drivers
#define MAXIMUM_STEPPER_RATE 400000 // 400000 : Maximum for TMC2xxx stepper drivers
```

After checking these steps print again....

boelle, 2019-11-03:
@valfar999 problem gone?

valfar999, 2019-11-03:
Probably solved the problem. Most errors were caused by outdated FW in the BTT TFT24 display. After the update, illogical engine steps and various messages disappeared during printing. Stopping still occurred, but there seems to have helped increase the buffer value in configuration_adv. Now only occasionally throws a problem with heating hotend, but it's not regular and after restart everything goes fine.

boelle, 2019-11-11:
> Probably solved the problem.

will close as issue seems gone

closed 2019-11-11 by boelle