MarlinFirmware/Marlin #15740: [BUG] probe crashes into bed after G29 completes

hapklaar, 2019-10-31:
### Description

After running and completing G29, probe goes up a bit, then goes down hard and crashes into bed

### Steps to Reproduce

1. Turn on printer
2. G28 then G29
3. Observe

**Expected behavior:** Expect probe to return home, just like after G33

**Actual behavior:** Probe crashes into bad at last probe point

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3794267/Configuration.zip)
* Provide pictures or links to videos that clearly demonstrate the issue.
https://photos.app.goo.gl/XNsd8faSAvubtPLN6
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2019-11-11:
Is the problem solved if you uncomment `PAUSE_BEFORE_DEPLOY_STOW`?

This option is enabled in the example config for your printer, but disabled in your config. It is needed so that the printer prompts you to remove your probe before performing other moves. I suspect your printer is moving as if you have already removed the probe, causing the collision.

Let me know whether that solves the problem for you. If the problem is not solved I can try it on my printer.

hapklaar, 2019-11-11:
> Is the problem solved if you uncomment `PAUSE_BEFORE_DEPLOY_STOW`?

When I enable this feature, message to stow the probe is given after G29 completes and after removing the probe and pressing the knob to continue, the head travels down. If the probe would have still been there, it would have crashed.

So it seems like the G29 process doesn't return the probe to home after completing as you would expect, but instead lowers the head. This seems incorrect behavior to me.

sjasonsmith, 2019-11-11:
It sounds like it is behaving correctly. Your configuration defines `Z_AFTER_PROBING` as 10, which explains the downward movement.

Commenting this out will probably eliminate the downward move. You could also increase it to a larger value, according to your preference.

You might also try `Z_PROBE_END_SCRIPT`. As an example, you could script a `G28` to occur automatically after the `G29`.

hapklaar, 2019-11-11:
Ah I see, thanks for explaining. What is the practical use for <Z_AFTER_PROBING> after G29 exactly? Can't think of a reason to not G28, like G33 does after it completes.

shitcreek, 2019-11-12:
In the case of a TouchMi probe, the print head needs to move down farther for the magnet to catch and hold the probing rod - which is why `Z_AFTER_PROBING` is required for the use of a TouchMi probe.
I don't think this is a bug as it is doing exactly what it should. As @sjasonsmith has mentioned, you could easily script G28 to occur after G29 - I'd recommend enabling `RESTORE_LEVELING_AFTER_G28` if you are going to do it. As an aside, I hope you are not leveling the bed before each print - it not be necessary.

closed 2019-11-12 by hapklaar