MarlinFirmware/Marlin #15743: [BUG] Crash on M24

GilDev, 2019-10-31:
### Description

My Creality Ender-3 running Marlin 1.1.9 bugfix reboots when I resume a print from SD card.

### Steps to Reproduce

1. Print something
2. M25
3. M24

**Expected behavior:** Resume printing.

**Actual behavior:** Printer reboots.

#### Additional Information

[Configuration files](https://github.com/MarlinFirmware/Marlin/files/3795705/Configuration.zip)

boelle, 2019-10-31:
use 2.0.x instead

GilDev, 2019-11-01:
Yeah I am now, spend an hour migrating my config yesterday, but haven’t tried these steps yet (and will not be able to in the following weeks).

boelle, 2019-11-16:
have you been able to test ?

GilDev, 2019-11-16:
No sorry, still don't have access to my printer for 4 weeks.

boelle, 2019-12-04:
will close this one, we dont work active on 1.1.9 anymore and 2.0 was just released yesterday

closed 2019-12-04 by boelle