MarlinFirmware/Marlin #15744: [BUG] Marlin bugfix 2.0: SKR V1.1 PRO with TMC 2130 Tripple extruder compilation error

phoenixalien, 2019-10-31:
Hello,

I have an issue with compiling Marlin bugfix-2.0.x for SKR v1.1 PRO with TMC 2130, where I use all 3  extruders E0, E1, E2.
The error during compiling that I get is as following:
Marlin\src\module\stepper\trinamic.cpp:47:84: error: conflicting declaration 'TMCMarlin<TMC2130Stepper, 'E', '2', (AxisEnum)5> stepperE2'

I have added config and config_adv files, but I don't think that issue is in them.

Thanks in advance

boelle, 2019-10-31:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

phoenixalien, 2019-11-01:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/3796998/Config.zip)
Zip with configs are added

Thanks

phoenixalien, 2019-11-01:
Found a bug in trinamic.h
// E2 Stepper
#if AXIS_IS_TMC(E2)
  extern TMC_CLASS_E(**1**) stepperE2;
It's the same for all E steppers 2 and higher

closed 2019-11-01 by phoenixalien