MarlinFirmware/Marlin #15745: [BUG] REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER needs help

ruggb, 2019-11-01:
I don't know when it changed, but in build bugfix-2.0.x.9e8021d this controller produces a scrambled screen. It may have something to do with u8glib or arduino since I have updated those recently. When I tried to go back to bugfix-2.0.x.9a515cb it was also scrambled and was not previously.
I found this on the reprap forum, I think. With those three lines added it works.

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_NS(0)   //** scrambled screen without these
#define ST7920_DELAY_2 DELAY_NS(250) //**
#define ST7920_DELAY_3 DELAY_NS(250) //**

lordwolfchild, 2019-11-01:
I have experienced the same problem. I could nail down the problem somewhere between V1.1.8 and V1.1.9.

I have a MKS Gen 1.4 Motherboard and a RepRap Discount full graphic smart controller. I got 1.1.8 and 1.1.9  both downloaded and changed only the MOTHERBOARD to my Baseboard and activated defined REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. With 1.1.8 everythings looks crispy and nice, with 1.1.9  the graphic output shows random shifted lines with each screen update. 

Adding the additional defines ruggb posted above, everything works just great with my config and 1.1.9 upwards.

ruggb, 2019-11-08:
reopened on #15833

thisiskeithb, 2019-11-08:
Providing configs and more information about your specific hardware would be helpful.

ruggb, 2019-11-08:
The bug was confirmed. What does that mean?

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3824653/Marlin.zip)

sjasonsmith, 2019-11-24:
I finally pulled my display out of its box so I could try to reproduce this.

It is working fine for me, with both an SKR 1.3 and an Atmega2560/RAMPS.

thisiskeithb, 2019-11-24:
> It is working fine for me, with both an SKR 1.3 and an Atmega2560/RAMPS.

Same for me on a multitude of boards/configs.

boelle, 2019-12-02:
> The bug was confirmed. What does that mean?
> 
> [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3824653/Marlin.zip)

means that more than 1 person has the problem

boelle, 2019-12-07:
@ruggb since 2.0 was just released a few days ago has this changed this issue at all?

facusenn, 2019-12-10:
I had the same problem, a glitchy screen using Marlin bugfix 2.0 in at least 3 new arduino Mega+reprap full graphic controller, had to solve it using this solution:

Copy this and paste it somewhere in config.h outside of the conditional statements:

**#define ST7920_DELAY_1 DELAY_NS(50)
#define ST7920_DELAY_2 DELAY_NS(188)
#define ST7920_DELAY_3 DELAY_NS(50)**

**If this doesn't work try increasing the numbers inside the brackets.** 
It's a timing issue and delays are in nanoseconds.
This solution its the same as @ruggb proposed but with different numbers.
I don't know why but it seems hardware related, some screens do it and others don't.

[This video represents the issue](https://www.youtube.com/watch?v=yh-PQ7BpZl8)

ruggb, 2019-12-11:
@boelle Here are my findings. Maybe you can do something with it.
Ver 2.0 released 5103cdc
I viewed one signal as representative. 
Noise does not appear to be a factor as someone suggested.
What is a major diff is the signal frequency => no delay is 1.3MHz; with delay it is 850kHz
The pulse width is 2x with the delay. 200ns vs 400ns

X10 probe no delay
[image]
X10 probe no delay
[image]
X10 probe w/delay
[image]
X10 probe w/delay
[image]

tfgcampos, 2020-05-14:
I had the same issue when using Arduino IDE for compiling. Adjusting the delay worked partially, as I didn't find an optimal combination between delay_1, 2 and 3 to give me a sharp screen.

After giving a try on PlartformIO, the issue was solved. Probably related to u8glib used in Arduino IDE.

ruggb, 2020-05-15:
Maybe one of these days I will try PlatformIO but I am still trying to figure out Arduino.
But this totally worked for me since I did it.

boelle, 2020-06-23:
@ruggb 
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

sjasonsmith, 2020-07-04:
I am closing this issue as it is quite old and it isn't clear that any action is needed. Many users use these screens every day, and many timing fixes have gone in since this was originally reported over 8 months ago. A new issue can be opened if specific combinations are found to be problematic.

closed 2020-07-04 by sjasonsmith