MarlinFirmware/Marlin #15749: [BUG] Marlin 2.0 bug fix - LPC1768 case light PWM freqwu

towe96, 2019-11-01:
### Description

When using the "case light" feature on a SKR 1.3 board, the light can't be set to a bearable frequency when not at 0 or 100% duty cycle.

When using it with hardware PWM or fast hardware PWM, the light only flickers shortly when changing the brightness.

When using software PWM, the light works as expected, but has a far too low frequency to be bearable - it's like a strobe light.
Changing the SOFT_PWM_SCALE does not alter the PWM frequency on the case light, but it does affect the parts cooling fan.

### Steps to Reproduce

1. Use a SKR 1.3
2. Enable the case light feature with either hardware or software PWM on the E1 connector

**Expected behavior:**
Software-controllable case light without flickering

**Actual behavior:**
Either no light, or flickering light

boelle, 2019-11-01:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

towe96, 2019-11-01:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/3797541/Configuration.h.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/3797542/Configuration_adv.txt)

boelle, 2019-11-08:
@towe96 still an issue i assume?

towe96, 2019-11-11:
Still an issue. I'll rebase the firmware today and check if there's been any changes.

boelle, 2019-11-21:
@towe96 was there any changes?

DenisSemenov1970, 2019-11-22:
I have a similar problem. In the latest firmware there is no reaction to SOFT_PWM_SCALE. I usually use SOFT_PWM_SCALE 2. Now any value of SOFT_PWM_SCALE is perceived as 0. A terrible squeak from the fans. Board BigTree_SKR_Pro.  Firmware fresh.

boelle, 2019-11-24:
@towe96 have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-12-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-03 by boelle