MarlinFirmware/Marlin #15750: [BUG] AVR2560 home with interrupt has been broken

GMagician, 2019-11-01:
There is something in recent PR that broke homing with interrupt with AVR.
I have endstops on X and Y (standard inputs) and Y fails...
M119 report correctly triggered but Y move to negative and never stops...

I went back to c80eda073 and there it works. I strongly suppose problem is in 35b1149d9 (Patch es int tests for Arduino 1.8.10)

I'll investigate....

GMagician, 2019-11-01:
```c++
if (moreDigitalPinToPCMSK(pin) != nullptr) {
```
whill never return `nullptr`

GMagician, 2019-11-01:
I think I have to read mega2560 datasheet to understand changes...

GMagician, 2019-11-01:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3797357/Configuration.zip)

GMagician, 2019-11-01:
Maybe better to reopen until merge has been done....

minorminer59, 2019-11-02:
I can confirm this behavior. When homing the bed continues to hit Y at the Min.

GMagician, 2019-11-02:
@minorminer59 you can use my PR to solve it, waiting it to be merged

PR #15758 (2019-11-02): Fix mega interrupt handling on PJ
PJ pins are D14 and D15 not D13 and D14, so all new logic has to be changed.

This fix #15750
closed 2019-11-02 by thinkyhead