MarlinFirmware/Marlin #15751: [BUG] Sbase: No communication via hardware serial 2.

karabas2011, 2019-11-01:
Last marlin 2.0 on MKS Sbase.1
hardware serial Aux1 connected to raspberry and proved to work on smoothie firmware
Problem: 
No communication at any speed. No garbage on screen, no anything.
Configiration.h:

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 250000

boelle, 2019-11-01:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Bergerac56, 2019-11-01:
Yes, it will be easier with configuration files ;)
I have a MKS TFT32 connected to PORT_2 and it works without any problem with the last bugfix release at least.

karabas2011, 2019-11-01:
Do you set pins for AUX1 or comment them out in SBASE_PINS.h ?

Bergerac56, 2019-11-01:
No I use the pins definitions as they are.

Bergerac56, 2019-11-01:
Please post your config files and your pins.h file so I can compare them with mine

karabas2011, 2019-11-01:
I found my problem, last hardware changes? wrong wireing

closed 2019-11-01 by karabas2011