MarlinFirmware/Marlin #15753: [BUG] ABL points not accurate when dealing with large bed size.

paulpenney, 2019-11-01:
### Description

Hello everyone - I am submitting this as a bug because I truly do believe it is a marlin-specific bug. At our shop, we are currently dealing with a number of large-bed printers. We have the following printers we are working with: 

1) Modix Big 120X (600 x 1200 MM bed size) UBL (Aluminum bed with PEI)
2) Modix Big 60 (600 x 600 MM bed size) UBL (Aluminum bed with PEI)
3) Creatbot D600 Pro (600 x 600 MM bed size) Bilinear ABL (Glass Bed)

They all have a BL touch V2.2 or higher. All these printer seem to share the same issue - when the printer are either calibrated or levelled using G29 commands, the ABL system doesn't seem to be able to keep up with the extreme variations that a large bed can produce. After producing either a UBL mesh or an BILIN-ABL mesh, doing G26 and full bed validation tests reveals that the nozzle offset has significant variations (sometimes out by 0.3-0.4 when adjusting manually with a G29 P4). Some places are very squished, while others are printing almost in the air, and aren't squished enough to even adhere.

The Modix printers seem to print low in the centre of the beds (very squished) and high near and  around the perimeters. They are calibrated using a 10 x 10 grid (Big 60) or a 10 x 15 grid (Big 120) - The points certainly seem dense enough, and the behaviour isn't that the points are perfect at the probe point and then break out from there - the probe points themselves at the point are high and low. I have personally looked at the source code for these (Marlin 1.1.9, modern release) and have determined that the probe - nozzle offset is correct. The Z offset is set correctly, and the UBL meshes are being properly stored, recalled, and "smart filled". I want to say specifically that this problem isn't just something that is present in smart filled regions, and it occurs in areas where the bed is being actually physically probed by the BL touch. I have tried compiling the latest bug fix 2.0 and seeing if that solves the issue, but it doesn't seem to (created a new mesh after updating software, cleared EEProm properly, etc) 

The Creatbot D600 is more "bowl" shaped and dips in the centre physically when measured with a straight ruler. With this printer, using bilinear ABL, the print is very squished around the edges and "Normal" in the centre - basically the trend is that there doesn't seem to be enough adjustment in the Z axis to keep up with the slope of the bed curve. This printer is running 1.1.9 as well. 

Both printers we have confirmed with calibration and small-footprint prints do not have any extrusion issues. They also do not have any filament binding issues. Creatbot is 2.85mm and Modix printer are 1.75mm. 

### Steps to Reproduce

1) All printers heated to 60 deg on bed with time for the heat to settle in and stabilize. All printers are four corner levelled three times to ensure all lead screws are even. All printers are checked for squareness, loose belts, loose parts, mechanical fit and finish - and they all seem to have very sturdy designs, especially the D600 all-metal one piece enclosure. All nozzles heated to 200 deg with filament loaded for testing. BL touches are confirmed steady, secure with no movement. EEPROM memory has been cleared properly and ready for new calibration.

2) Calibrate. For modix printers - G29 P1 to make a UBL grid, G29 P3 to fill the perimeter, G29 S1 to save to slot 1, G29 L1 to load the grid, G29 T to verify, G29 A to turn on UBL, M500 to save settings. M503 to confirm settings stick. Check again after power cycle. No EEPROM issues. Start Gcode has G29 A, G29 L1, G29 J (have tried with and without J to see if it makes a difference - none made) 

For D600 - G29 after G28 in start Gcode. Confirm using pronterface that the grid is being created (it spits it out in the feed). 

Both printers have fade height 0.

**Expected behavior:** [What you expect to happen]

During printing, the probe-Z offset of the nozzle will keep up with the contours of the bed and produce a "flat" first layer. 

**Actual behavior:** [What actually happens]

During printing, the probe-Z offset is variable and seems to either over or under compensate in certain areas of the bed (I am leaning toward under-compensate based on the behaviour of the D600). Feeling the Z lead screws, they are turning gradually during X-Y movement so I know that some corrections are being made. If the print has a small footprint, it usually works fine as we can rig the Z offset for that area of the bed - in this way, we have determined that the Z step calibration is correct from calibration pieces, and in general the printers are printing nominally - only when we try and print objects that span a large area of the bed in the first layer do we run into these issues, or multiple objects throughout the whole bed.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[Modix Config.zip](https://github.com/MarlinFirmware/Marlin/files/3798082/Modix.Config.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.

Images of the Modix Print, front of the bed (Squished) and back of the bed (Almost air printing)

[image]
[image]
[image]

Images of Creatbot border of print near edges (very squished) and near centre (looks good). 

[image]
[image]

boelle, 2019-11-08:
@paulpenney i assume this is still an issue?

paulpenney, 2019-11-08:
Yes indeed. As of now we don’t have a solution outside of using G29 P4 to manually adjust the points - which on the larger 1200 x 600mm printer literally takes hours. Thanks for checking!

boelle, 2019-11-20:
@Roxy-3D do you know what could be done here?

paulpenney, 2019-11-20:
It is interesting - I know this has been brought up before in other threads and it seems to be a recurring theme. Just on my background, I've assembled probably 10 printers from scratch, have a fairly deep understanding of Marlin and have worked with many of these large printers with ABL, and all of them seem to have the same issue. It seems like the larger bed size and large range of variance between points seems to cause the issues.

boelle, 2019-11-20:
without knowing i think the biggest problem might be if the variance is to big

the bed size as such would only spread the probe points on a larger area, but then again that is just my simple thinking

boelle, 2019-11-20:
just curious, how much is the variance between the highest point and the lowest one?

paulpenney, 2019-11-22:
> just curious, how much is the variance between the highest point and the lowest one?

I believe about 1.5-2mm

Vertabreak, 2019-11-22:
Unsure if its relevant but i recently discovered my probe was running at 3V and it was killing accuracy. checking my board and probe supported 5V then i force set 5V and now the probe works correctly.

boelle, 2019-11-24:
@paulpenney checked if voltage to probe makes a difference?

paulpenney, 2019-11-24:
Hey y’all, 

Probe accuracy is not an issue - repeatability tests confirmed a high degree of accuracy, and this is happening in both the new 3.1 BL touch in 5V mode and the old 2.2 hardwired to 5v on different boards as well.

yet-another-average-joe, 2019-11-25:
I have such a (smaller than yours) problem on a much smaller printer (Tevo Tornado + mechanical dual Z, and open loop). I also have lots of metrology instruments.

First, I mesured the BLTouch repeatability, using the printer : a few micons, fine. Then I measured the printer repeatibility using a DTI in place of the hotend (0.002 mm = 2 microns resolution, +-2.5 mm range). Repeatability is OK : +- 0.002mm to 0.004mm (+-2 to 4 microns). Fine. This means that Z-axis and BLTouch have a great repeatability.

But repeatability is NOT accuracy !

Using another DTI (0.01mm = 10 microns resolution, 50mm range), I measured the kinematics itself, over 50mm. System repeatability is just fine. But accuracy is just terrible ! +- 0.1mm over 1mm at worst !. The two leadscews were tested with the same results. They are just crap.

All instruments are either Swiss made (Tesa, Roch) or Japan made (Mitutoyo). No crappy chinese capacitive calipers or whatever for these measurements.

The BLTouch does NOT probe at the exact nozzle height ! The BLTouch is just part of the probing system, and the result cannot be better than the leadescews accuracy. As is : +-0.005mm (repeatability) +- 0.1mm (kinematics accuracy) ; the BLTouch + kinematics + ABL software CANNOT be trusted for better than +-0.105mm (even if results are slightly better : I'd say I get +- 0.05mm at bed leveling, this is the order of magnitude I have to compensate everytime with babystepping). I gave up measuring the brim thickness with a (swiss) micrometer and changing Z offsets and so on. Not worth the effort.

6 month ago, I shot some videos that I never edited and uploaded to Youtube. Maybe I should... But it is time consuming, and I'm out of time. It is self explanative about the repeatability vs accuracy on a consumer 3D printer.

It's probably even worse if belts/pulleys are used for the Z-axis. I hope your printers have machined/grinded leadscrews, or precision grinded ball screews (those used on machine tools). If your leadscrews are chinese ordinary ones, I fear you will never get perfect results with a BLTouch.

Workarounds :

- babystepping
- manual grid for bed leveling
- piezo probing

When I level the bed (before each and every print) , I always preheat the bed and the hotend for half an hour, and I avoid air movements around the printer, at least at the begining. And I always have to babystep (more or less).

Problems are probably much more difficult to solve with so large machines.

boelle, 2019-11-25:
@yet-another-average-joe so to sum if up, if you use shitty crap china stuff the printer will perform likewise

if you pay a bit more to get quality precission parts you get likewise printer?

yet-another-average-joe, 2019-11-25:
I wouldn't say that ! Many makers get great prints even with the cheapiest crappiest machines. But the first layer is not always perfect. And after calibration, if an accurate print is needed, print two times : first prototype -> measuring -> compensating -> final part. And this way +-0.1mm is easy to get. It's incredible ! On my side I nearly stopped machining on my Bridgeport and my lathes for most of my makes. I got the printer for part prototyping before machining them, for a small lathe CNC retrofit project : since then (2 years), the (el cheapo, chinese, rolled...) ballscrews, steppers and drivers are still on a shelf, and the lathe is still apart !
 
Cheap crappy 3D printers are incredible. So light and so cheap, so great softwares, and so great results with so tiny machines. Materials are cheap, tooling is dirt cheap (nozzles) !

For sure even chinese rolled ballscrews would be better ; but would require a rigid structure and so on to take advantage of the better specs. Unfortunately, it is not a bit more to pay. It is a lot more :-( And they take a lot of room, this means the printer has to be built around the ballscrews and linear rails.

Leadscrews I have are +-0.1mm. It is crap, but it is within specs for consumer 3D printing. I think nobody except professionals with $$$$$$ machines need or expect better.

I just wanted the OP to consider the real accuracy, and think of the machine before thinking of the software or the probe. He's needs are probably very different from the average joe. But physics are the same for everyone, meaning that a 10 times larger 3D machine will have 10^3 = 1000x larger problems to solve.

Maybe the OP problem could be solved with piezo probing. This is a technology I plan to experiment with : the nozzle being the probe (the transmitter !), the repeatability being what it is (near to perfect), the kinematics inaccuracies between the probe height and the nozzle would be zero.

boelle, 2019-11-25:
yep, the frame needs to be good, and the lead screw or ball screw to some extent dictate the frame

but yes the old saying: you get what you pay for still comes in to my mind

if i want a good machine i need to pay a lot more

yet-another-average-joe, 2019-11-25:
Even cheap machines are probably good. Look at the well known youtubers who print for a living. The best they have are often Prusa's MK3. I'm a beginer, and I just slightly modded my printer ; the only real problems come from the cheapest filament I can find (tiny over/under extrusion due to diameter inconsistencies), and some ringing (the frame, as you wrote !). It's only aesthetics. Just finished a print. One hole had to be 25mm. It is within -0.05mm + 0.04 mm. And everybody on earth gets this with even the cheapest printer, out of the box.

I just measured leadscrews on my (european) machines : lathe = 35mm (2 tons), milling machine (X an Y only) = 28mm (1.3 ton). getting +-0.02 is good, +-0.01 needs pratice (professional skills). A cheap chinese printer does nearly the same ! This makes me think of a famous grandpa youtuber (myfordboy), a great engineer/machinist/founder : his channel is now about 3D printing !

I think that we get a lot more that what we pay for. Thanks to open source and open hardware, and thanks to chinese people.

It's been the same with paper printers. First one I had was a noisy anchor boat, using perf paper. Last one is a laser color printer, half the price, and spitting and scanning double sided sheets with perfect graphics a light speed (for pro usage). In between, a €50 inkjet does a far better job than my first one.

(I realize that we are 200% off topic !)

paulpenney, 2019-11-25:
Bringing things back on topic: The Modix has four ball screws (one on each corner of the bed and each with its own motor), and it is built like a tank - using all genuine parts - Hiwin linear rails, e3D hotend, all metal frame parts and 4040 extrusions, etc. Again, the probe accuracy isn't the issue here- what we are seeing is a uniform distortion of the grid itself over large surfaces, not random points that are inaccurate here and there - the mesh gets further away from the bed the further up the Y axis the head travels.

boelle, 2019-11-25:
but what about on the X axis?

@Roxy-3D do you have a clue what might be going on there? would more probe point on each axis help? i know there currently  is a limit but i thought that since this might be a larger bed than what would be "normal" it might need more probe points

AnHardt, 2019-11-25:
When the error is systematic it could be a variation  of [this](https://github.com/AnHardt/Marlin/issues/84) mechanical problem.
This kind of problems can't be compensated in Marlin.

paulpenney, 2019-11-25:
The X axis is better but not perfect. Currently the Modix probes 150 points, and those points are about 5-6 cm from eachother – not too bad in terms of granularity.   Sent from Mail for Windows 10 From: Bo HerrmannsenSent: November 25, 2019 2:06 PMTo: MarlinFirmware/MarlinCc: paulpenney; MentionSubject: Re: [MarlinFirmware/Marlin] [BUG] ABL points not accurate when dealing with large bed size. (#15753) but what about on the X axis?@Roxy-3D do you have a clue what might be going on there? would more probe point on each axis help? i know there currently is a limit but i thought that since this might be a larger bed than what would be "normal" it might need more probe points

paulpenney, 2019-11-25:
> When the error is systematic it could be a variation of [this](https://github.com/AnHardt/Marlin/issues/84) mechanical problem.
> This kind of problems can't be compensated in Marlin.

Is that if the actual head of the printer were to be skewed? Or you are saying that perhaps the rails on either side of the printer may be twisted? The thing seems straight as an arrow - how would we test for that besides visual confirmation?

boelle, 2019-11-25:
hmm... the poorman check would be to use a caliper and check if each rail are straight on the frame

then check if the end of each rail is the same distance apart

AnHardt, 2019-11-25:
At both sides go down until the probe triggers. Measure/compare the gap between nozzle and bed. If the gap is not equal the software has no chance to compensate.

boelle, 2019-11-25:
ahh yes, sorry for my mis-guidance

its the the rails that are not keep flat or they are twisted right?

yet-another-average-joe, 2019-11-25:
The only way to know if there is a software error :

Make a DTI fixture so the DTI will be aligned with the BLTouch. Probe with the BLTouch, keep the results. Move the Z axis to Z = zero. Remove the hotend, and mount the DTI fixture with the DTI. Probe the grid at the exact points where Marlin probes the bed. Write down the DTI readings. Compare with the firmware cartography.

(The DTI applies little pressure on the bed, and will slighly move it down - it also slighly depend on the position, center or corners, the springs forces resultant being different. On my printer, I measured up to 0.03mm using a second - 2 microns, lever - DTI while probing with the first one. It is nearly negligible)

[image]

boelle, 2019-12-01:
@paulpenney have you had time to test with a dialgauge like @yet-another-average-joe suggested?

YAost, 2019-12-01:
I'm surprised no one mentioned the "Fade height" feature. I had all sorts of problems when it was set too low (about 1-2 mm), such as delamination between first and second layer, not even width of printed lines on first layer. With value of 10 most of the problems were solved.
Though, on my setup (extended Tevo Tornado with buildarea of 300x600) I have a strong feeling that it still isn't working properly. With all the levelings done (same setup as @paulpenney mentioned at the beginning, besides fade height), there is more squish at the edges (printbed has drop at the middle). 

As an alternative for dialgauge testing method, you can also use paper as a probe. As far as I know, when sending commands manually (or moving axis via LCD), leveling should be working, so difference in nozzle pressure against paper will be noticable.

boelle, 2019-12-07:
@paulpenney i guess that the release of 2.0 (not 2.0.x) did not change this?

YAost, 2019-12-14:
After two weeks of tweaking I can finally conclude that system is working as it should on my large setup (300x600). 
Details about it:

- MKS Gen L, Marlin 2.0
- Tevo Tornado, X axis extended (still V-slot)
- Double Z axis with independent motors without belt sync
- Huge warping aluminum bed with glass on it
- BFPTouch variation to fit custom carriage design (basically: optical endstop with spring loaded probe, servo)

Problems that led ABL to fail in some regions of bed:

- Double Z not in sync (what happens quite frequently), as a result: not leveled X-axis
- Spring loaded probe tilted carriage creating unaccuracy
- Backlashes in overweighted X-carriage
- Z-axis flexible couplers messed with double Z leveling

What I've done to prevent such issues:

- Simply lowering Z speed when probing made great impact on original setup (with all backlashes included).
- Extra probing (fast and slow)
- Removing spring from BFPTouch improved accuracy (according to repeatability test), removed tilt in carriage
- All sorts of fixing with carriag, reducing backlash 
- Removing flexible couplers
- Double check z axis leveling before probing and printing

The only problem left is probe area. Probe offset from nozzle is { 25, -22 }, when doing G29 P1 X0 Y0 it starts to probe approx. in (30, 32), when it clearly can at (0, 22). The same goes for X300 Y600, it doesn't go past Y250, despite any changes in MESH_MIN_ or MESH_MAX_ and other. But I guess, it should be discussed in separate thread.

boelle, 2019-12-24:
@paulpenney is the issue still the same with all the updates in the last 29 days?

paulpenney, 2019-12-24:
Sorry for being missing in action everyone, unfortunately I just don’t have the time here to finish the dial-gauge method of checking the bed - but the issue persists on two identical large bed machines built by two different people. What we have discovered is that using a duet and its firmware actually solves the issue. Is there something that duet does in its firmware that marlin isn’t that makes it more robust?  

Paul

boelle, 2020-01-07:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-07 by boelle