MarlinFirmware/Marlin #15754: [BUG] M261 not worked on SKR 1.3

aleknest, 2019-11-01:
I use Marlin 2.0 "bugfix"
M261 g-code not return answer. As i see problem in Wire library, RequestFrom method
M261 on Mega2560 worked fine.

boelle, 2019-11-03:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

aleknest, 2019-11-03:
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/3801721/conf.zip)

ellensp, 2019-11-11:
What pins are you trying to use as i2c?   
As far as I can see you need to use  2 of these pins, but they are already used for other things

P0.0  SDA1 is used for E1DIR
P0.1  SCL1 is used for E1STP
P0.10 SDA2 is used for E1EN
P0.11 SCL2 is used for E0DIR
P0.19 SDA1 is used for YSTP
P0.20 SCL1 is used for YDIR
P0.27 SDA0 is used for SD card Data
P0.28 SCL0 is used on pin 9 of EXP1 (ie LCD connector)

aleknest, 2019-11-11:
> What pins are you trying to use as i2c?
> As far as I can see you need to use 2 of these pins, but they are already used for other things

P0.0 SDA1 is used for E1DIR
P0.1 SCL1 is used for E1STP
I turn off this pins for E1 in pins file.
i2c worked for transmit, but don't worked for receive.

BTW i moved to software UART for my device.

boelle, 2019-11-24:
@aleknest have you tried to ask for help any of these places?

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2019-12-03:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-12-03 by boelle

robin-gdwl, 2020-04-23:
Having the same issue.