MarlinFirmware/Marlin #15755: TouchMi - probing setting stop after 6 probing point

kramer04, 2019-11-01:
### Description
I've an Ender3 with a SKR-MINI-E3 TMC2209 and a TouchMi

when i launch a print, probing point stop at 6 and not at 36

### Steps to Reproduce
Launch a print or a level bed

**Expected behavior:** 
I expect 36 probing points

**Actual behavior:** [What actually happens]
probing only 6 points

#### Additional Information
I did not have this issue with the previous version of Marlin bugfix2.0.x
[config.zip](https://github.com/MarlinFirmware/Marlin/files/3798719/config.zip)

DiegoGiovany, 2019-11-01:
Same here, with ir probe and bilinear auto bed level. Just the first 6 points are probed

tharr916, 2019-11-01:
Very similar results with bltouch and  a 7x7 grid - the first 14 points are probed

thisiskeithb, 2019-11-01:
> Very similar results with bltouch and a 7x7 grid - the first 14 points are probed

Same issue here.

thisiskeithb, 2019-11-01:
This has been fixed with ad28da7eb4fe720776daf929d0a25ba39e6b6ca8.

boelle, 2019-11-03:
closing as its been fixed with https://github.com/MarlinFirmware/Marlin/commit/ad28da7eb4fe720776daf929d0a25ba39e6b6ca8

closed 2019-11-03 by boelle

gannon145, 2019-11-20:
ad28da7 did not fix this for me. I still only get 6 of 9 probe points.

dmonty2, 2019-11-24:
Same here.  My source code has the patch and only 6 first points are probed, front and middle.  The last 3 rear points are not done.