MarlinFirmware/Marlin #15756: Power loss recovery error using mesh bed leveling as a calibration method

camilorodas, 2019-11-01:
Good morning to the whole community, I find an error in the bugfix version 2.0, when used as an MBL calibration method, when activating the PLR ​​system, the printer never returns to the calibration or offset it had at each point of Z , so whenever it comes back it starts 2mm above or in the worst case even more, I have found the solution to this problem in line 375 and 376 of the library power_loss_recovery.cpp

    gcode.process_subcommands_now_P (PSTR ("G1 Z0 F200"));
    sprintf_P (cmd, PSTR ("G92.9 Z% s"), str_1);

The error can be corrected by adding the line:
 gcode.process_subcommands_now_P (PSTR ("M211 S0"));
which turns off the limit switches at the time of making G28 and that marlin allows movements below Z0, if now it is written:
gcode.process_subcommands_now_P (PSTR ("M211 S0"));
 gcode.process_subcommands_now_P (PSTR ("G1 Z-2 F200"));
 sprintf_P (cmd, PSTR ("G92.9 Z% s"), str_1);

the printer now when retaking the position will lower 2 mm that it needed, as this is a bad thing to be compiling by manual error, I wanted to add an editable bar in the bed level menu but when adding it gives me this error on console:
Error: EEPROM datasize error.
echo: Index: 544 Size: 620
 loading the EEPROM memory I want my variable called z_power_loss_offset to be saved in the EEPROM, I created this variable in the libraries:
g29.cpp // mbl
mesh_bed_leveling.h
mesh_bed_leveling.cpp
configuration_store.cpp

I thank everyone who can help me, God bless you, greetings. I add my modified marlin code for my 3D printers.

[Marlin ROAL uLTIMA VERSION 2019 2.zip](https://github.com/MarlinFirmware/Marlin/files/3798892/Marlin.ROAL.uLTIMA.VERSION.2019.2.zip)

boelle, 2019-11-11:
where do you download marlin from? marlin does not have any ultima version

boelle, 2019-11-21:
will close this one as its not 100% certain that we deal with marlin downloaded from
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

the zip attached suggest that its an ultimate version and marlin does not have that

closed 2019-11-21 by boelle