MarlinFirmware/Marlin #15759: [BUG] 2.0.x Bilinear ABL probes only part of the bed

mat2k, 2019-11-01:
Bilinear bed leveling seems to be broken in the latest (downloaded oct 31) Marlin bugfix 2.0.x build. In my setup, it probes some points (for a 5x5 grid, the first 10 points; for a 4x4 grid, only the first 4 points), then finishes normally and executes the z probe end script. (Bed Leveling is set to Off after the G29).
M111 S32 shows this:
`>>> G29
SENDING:G29
  current_position= X0.00 Y45.00 Z5.00 : >>> G29
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X30.00 Y-15.00 Z-1.20 (Right-Front & Below Nozzle)
Auto Bed Leveling: BILINEAR
 (disabled)
  current_position= X0.00 Y45.00 Z5.00 : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to X0.00 Y45.00 Z11.20
<<< do_blocking_move_to
>>> do_blocking_move_to X0.00 Y45.00 Z11.20
<<< do_blocking_move_to
>>> probe_at_point(198.00, 0.00, raise, 0, probe_relative)
  current_position= X0.00 Y45.00 Z11.20 :
>>> do_blocking_move_to X168.00 Y15.00 Z11.20
<<< do_blocking_move_to
  current_position= X168.00 Y15.00 Z11.20 : set_probe_deployed
deploy: 1
  current_position= X168.00 Y15.00 Z11.20 : >>> run_z_probe
  current_position= X168.00 Y15.00 Z11.20 : >>> do_probe_move
>>> do_blocking_move_to X168.00 Y15.00 Z-1.80
<<< do_blocking_move_to
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
  current_position= X168.00 Y15.00 Z1.15 : sync_plan_position
  current_position= X168.00 Y15.00 Z1.15 : <<< do_probe_move
  current_position= X168.00 Y15.00 Z1.15 : <<< run_z_probe
>>> do_blocking_move_to X168.00 Y15.00 Z2.15
<<< do_blocking_move_to
<<< probe_at_point
>>> probe_at_point(135.00, 0.00, raise, 0, probe_relative)
  current_position= X168.00 Y15.00 Z2.15 :
>>> do_blocking_move_to X105.00 Y15.00 Z2.15
<<< do_blocking_move_to
  current_position= X105.00 Y15.00 Z2.15 : set_probe_deployed
deploy: 1
  current_position= X105.00 Y15.00 Z2.15 : >>> run_z_probe
  current_position= X105.00 Y15.00 Z2.15 : >>> do_probe_move
>>> do_blocking_move_to X105.00 Y15.00 Z-1.80
<<< do_blocking_move_to
echo:busy: processing
  current_position= X105.00 Y15.00 Z1.19 : sync_plan_position
  current_position= X105.00 Y15.00 Z1.19 : <<< do_probe_move
  current_position= X105.00 Y15.00 Z1.19 : <<< run_z_probe
>>> do_blocking_move_to X105.00 Y15.00 Z2.20
<<< do_blocking_move_to
<<< probe_at_point
>>> probe_at_point(72.00, 0.00, raise, 0, probe_relative)
  current_position= X105.00 Y15.00 Z2.20 :
>>> do_blocking_move_to X42.00 Y15.00 Z2.20
<<< do_blocking_move_to
  current_position= X42.00 Y15.00 Z2.20 : set_probe_deployed
deploy: 1
  current_position= X42.00 Y15.00 Z2.20 : >>> run_z_probe
  current_position= X42.00 Y15.00 Z2.20 : >>> do_probe_move
>>> do_blocking_move_to X42.00 Y15.00 Z-1.80
<<< do_blocking_move_to
  current_position= X42.00 Y15.00 Z1.20 : sync_plan_position
  current_position= X42.00 Y15.00 Z1.20 : <<< do_probe_move
  current_position= X42.00 Y15.00 Z1.20 : <<< run_z_probe
>>> do_blocking_move_to X42.00 Y15.00 Z2.20
<<< do_blocking_move_to
echo:busy: processing
<<< probe_at_point
>>> probe_at_point(9.00, 0.00, raise, 0, probe_relative)
  current_position= X42.00 Y15.00 Z2.20 :
>>> do_blocking_move_to X-21.00 Y15.00 Z2.20
<<< do_blocking_move_to
  current_position= X-21.00 Y15.00 Z2.20 : set_probe_deployed
deploy: 1
  current_position= X-21.00 Y15.00 Z2.20 : >>> run_z_probe
  current_position= X-21.00 Y15.00 Z2.20 : >>> do_probe_move
>>> do_blocking_move_to X-21.00 Y15.00 Z-1.80
<<< do_blocking_move_to
  current_position= X-21.00 Y15.00 Z1.17 : sync_plan_position
  current_position= X-21.00 Y15.00 Z1.17 : <<< do_probe_move
  current_position= X-21.00 Y15.00 Z1.17 : <<< run_z_probe
>>> do_blocking_move_to X-21.00 Y15.00 Z2.17
<<< do_blocking_move_to
<<< probe_at_point
>>> probe_at_point(16074.00, 0.00, raise, 0, probe_relative)
  current_position= X-21.00 Y15.00 Z2.17 :
  current_position= X-21.00 Y15.00 Z2.17 : set_probe_deployed
deploy: 0
>>> do_blocking_move_to X-21.00 Y15.00 Z2.17
<<< do_blocking_move_to
  current_position= X-21.00 Y15.00 Z2.17 : > probing complete
<<< G29
>>> do_blocking_move_to X-21.00 Y15.00 Z5.00
<<< do_blocking_move_to
echo:busy: processing
Z Probe End Script: G1 X0 Y0 F10000
`

I'm using a Mega 2560 with RAMPS 1.4, this worked perfectly on a previous build of 2.0.x

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3799143/Marlin.zip)

mat2k, 2019-11-01:
UBL also seems to do this, the 5x5 grid doesn't finish (probes 20 points and stops)

Masanetz, 2019-11-01:
Rolling back the changes of commit e3ddf6e made it work again for me...

thisiskeithb, 2019-11-01:
> Rolling back the changes of commit e3ddf6e made it work again for me...

Same here.

thisiskeithb, 2019-11-01:
This has been fixed with ad28da7eb4fe720776daf929d0a25ba39e6b6ca8.

boelle, 2019-11-03:
closing this as it seems to be fixed with https://github.com/MarlinFirmware/Marlin/commit/ad28da7eb4fe720776daf929d0a25ba39e6b6ca8

closed 2019-11-03 by boelle