MarlinFirmware/Marlin #15761: [BUG] TMC2130 LIN_ADVANCE not turning

psavva, 2019-11-01:
Extruder doesn't move at all when using LIN_ADVANCE and TMC2130 driver when MINIMUM_STEPPER_PULSE is so to 0 as per the recommendation in configuration_adv.

If I set MINIMUM_STEPPER_PULSE to 2 with LIN_ADVANCE enabled, extruder works as it should.

If I set MINIMUM_STEPPER_PULSE to 0 with LIN_ADVANCE undefined, Extruder works as it should.

If I set MINIMUM_STEPPER_PULSE to 0 with LIN_ADVANCE to 0 or any other value , Extruder does not work.

I can provide my config files, but won't help much since you need to change values of LIN_ADVANCE and MINIMUM_STEPPER_PULSE to simulate using TMC2130 driver

psavva, 2019-11-03:
@Sebastianv650 Any ideas on this one?

Gerggg, 2019-11-04:
You can try the #define SQUARE_WAVE_STEPPING in configuration_adv.h

psavva, 2019-11-04:
I'm not after a workaround, we need to understand the root cause of the issue and fix it really.

The workaothat works is setting the minimum step pulse to 2 for tmc2130 drivers.

@tcm0116 @teemuatlut  maybe you guys have an idea what the issue might be?

ManuelMcLure, 2019-11-04:
We know exactly what the issue is - the code path for ``LIN_ADVANCE`` can result in step pulses that are too short for TMC drivers with the default setting of ``MINIMUM_PULSE_WIDTH``. Setting ``SQUARE_WAVE_SETTING`` or increasing the ``MINIMUM_PULSE_WIDTH``  so the pulses are long enough _is_ the correct solution. It can be argued that this should happen automatically if the extruder has a TMC driver.

psavva, 2019-11-04:
In that case, looks like we need a conditional or sanity check :)

AnHardt, 2019-11-04:
@ManuelMcLure 

That's only half of the truth. Either:
the TMC have an other internal clock frequency than we expect, or
we misinterpret the datasheets at an other place, or
the datasheets are wrong, or
Marlins clock to measure the pulse length is wrong, or
due to prefetch, or some other speed up technology, the time to execute the code without delays is much shorter than we expect.

So far the theory.
Additionally the STEP connection between CPU and TMC could have enough capacity/inductivity to prevent the signal from rising fast enough to reach the needed voltage for long enough. For the ST7920 based displays we have similar problems.

What ever delay we set as default might be unnecessarily long for "well behaving" systems and may waste time.

Maybe one day we will know the real reason.

In the meantime we could set up a bot here. The keywords ("LA"|"linear advance")&"TMC"&"Extruder"&("does not move"|"loosing steps") could trigger a "Increase delays or use SQUARE_WAV" answer.  ;-)  Or we could set the delays and hope nobody tries to recalculate them.

ManuelMcLure, 2019-11-05:
The problem is that for TMC we by default set ``MINIMUM_STEPPER_PULSE`` to 0 which means "the shortest pulse we can generate" - the resulting pulse is much shorter with ``LIN_ADVANCE`` than without it.
The proper solution is to allow specifying the minimum pulse length in nanoseconds instead of microseconds so we can specify a pulse length between 0 and 1us - that would allow us to specify a correct value for TMC drivers that would be honored whether ``LIN_ADVANCE`` is set or not.
Of course, this only applies to TMC standalone drivers which can't be run in ``SQUARE_WAVE_STEPPING`` mode - if the driver supports that option it's much better to use that instead.

teemuatlut, 2019-11-05:
https://github.com/MarlinFirmware/Marlin/pull/14195

psavva, 2019-11-11:
PR merged

closed 2019-11-11 by psavva