MarlinFirmware/Marlin #15763: [BUG] Compilation Error (2.x)

psavva, 2019-11-02:
I think this may have broken LPC1768

https://github.com/MarlinFirmware/Marlin/pull/15762

In file included from Marlin\src\HAL\HAL_LPC1768\HAL.cpp:26:
Marlin\src\HAL\HAL_LPC1768\../shared/Delay.h: In function 'void enableCycleCounter()':
Marlin\src\HAL\HAL_LPC1768\../shared/Delay.h:49:7: error: 'DWT' was not declared in this scope
       DWT->CYCCNT = 0;
       ^~~
Marlin\src\HAL\HAL_LPC1768\../shared/Delay.h:50:20: error: 'DWT_CTRL_CYCCNTENA_Msk' was not declared in this scope
       DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;
                    ^~~~~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\HAL_LPC1768\../shared/Delay.h:50:20: note: suggested alternative: 'MPU_CTRL_HFNMIENA_Msk'
       DWT->CTRL |= DWT_CTRL_CYCCNTENA_Msk;
                    ^~~~~~~~~~~~~~~~~~~~~~
                    MPU_CTRL_HFNMIENA_Msk
Marlin\src\HAL\HAL_LPC1768\../shared/Delay.h: In function 'volatile uint32_t getCycleCount()':
Marlin\src\HAL\HAL_LPC1768\../shared/Delay.h:53:61: error: 'DWT' was not declared in this scope
     FORCE_INLINE volatile uint32_t getCycleCount() { return DWT->CYCCNT; }
                                                             ^~~
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1

Config Attached: 

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3800155/Configs.zip)

buddybu, 2019-11-02:
I am also seeing this issue.  I can provide my configs if necessary.

psavva, 2019-11-02:
Once I revert f0d952d33ac7b2fb491fc356ce1d399f4460d5e4  then all is good.
I'm looking at the updated implementation and logic looks good.
How does WITHIN work?  Maybe the root cause of the issue?

randellhodges, 2019-11-02:
There is debate on the interwebs if BETWEEN is inclusive. It boils down to general usage is inclusive, aka "number between 1 and 10" but usually with physical objects, it isn't inclusive.

The choice to call it WITHIN was probably a choice to side step the ambiguity of inclusive since it implies inclusive. At least, that is how I read it. I could also be 100% wrong :)

closed 2019-11-03 by psavva