MarlinFirmware/Marlin #15764: Can'Compile Marlin 2.0 bugfix for Rumba32 Board

RipVW70, 2019-11-02:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

RipVW70, 2019-11-02:
Compiling in Arduino 1.8.9 and can't get past the following:

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "3D printer boards, RUMBA32, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), CDC (no generic 'Serial'), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

Build options changed, rebuilding all
WARNING: library U8glib claims to run on (avr, sam) architecture(s) and may be incompatible with your current board which runs on (stm32) architecture(s).
In file included from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\hal.h:99,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal.h:26,
                 from sketch\src\inc/MarlinConfig.h:30,
                 from sketch\src\Marlin.h:24,
                 from sketch\src\Marlin.cpp:31:
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:26: error: variable or field 'Step_Handler' declared void
 extern void Step_Handler(stimer_t *htim);
                          ^~~~~~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:26: error: 'stimer_t' was not declared in this scope
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:26: note: suggested alternative: 'timer_t'
 extern void Step_Handler(stimer_t *htim);
                          ^~~~~~~~
                          timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:36: error: 'htim' was not declared in this scope
 extern void Step_Handler(stimer_t *htim);
                                    ^~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:36: note: suggested alternative: 'tm'
 extern void Step_Handler(stimer_t *htim);
                                    ^~~~
                                    tm
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:26: error: variable or field 'Temp_Handler' declared void
 extern void Temp_Handler(stimer_t *htim);
                          ^~~~~~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:26: error: 'stimer_t' was not declared in this scope
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:26: note: suggested alternative: 'timer_t'
 extern void Temp_Handler(stimer_t *htim);
                          ^~~~~~~~
                          timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:36: error: 'htim' was not declared in this scope
 extern void Temp_Handler(stimer_t *htim);
                                    ^~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:36: note: suggested alternative: 'tm'
 extern void Temp_Handler(stimer_t *htim);
                                    ^~~~
                                    tm
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:131:9: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
 typedef stimer_t stm32_timer_t;
         ^~~~~~~~
         timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:137:8: error: 'stm32_timer_t' does not name a type; did you mean 'hal_timer_t'?
 extern stm32_timer_t TimerHandle[];
        ^~~~~~~~~~~~~
        hal_timer_t
In file included from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/STM32F4xx/stm32f4xx_hal_conf_default.h:419,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/STM32F4xx/stm32f4xx_hal_conf.h:13,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:30,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/stm32_def.h:34,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/clock.h:43,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/wiring_time.h:23,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/wiring.h:38,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/Arduino.h:32,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\shared\marduino.h:36,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\hal.h:28,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal.h:26,
                 from sketch\src\inc/MarlinConfig.h:30,
                 from sketch\src\Marlin.h:24,
                 from sketch\src\Marlin.cpp:31:
[... cut ...]
D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_tim.h:1244:6: note: in definition of macro '__HAL_TIM_GET_AUTORELOAD'
    ((__HANDLE__)->Instance->ARR)
      ^~~~~~~~~~
Error compiling for board 3D printer boards.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I am new to Marlin/Arduino and don't know where to go from here. Can anyone help?????

RipVW70, 2019-11-02:
[MyConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/3800179/MyConfigFiles.zip)

I am new to marlin,Arduino so I get lost a little when I get errors in compiling and am looking for help with the following errors.
The Marlin version is 2.0 bugfix and the board I am trying to flash is Rumba 32.

Compiling the attached files above in Arduino 1.8.9 and get the following errors:

WARNING: library U8glib claims to run on (avr, sam) architecture(s) and may be incompatible with your current board which runs on (stm32) architecture(s).
In file included from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\hal.h:99,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal.h:26,
                 from sketch\src\inc/MarlinConfig.h:30,
                 from sketch\src\Marlin.h:24,
                 from sketch\src\Marlin.cpp:31:
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:26: error: variable or field 'Step_Handler' declared void
 extern void Step_Handler(stimer_t *htim);
                          ^~~~~~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:26: error: 'stimer_t' was not declared in this scope
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:26: note: suggested alternative: 'timer_t'
 extern void Step_Handler(stimer_t *htim);
                          ^~~~~~~~
                          timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:36: error: 'htim' was not declared in this scope
 extern void Step_Handler(stimer_t *htim);
                                    ^~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:122:36: note: suggested alternative: 'tm'
 extern void Step_Handler(stimer_t *htim);
                                    ^~~~
                                    tm
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:26: error: variable or field 'Temp_Handler' declared void
 extern void Temp_Handler(stimer_t *htim);
                          ^~~~~~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:26: error: 'stimer_t' was not declared in this scope
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:26: note: suggested alternative: 'timer_t'
 extern void Temp_Handler(stimer_t *htim);
                          ^~~~~~~~
                          timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:36: error: 'htim' was not declared in this scope
 extern void Temp_Handler(stimer_t *htim);
                                    ^~~~
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:123:36: note: suggested alternative: 'tm'
 extern void Temp_Handler(stimer_t *htim);
                                    ^~~~
                                    tm
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:131:9: error: 'stimer_t' does not name a type; did you mean 'timer_t'?
 typedef stimer_t stm32_timer_t;
         ^~~~~~~~
         timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:137:8: error: 'stm32_timer_t' does not name a type; did you mean 'hal_timer_t'?
 extern stm32_timer_t TimerHandle[];
        ^~~~~~~~~~~~~
        hal_timer_t
In file included from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/STM32F4xx/stm32f4xx_hal_conf_default.h:419,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/STM32F4xx/stm32f4xx_hal_conf.h:13,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:30,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:250,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/stm32_def.h:34,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/stm32/clock.h:43,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/wiring_time.h:23,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/wiring.h:38,
                 from D:\Users\ggibb.DESKTOP-2AML7RE\Documents\ArduinoData\packages\STM32\hardware\stm32\1.7.0\cores\arduino/Arduino.h:32,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\shared\marduino.h:36,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\hal.h:28,
                 from d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal.h:26,
                 from sketch\src\inc/MarlinConfig.h:30,
                 from sketch\src\Marlin.h:24,
[... cut ...]
 typedef stimer_t stm32_timer_t;
         ^~~~~~~~
         timer_t
d:\users\ggibb~1.des\appdata\local\temp\arduino_build_850226\sketch\src\hal\hal_stm32\timers.h:155:5: note: suggested alternative: 'timeradd'
     TimerHandle[timer_num].handle.Instance->EGR |= TIM_EGR_UG; // Generate an immediate update interrupt
     ^~~~~~~~~~~
     timeradd

Error compiling for board 3D printer boards.

This is repeated many times more over several pages. I have tried tracking the paths but don't know what I can do once I get there

FLYmaker, 2019-11-02:
USE stm32duino ide1.50

RipVW70, 2019-11-03:
Thanks FLYmaker
Tried 1.50. It got rid of the timer issues but still would not compile. Came up with other issues such as "too many files" or similar. Also the Rumba32 boars requires STM32cubeprogrammer in DFU mode to upload. This was not available with 1.50.

FLYmaker, 2019-11-03:
usb support   ->CDC(generic

boelle, 2019-11-03:
have you tried to use vscode+platform.io?

RipVW70, 2019-11-03:
Thanks FLYmaker - Tried CDC (generic & the recommended (no generic - makes no difference.

boelle - new to all this & have never tried your suggestion. Looking at it now &  will have to work through the tutorials so there will be a learning curve. Thanks for the suggestion.

RipVW70, 2019-11-08:
Thanks to those that helped. Solved the problem by using an earlier version of STM32duino core 1.6.1 and compiling after every change made. Found that the current libraries did not like the Reprap discount full graphic smart controller being defined. Changed that for Reprap smart controller and marlin 2.0 compiled.

closed 2019-11-08 by RipVW70