MarlinFirmware/Marlin #15765: [BUG] [bugfix-2.0] Ramps and Viki LCD not compiling

chepo92, 2019-11-02:
### Description

I'm trying to compile marlin 2 with RAMPS and viki 1 LCD (i2c), i'm on the way of resolving this but I need to move some include libraries, so need some advice

> In file included from C:\Users\Axel\AppData\Local\Temp\arduino_build_755273\sketch\src\module\configuration_store.cpp:53:0:
> 
> c:\users\axel\appdata\local\temp\arduino_build_755273\sketch\src\lcd\ultralcd.h: In static member function 'static bool MarlinUI::button_pressed()':
> 
> c:\users\axel\appdata\local\temp\arduino_build_755273\sketch\src\lcd\ultralcd.h:161:19: error: 'BUTTON_SELECT' was not declared in this scope
> 
>      #define B_MI (BUTTON_SELECT << B_I2C_BTN_OFFSET)

After moving the include libraries from ultralcd_HD44780.h line 67 to Conditionals_LCD.h line 273 and adding some missing conditionals for heat bed i get this:

> C:\Users\Axel\AppData\Local\Temp\arduino_build_345136\sketch\src\lcd\ultralcd.cpp:71:7: error: 'lcd' was not declared in this scope
> 
>        lcd.buzz(duration, freq);

### Steps to Reproduce

1. Uncomment line ~1835   #define LCD_I2C_VIKI
2. Compile
3. Get the first bug
4. Move include libs 
5. Get the second bug

**Expected behavior:** Compile successful

**Actual behavior:** get some bugs

#### Additional Information

LCD is working with 1.1 bugfix

bug catching in: https://github.com/chepo92/Marlin/tree/vikiLCD-bugfix-2.0

chepo92, 2019-11-02:
This is similar to #14812 , so Tried https://github.com/Ludy87/Marlin/tree/bf2_buzzer2 after abug and adding 
> 	#include <Wire.h>
> 	#include <LiquidTWI2.h>

to ultralcd.h , it throws: 

> C:\Users\Axel\AppData\Local\Temp\arduino_build_811531\sketch\src\lcd\HD44780\ultralcd_HD44780.cpp:390:62: error: 'next_button_update_ms' was not declared in this scope
> 
>          if ((slow_bits & (B_MI | B_RI)) && PENDING(millis(), next_button_update_ms)) // LCD clicked

                                                              ^

boelle, 2019-11-03:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

chepo92, 2019-11-03:
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/3801781/Configs.zip)

PR #15811 (2019-11-06): Viki 1.0 LCD 
### Description

Added conditionals led status when heat bed no enabled (throwed errors)
Changes pin definition for ramps when Viki I2C enabled (to match official wiring diagram)
Tested with ramps and Viki 1.0 hardware

### Benefits

solves compiling issues for viki 1 LCD I2C using Ramps board

### Related Issues
fixes:
#15765 (tested working)
closed 2019-11-06 by thinkyhead