MarlinFirmware/Marlin #15766: ADVANCED_PAUSE_FEATURE and NOZZLE_PARK_FEATURE no longer heating bed

Mesmer88, 2019-11-02:
### Description

Marlin 2.0 Bugfix: when activating ADVANCED_PAUSE_FEATURE and NOZZLE_PARK_FEATURE printer won't heat up heatbed when starting a gcode print. If i order heat bed from the LCD menu it works.
Also with the 2 options heated bed crashed into the nozzle.
When removing the two options all goes back to normal

Hardware: SKR mini E3 dip v1.0 , Ender 5, TMC2208

### Steps to Reproduce

1. Activate ADVANCED_PAUSE_FEATURE and NOZZLE_PARK_FEATURE
2. Start a print

**Expected behavior:** 
Bed starts to heat then hot end

**Actual behavior:** [What actually happens]
Bed is not heating. LCD indicator shows heater element is on.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

[park_fail.zip](https://github.com/MarlinFirmware/Marlin/files/3800278/park_fail.zip)

boelle, 2019-11-08:
@Mesmer88 i assume this is still an issue ?

Mesmer88, 2019-11-08:
Yes, also tried updating to latest Marlin when i first reported it.
Wondering it anyone else can replicate the problem.

boelle, 2019-11-11:
no worked fine for me, no problem with bed heating

boelle, 2019-11-21:
@thinkyhead @shitcreek time to have a look here before the issue gets lost and not fixed

Mesmer88, 2019-11-30:
Today i tried to retest the bug:
I upgraded to latest Marlin 2.0 bug-fix, used new config files.
Did two tests, one firmware without ADVANCED_PAUSE and one with it.

Again ADVANCED PAUSE was causing erratic behavior on the printer. This time when hitting "Print from media" the printer would crash and buzzer sounds like siren. After power cycling the printer it would not boot up.

I removed ADVANCED_PAUSE and NOZZLE_PARK, all goes back to normal.

boelle, 2019-12-07:
@Mesmer88 since 2.0 was just released a few days ago has this changed this issue at all?

Mesmer88, 2019-12-08:
Just tried it now, good news: seems to be working fine with 2.0 release.

closed 2019-12-08 by Mesmer88