MarlinFirmware/Marlin #15767: [BUG] SRK mini E3 (again) wont boot up

reloxx13, 2019-11-02:
### Description
SKR mini E3 wont boot up if selfcompiled.

The default manufacturer bin works.
If i download the sourcecode from BTT´s Marlin and selfcompile, it wont work.

Latest Marlin compiled fine and wont boot up, too.

So i guess its a bugged library which brokes the boot up in the last 20 days.

I tried to go back to stm lib 5.5.0 and it wont boot up, too.

**Actual behavior:** [What actually happens]
Firmware flashes (firmware.cur gets created/renamed)
but screen stays blank.

#### Additional Information
My config can be found here:
https://github.com/reloxx13/Marlin/tree/my-skr-mini-e3/Marlin

BTT Version i tried:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.0

Lord-Quake, 2019-11-02:
I've been having similar problems.
In addition sometimes the renaming of firmware.bin to firmware.cur would take place but firmware.bin would not get deleted.
My issues have been randomly.
I'm using SKR mini E3 V1.2.

bojanpotocnik, 2019-11-02:
I had the same problem and after two days of hair-pulling I solved it with rather strange conclusion, as described in [Can't flash firmware - board won't boot](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/49#issuecomment-547310299). I have no idea what is the sorcery here, however I can confirm that if the firmware is compiled (fresh checkout of my fork, Python 3.7, venv) on my Windows 10 (1809 build 17763.805) the firmware is flashed but the screen will stay blue, however if compiled in the fresh Debian 10 virtual machine it works (.bin file also in my fork).

As for actually flashing the board, there is the same behaviour as described by @Lord-Quake . Sometimes the `firmware.bin` is left as is, only the `FIRMWARE.CUR` is created (being the exact copy of the `firmware.bin`). Sometimes the `firmware.bin` is deleted, but usually not. It seems to me that if I name the file `FIRMARE.BIN` then it is *moved*/*renamed* to `FIRMWARE.CUR`, but if it's named `firmware.bin` it is *copied* to the `FIRMWARE.CUR`. Despite that, the firmware is successfully flashed in all cases. Maybe bootloader has case-insensitive string comparison of the `FIRMWARE.BIN` when flashing and copying, and case-sensitive when deleting.

Lord-Quake, 2019-11-03:
@timryder made an interesting comment in https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/49#issuecomment-547310299.
`firmware.bin` cannot be larger than 224kb. Otherwise it overwrites the flash memory used as simulated EEPROM space. This could very well be the cause of our problems.

reloxx13, 2019-11-03:
we had the problem with flash memory already sorted out in one of the other hundret skr issues.

i tried with and wihout flash memory enabled and i also disabled some features to exclude the firmware size.
i had sd memory enabled 2 weeks ago when it stopped working.

thisiskeithb, 2019-11-03:
@reloxx13: Does this firmware work for you? [reloxx13-firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/3801358/reloxx13-firmware.bin.zip)

It was compiled from your repo on macOS and is under the 224kb limit.

reloxx13, 2019-11-03:
> @reloxx13: Does this firmware work for you? [reloxx13-firmware.bin.zip](https://github.com/MarlinFirmware/Marlin/files/3801358/reloxx13-firmware.bin.zip)
> 
> It was compiled from your repo on macOS and is under the 224kb limit.

yes, this one worked.

thisiskeithb, 2019-11-03:
> yes, this one worked.

Interesting. But if compiled on Windows it doesn't?

reloxx13, 2019-11-03:
yep, i noted in another issue already:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/49#issuecomment-549112276

> we know this issue with failing firmware from windows pc´s from Tasmota firmware.
> Its hapenning again in marlin since 20-30 days.
> 
> ps: my tasmota devices start to be instable and restating too since i updated around 7 days back with windows + vsc + pio

boelle, 2019-11-03:
> Otherwise it overwrites the flash memory used as simulated EEPROM space. This could very well be the cause of our problems.

who else can confirm this?

ManiRX100, 2019-11-03:
> 
> 
> > Otherwise it overwrites the flash memory used as simulated EEPROM space. This could very well be the cause of our problems.
> 
> who else can confirm this?

I faced this issue when I flashed firmware larger than 224kb. I went back and ran the old BTT Marlin 2.0 I had and used a 215kb firmware. It is working fine now.

reloxx13, 2019-11-03:
i feel like talking to trees.

its not the fw size, its windows.

i builded a fw on a debian buster now which works fine, same sourcecode like on windows. 223kb with sd flash.

stop going offtopic.

boelle, 2019-11-03:
sorry if i did go offtopic

opiswahn, 2019-11-03:
So, how to work around this? Compiling on linux is not a good solution :)

boelle, 2019-11-03:
or use the correct compile tool on windows :-D

Lord-Quake, 2019-11-03:
OK, I will confirm that the file size has nothing to do with our problem.
E.g. loaded a fresh FW version to the printer.
New FW was installed and FIRMWARE.CUR was created. However firmware.bin was not deleted.
I will no longer mention firmware size issue ;-)

opiswahn, 2019-11-03:
> or use the correct compile tool on windows :-D

And that would be? I use VSC +PIO, also tried Atim+PIO, both doesn't work

boelle, 2019-11-03:
i use that too and no issue here

opiswahn, 2019-11-03:
huh... is there any special setup within VSC and PIO that needs to be set? I only have that problem with the SKR E3 Mini v1.2. I have another SKR 1.3, that works without any problems. Both compiled on the same maschine and same VSC+PIO...

boelle, 2019-11-03:
platformio.ini needs to be correct, or you can use the correct autobuild

i use the first as that makes sure it builds correct if i just hit build

opiswahn, 2019-11-03:
Sure, i've edited the ini file for the right board type. As i said, for the SKR1.3 it works, but as soon as i change the settings to the E3 Mini it doesn't work.
It Compiles, then  i can sometimes upload to the sd (sometimes it won't flash correctly), but the device won't boot in einter case. If i use a precompiled firmware everything works like intended...

I don't have any clues what's going wrong here... The build environment seems to be ok as it works for the SKR1.3. Even if i try the exact repo that was used for a precompiled firmware it just won't boot up...

boelle, 2019-11-03:
bad hardware maybe? or the mini needs a different setting?

opiswahn, 2019-11-03:
HW can't be faulty, as i said, precompiled one works just fine... I used the settings provided by SKR and other people for the board...

boelle, 2019-11-03:
so it must be a setting then ?

opiswahn, 2019-11-03:
They seem to work for some people, not for others... idk...

reloxx13, 2019-11-03:
did u read that thisiskeithb confimred it? 

same source code, vsc+pio windows = not working
same source code, vsc+pio mac = working
same source code, vsc+pio linux = working

how can u say its settings?

post ur vsc version, installed pio core version, which OS and which board.

Win10x64 Pro 1903
VSC: 1.39.2
PIO Home: 3.0.0-beta.4
PIO Core: 4.1.0rc4
SKR mini E3 v1

opiswahn, 2019-11-03:
Same here, just different HW version: 

Win10Pro1903
VSC 1.39.2
PIO Core 4.1.0rc6
PIO Home 3.0.0-beta.4
SKR Mini E3 V1.2

--> Not working

boelle, 2019-11-03:
Win10x64 Home Build 18362
VSC: 1.39.2
PIO Home: 2.3.3
Re-ARM Ramps

where do i find the pio core ver?

reloxx13, 2019-11-03:
> where do i find the pio core ver?

on the pio home site

[image]

boelle, 2019-11-03:
post updated

reloxx13, 2019-11-03:
gotcha, after downgrading to pio core 4.0.3 its working.

pio-core 4.1.0rc4 > is making troubles.

Open an admin cmd

Be sure using the right python and pip.
You can see i have two python installations, i have to use the pio python
```
C:\WINDOWS\system32>where pip
C:\Users\Nils\.platformio\python27\Scripts\pip.exe

C:\WINDOWS\system32>where python
C:\Users\Nils\AppData\Local\Microsoft\WindowsApps\python.exe
C:\Users\Nils\.platformio\python27\python.exe
```

Upgrade pip (optional)
```
pip --version
C:\Users\Nils\.platformio\python27\python.exe  -m pip install --upgrade pip
pip --version
```

Upgrade all packages managed by pip (optional)
```
pip list --outdated --format=freeze | grep -v '^\-e' | cut -d = -f 1  | xargs -n1 pip install -U
```

open VSC go to files -> settings   
search for platformio   
untick "use dev"
close VSC

[image]

Force re-install of latest stable pio core
```
pip install "platformio>=4.0" --force-reinstall
```

open VSC and check the core version on PIO home site

[image]

Lord-Quake, 2019-11-03:
I never use RC versions and have been only using Home 2.3.3·Core 4.0.3.
Still have the problems.

reloxx13, 2019-11-03:
@Lord-Quake still try the force reinstall part, maybe platformio got damaged by an update or cache or whatever ^^
post feedback if it worked or not

thisiskeithb, 2019-11-03:
I updated my original post with system/version specs:

- macOS Mojave 10.14.6
- VSCode: 1.39.2
- PIO Home: 2.3.3
- PIO Core 4.0.3
- SKR 1.3

I don't plan to upgrade to test if it's the PIO Core RC 🙂

sl1pkn07, 2019-11-03:
works for me in linux with platformio from git (3de2d84e). tested with ReARM and SKR E3 DIP platforms

reloxx13, 2019-11-03:
i re-tried with Core 4.1.0rc6 and its working, too.

so im clueless, the simple reinstalling has fixed it for me.

i had no pc restart between it and used the same sourcecode as i did before.

opiswahn, 2019-11-03:
@reloxx13 f**k my, it's working, you are my hero :) the weekend is saved (well, what's left of it)

boelle, 2019-11-03:
so where do we stand?

bojanpotocnik, 2019-11-03:
> I only have that problem with the SKR E3 Mini v1.2. I have another SKR 1.3, that works without any problems. Both compiled on the same maschine and same VSC+PIO...

> As i said, for the SKR1.3 it works, but as soon as i change the settings to the E3 Mini it doesn't work.

> I don't have any clues what's going wrong here... The build environment seems to be ok as it works for the SKR1.3.

Note that SKR 1.3 uses LPC1768 but the SKR Mini E3 uses STM32F103. Maybe there is a problem in the chip support library.

Lord-Quake, 2019-11-03:
I'm using the exact same environment to program the FW for both cards: 
With the BIGTREETECH SKR MiNi V1.1 I have no issues.
With the BIGTREETECH SKR MiNi E3 V1.2 I have the issues e.g. firmware.bin not getting deleted.

boelle, 2019-11-03:
> Note that SKR 1.3 uses LPC1768 and the SKR Mini E3 uses STM32F103. Maybe there is a problem in the chip support library.

ivankravets, 2019-11-03:
Hi all,

I'm not sure that the issue is caused by PlatformIO Core. What is your `platformio.ini`? Which dev/platform did you use? Maybe, that update to it caused a problem.

reloxx13, 2019-11-03:
Hey,

this is my platformio.ini
https://github.com/reloxx13/Marlin/blob/my-skr-mini-e3/platformio.ini

bojanpotocnik, 2019-11-03:
Sorry for offtopic, just making things clear:
> With the BIGTREETECH SKR MiNi V1.1 I have no issues.
> With the BIGTREETECH SKR MiNi E3 V1.2 I have the issues e.g. firmware.bin not getting deleted.

@Lord-Quake firmware.bin not getting deleted is a different problem... In my [experience](https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-549087981) the `firmware.bin` is sometimes _moved_ (_renamed_) and sometimes _copied_ to the `FIRMWARE.CUR`. The firmware is, however, always successfully updated. The issue in this issue is if the board actually boots after such update or only the GLCD backlight is lit (and stuck there). 

Adding to my linked comment above, I discovered that the `FIRMWARE.BIN`/`firmware.bin` does not matter. I reformatted the factory 128 MB μSD card (FAT, 2k allocation size) and now the `firmware.bin` is always _moved_ to the `FIRMWARE.CUR` :man_shrugging:

In any case, this is related to the Maple Bootloader not the Marlin FW.

boelle, 2019-11-03:
@reloxx13 can you confirm what @bojanpotocnik just wrote?

Lord-Quake, 2019-11-03:
I wrote "e.g". I also had the printer not booting after FW install.

Here my [platformio.zip](https://github.com/MarlinFirmware/Marlin/files/3802010/platformio.zip)

reloxx13, 2019-11-03:
@boelle yep, he is right. the fw got flashed for my issue (.cur always got renamed) and not getting flashed is another issue, mostly related to a corrupted sd card or wrong file size.

ivankravets, 2019-11-03:
It's difficult for me to reproduce this issue. Please note that the most important role here plays a development platform and not PlatformIO Core. We always recommend using the latest PlatformIO Core.
The upcoming PlatformIO Core 4.1 will Project Memory Inspection tool. See https://community.platformio.org/t/platformio-home-3-0-beta-and-powerful-project-inspection/10187

So, you can inspect the same project on Windows and Unix and compare section/symbol tables.

If you use ST STM32 dev/platform in pair with Arduino framework, there is 5.7.0 version with a lot of bug fixes. You can try it. See https://github.com/platformio/platform-ststm32/releases

boelle, 2019-11-03:
@reloxx13 so we still have an issue or is this one solved? sorry i'm confused

opiswahn, 2019-11-04:
The issue with firmware not being deleted/renamed on SD still remains, but that is a separate topic i think.
The error with not beeing able to boot the firmware has been resolved by using the 'right' PIO version, though that is somewhat not really satisfying, as many people can potentionally run into that in the future...

reloxx13, 2019-11-04:
> @reloxx13 so we still have an issue or is this one solved? sorry i'm confused

heya @boelle ,

we found a "solution" by reinstalling PIO core, but not the cause of it.
The stable and the dev version does work after a reinstalling pio core with pip.

We can let this open, but im sure this will be an endless open issue cause its hard to debug and randomly occurs.

As long as we dont find the source of the issue, some ppl will run in it sooner or later.

boelle, 2019-11-04:
yeah :-/ will let it open

you think it would help if the first post is edited to give the "fix" ?

ivankravets, 2019-11-04:
> we found a "solution" by reinstalling PIO core, but not the cause of it

@reloxx13 why do you need that? I've not still received an answer about PIO Core. Do we have a problem with PIO Core 4.1RC6? If yes, please explain how to reproduce and we will fix.

You can try RC via `pio upgrade --dev`. Rollback to stable 4.0.3 version via `pip install "platformio==4.0.3"`

reloxx13, 2019-11-04:
> @reloxx13 why do you need that? I've not still received an answer about PIO Core. Do we have a problem with PIO Core 4.1RC6? If yes, please explain how to reproduce and we will fix.
>
> You can try RC [...]

The builded binary did not let the 3d printer boot, the screen stayed blank.   

We cannot reproduce it cause it hit in randomly and a reinstall of pio core fixes it while no sourcecode got changed. clearing cache, libs and platforms did not help. i also tried latest 5.7.0 stm lib and it did not work and also 5.5.0 which worked before did not work.   

rc and stable are working again now after reinstalling.   

I just hitted pull on marlin and merged it in my fork. and from one day to another the build is not booting.   

i mentioned the tasmota firmware before, the devices with the windows build firmware did restart every few hours, while now i use a linux build works fine. and before i updated the winows verison worked fine, too.   
we had this issue a few month (maybe a year now?) back already, than the tasmota devs decided to set up a build server to provide all diffrent fw env. cause it was not reproducable nor fixable.

if i got some time ill rebuild a tasmota version on windows and im pretty sure it will work again after the reinstalling of pio-core.   

my guess is sth is damaging the pio core installation over time on windows.    
it builds fine but the binary seems to be corrupt.

dinamitemic, 2019-11-08:
Same problem here. I'm on W10, board SKR_MINI_V1.1.
Downgraded to 1.9.2 version of platformio extension in VSC and everything restarted working as expected.

ivankravets, 2019-11-08:
VSCode extension does not have any influence on the final firmware. This is a wrapper around PlatformIO Core. Now, you can upgrade again to the latest version of the extension and everything should work.

What is your `pio --version`?

dinamitemic, 2019-11-08:
> VSCode extension does not have any influence on the final firmware. This is a wrapper around PlatformIO Core. Now, you can upgrade again to the latest version of the extension and everything should work.
> 
> What is your `pio --version`?

If you use the integrated piocore coming along with the extension it influence the version of it. Infact with the latest version of the extension I had PIO core v4.1.0. Downgrading I have v4.0.3 that is ok!

ivankravets, 2019-11-08:
PIO Core does not come with the extension. The extension installs the latest version. So, do you mean that firmware works with PIO Core 4.0.3 and does not work with PIO Core 4.1?

Could you try to upgrade again to 4.1 and re-test. Please open PlaformIO (Left Activity Bar) >  Misc> New terminal and type `pio upgrade`.

Try to Clean and build project again.

dinamitemic, 2019-11-08:
> PIO Core does not come with the extension. The extension installs the latest version. 
Are you shure about it?? I've installed a previous version of the extension from the extension menù>gear on the left of Platformio IDE>Install Another Version>Selected 1.9.2
>Could you try to upgrade again to 4.1 and re-test. Please open PlaformIO (Left Activity Bar) > Misc> New terminal and type pio upgrade. 
Late I'll try it, now just let me do what I wanted to do this morning before going in trouble with this.

ivankravets, 2019-11-08:
> Are you shure about it??

Yes, I'm. We (PlatformIO Team) created this extension.

> Late I'll try it, now just let me do what I wanted to do this morning before going in trouble with this.

We would be thankful for any tests. The PIO Core 4.1 was released yesterday.

thisiskeithb, 2019-11-08:
@ ivankravets: I’m seeing similar issues with PIO Core 4.1 on some BigTreeTech TFT35s (STM32-based): https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/126

Downgrading PIO core gives us firmware binaries that work.

[65 more replies]

closed 2019-12-03 by boelle