MarlinFirmware/Marlin #15768: [BUG] TURBO_BACK_MENU_ITEM breaks compilication again

NovaViper, 2019-11-02:
### Description
Everytime I try to compile, with the latest commit to Marlin 2.0 (15f94e5), the process fails

### Steps to Reproduce

1. Enable TURBO_BACK_MENU_ITEM
2. Run Build command in PlatformIO
3. Wait for the error to occur

**Expected behavior:** [What you expect to happen]
Not to crash and compile successfully

**Actual behavior:** [What actually happens]
Everytime I try to compile, I get this error: 

`
In file included from Marlin\src\lcd\menu\menu_advanced.cpp:31:
Marlin\src\lcd\menu\menu_advanced.cpp: In function 'void menu_advanced_settings()':
Marlin\src\lcd\menu\menu_advanced.cpp:713:10: error: invalid use of member function 'static void MarlinUI::goto_previous_screen()' (did you forget the 
'()' ?)
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_led.cpp.o
       ui.goto_previous_screen,
       ~~~^~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu.h:412:58: note: in definition of macro '_CONFIRM_ITEM_P'
       (encoderLine == _thisItemNr, _lcdLineNr, PLABEL, ##V);    \
                                                          ^
Marlin\src\lcd\menu\menu.h:418:42: note: in expansion of macro 'CONFIRM_ITEM_P'
 #define CONFIRM_ITEM(LABEL, V...)        CONFIRM_ITEM_P(GET_TEXT(LABEL), ##V)
                                          ^~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_advanced.cpp:704:5: note: in expansion of macro 'CONFIRM_ITEM'
     CONFIRM_ITEM(MSG_INIT_EEPROM,
     ^~~~~~~~~~~~
*** [.pio\build\LPC1768\src\src\lcd\menu\menu_advanced.cpp.o] Error 1
`

#### Additional Information
My configurations are all uploaded here: https://github.com/NovaViper/Marlin/tree/ender-3/Marlin
* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2019-11-03:
I can confirm that disabling `TURBO_BACK_MENU_ITEM` allows successful compilation on an SKR 1.3 build.

BastR, 2019-11-05:
Same problem with SKR PRO

sjasonsmith, 2019-11-09:
This came up in comments in #15655. I've identified the reason for failure and will post a PR soon.

sjasonsmith, 2019-11-11:
@NovaViper, this issue should be fixed now. Can you verify that it works properly for your printer, and close the issue if fixed?

NovaViper, 2019-11-12:
It no longer crashes! However I can't test to see if it works on my printer (I'm not currently at home for the week). I'll go ahead and close it since I can finally compile the firmware and will update once I get it flashed

closed 2019-11-12 by NovaViper