MarlinFirmware/Marlin #15770: [BUG] Cannot compile with ENABLE_INTERRUPTS_FEATURE: 'moreDigitalPinToPCMSK' was not declared in this scope

ansemjo, 2019-11-02:
## Description

The Marlin bugfix-2.0.x branch cannot be compiled for the Anet A8 (`sanguino_atmega1284p`) when the interrupts feature `ENABLE_INTERRUPTS_FEATURE` is enabled. Specifically the function used to check interrupt capability on pins does not seem to be defined:

```
In file included from Marlin/src/module/endstops.cpp:36:0:
Marlin/src/module/../HAL/HAL_AVR/endstop_interrupts.h: In function 'void pciSetup(int8_t)':
Marlin/src/module/../HAL/HAL_AVR/endstop_interrupts.h:64:32: error: 'moreDigitalPinToPCMSK' was not declared in this scope
   if (moreDigitalPinToPCMSK(pin) != nullptr) {
                                ^
In file included from Marlin/src/module/../inc/../HAL/./HAL_AVR/HAL.h:21:0,
                 from Marlin/src/module/../inc/../HAL/HAL.h:26,
                 from Marlin/src/module/../inc/MarlinConfig.h:30,
                 from Marlin/src/module/endstops.h:28,
                 from Marlin/src/module/endstops.cpp:27:
Marlin/src/module/../HAL/HAL_AVR/endstop_interrupts.h:65:66: error: 'moreDigitalPinToPCMSKbit' was not declared in this scope
     SBI(*moreDigitalPinToPCMSK(pin), moreDigitalPinToPCMSKbit(pin));  // enable pin
                                                                  ^
Marlin/src/module/../inc/../HAL/./HAL_AVR/../shared/Marduino.h:48:33: note: in definition of macro 'SBI'
   #define SBI(A,B) (A |= (1 << (B)))
                                 ^
Marlin/src/module/../HAL/HAL_AVR/endstop_interrupts.h:66:44: error: 'moreDigitalPinToPCICRbit' was not declared in this scope
     SBI(PCIFR, moreDigitalPinToPCICRbit(pin)); // clear any outstanding interrupt
                                            ^
Marlin/src/module/../inc/../HAL/./HAL_AVR/../shared/Marduino.h:48:33: note: in definition of macro 'SBI'
   #define SBI(A,B) (A |= (1 << (B)))
                                 ^
In file included from Marlin/src/module/endstops.cpp:36:0:
Marlin/src/module/../HAL/HAL_AVR/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
Marlin/src/module/../HAL/HAL_AVR/endstop_interrupts.h:102:49: error: 'digitalPinHasPCICR' was not declared in this scope
       static_assert(digitalPinHasPCICR(X_MIN_PIN), "X_MIN_PIN is not interrupt-capable");
                                                 ^
*** [.pio/build/sanguino_atmega1284p/src/src/module/endstops.cpp.o] Error 1
```

## Steps to Reproduce

I'm trying to build via the PlatformIO CLI Version 4.0.3 (most recent, installed via `pip`) but I've previously tried the Arduino IDE Version 1.8.9 and the Arduino CLI `arduino-cli` Version 0.3.7-alpha.preview as well.

Below are two scripts to reproduce with the default example configurations for the Anet A8 on the `bugfix-1.1.x` and `bugfix-2.0.x` branch. The `bugfix-1.1.x` branch works fine.

### `bugfix-2.0.x`

Shell script to reproduce:

```bash
#!/usr/bin/env bash

# enter temporary directory
cd "$(mktemp -d)" && pwd

# clone marlin source, checkout bugfix-1.1.x
git clone https://github.com/MarlinFirmware/Marlin.git ./
git checkout bugfix-2.0.x

# install and activate virtualenv
virtualenv venv && source venv/bin/activate

# install platformio
pip install platformio

# copy Anet A8 example config
cp config/examples/Anet/A8/Configuration* Marlin/

# compile with platformio
platformio run -e sanguino_atmega1284p
```

<details>
<summary>Full Build Log (click to expand)</summary>
[... 586 lines cut ...]

# install platformio
pip install platformio

# install liquidcrystal lib
pio lib install LiquidCrystal

# copy Anet A8 example config
cp Marlin/example_configurations/Anet/A8/Configuration* Marlin/

# compile with platformio
platformio run -e sanguino_atmega1284p
```

**Expected behavior:** The example configuration should compile without error with `ENDSTOP_INTERRUPTS_FEATURE` enabled.

**Actual behavior:** The compilation fails on the `bugfix-2.0.x` branch.

### Additional Information

The full build log can be seen above.

I have previously trief other build methods and used older commits of the `bugfix-2.0.x` branch and hit #15540 before. Issues #9632 and #10480 might be related too but I'm not sure.

Of course I can simply disable the interrupts feature and then the firmware compiles without error. But then why did it work before (and still does on the `bugfix-1.1.x` branch)?

GMagician, 2019-11-02:
@ansemjo please try my PR. I can compile using ANET default cnf

ansemjo, 2019-11-02:
Nice! Your pull request compiles correctly with `platformio`, yes.

Out of interest I retried compiling with `arduino-cli` again and I am once more hitting the errors from #15540:

```
$ arduino-cli compile --fqbn Sanguino\:avr\:sanguino\:cpu=atmega1284p Marlin/Marlin.ino 
In file included from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/endstops.cpp:36:0:
/tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
/tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h:119:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(X_MIN_PIN), "X_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /home/ansemjo/.arduino15/packages/arduino/hardware/avr/1.8.1/cores/arduino/Arduino.h:257:0,
                 from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../inc/../HAL/./HAL_AVR/../shared/Marduino.h:36,
                 from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../inc/../HAL/./HAL_AVR/HAL.h:21,
                 from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../inc/../HAL/HAL.h:26,
                 from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../inc/MarlinConfig.h:30,
                 from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/endstops.h:28,
                 from /tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/endstops.cpp:27:
/home/ansemjo/.arduino15/packages/Sanguino/hardware/avr/1.0.3/variants/sanguino/pins_arduino.h:112:70: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)        ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                     ~^~~~~~~
/tmp/arduino-sketch-CE86B61259E71E1DEABA5D41D890BEF6/sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h:119:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(X_MIN_PIN), "X_MIN_PIN is not interrupt-capable");

// repeated for Y_MIN_PIN and Z_MIN_PIN
```

GMagician, 2019-11-02:
I compiled ANET with arduino ide and compiles fine. What is arduino-cli?

ansemjo, 2019-11-02:
`arduino-cli` is from here: https://github.com/arduino/arduino-cli
I originally tried it because I wanted to automate builds with a Makefile without opening the IDE each time and before I knew about PlatformIO.

GMagician, 2019-11-02:
Can you try to compile with arduino ide? Just to know if it works and if issue is only on "-cli" version
I was able to compile!

ansemjo, 2019-11-02:
I'm not sure if I'm doing something wrong but the newest Arduino IDE doesn't work for me either -- it fails with exactly the same error message.

* Deleted `~/Arduino` and `~/.arduino15` to ensure clean environment.
* Downloaded and used Arduino 1.8.10 from [here](https://www.arduino.cc/download_handler.php?f=/arduino-1.8.10-linux64.tar.xz).
* Installed Sanguino board definition Version 1.0.3 via [Lauszus/Sanguino](https://github.com/Lauszus/Sanguino)
* Selected board `Sanguino` and processor `ATmega1284 or ATmega1284P (16 MHz)` from Tools menu.

```
In file included from sketch/src/module/endstops.cpp:36:0:
sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h: In function 'void setup_endstop_interrupts()':
sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h:119:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(X_MIN_PIN), "X_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
In file included from /tmp/arduino-1.8.10/hardware/arduino/avr/cores/arduino/Arduino.h:257:0,
                 from sketch/src/module/../inc/../HAL/./HAL_AVR/../shared/Marduino.h:36,
                 from sketch/src/module/../inc/../HAL/./HAL_AVR/HAL.h:21,
                 from sketch/src/module/../inc/../HAL/HAL.h:26,
                 from sketch/src/module/../inc/MarlinConfig.h:30,
                 from sketch/src/module/endstops.h:28,
                 from sketch/src/module/endstops.cpp:27:
/home/ansemjo/.arduino15/packages/Sanguino/hardware/avr/1.0.3/variants/sanguino/pins_arduino.h:112:70: error: reinterpret_cast from integer to pointer
 #define digitalPinToPCICR(p)        ( (((p) >= 0) && ((p) <= 31)) ? (&PCICR) : ((uint8_t *)0) )
                                                                     ~^~~~~~~
sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h:119:21: note: in expansion of macro 'digitalPinToPCICR'
       static_assert(digitalPinToPCICR(X_MIN_PIN), "X_MIN_PIN is not interrupt-capable");
                     ^~~~~~~~~~~~~~~~~
In file included from sketch/src/module/endstops.cpp:36:0:
sketch/src/module/../HAL/HAL_AVR/endstop_interrupts.h:135:7: error: non-constant condition for static assertion
       static_assert(digitalPinToPCICR(Y_MIN_PIN), "Y_MIN_PIN is not interrupt-capable");
       ^~~~~~~~~~~~~
```

ansemjo, 2019-11-02:
However, this is unrelated to the original issue that I opened here, I think. Discussion about the error around `error: reinterpret_cast from integer to pointer` should maybe continue in #15540?

boelle, 2019-11-03:
so since this compiles fine on platform.io would it be fair to say that this is not related to marlin?

GMagician, 2019-11-03:
This is related to arduino ide, @thinkyhead tryed to solve it but since new define are not present for other avr boards..he solved only 2560...
Since some define have to be added I think its better to leave old defines and add something only for "assertion" check, I have not figured out how to solve it. All supported avr boards have to be consideredto create new define and I have no idea how many boards needs to be added

boelle, 2019-11-03:
oki, will close as its related to compiler software and not marlin

we can reopen if the issue is truely within marlin

ansemjo, 2019-11-03:
I do not agree, @boelle. This issue still needs @GMagician's pull request #15771 merged to be fixed. As it stands, the current `bugfix-2.0.x` branch does **not** compile with the example Anet A8 configuration -- neither with the Arduino IDE *nor* with PlatformIO. See the scripts in my first comment to reproduce this.

I was simply saying that the next issue about `reinterpet_cast` around `&PCICR` is a different issue.

Please reopen this issue and / or merge #15771.

GMagician, 2019-11-03:
> I was simply saying that the next issue about reinterpet_cast around &PCICR is a different issue.

@ansemjo my PR now should also fix it

ansemjo, 2019-11-04:
@boelle, could you please re-open this issue? It *is* an issue within Marlin and @GMagician already has a fix for it. But that still needs to be merged.

boelle, 2019-11-04:
what pr # is it?

ansemjo, 2019-11-04:
It's PR #15771. That pull request also fixes issue #15540.

boelle, 2019-11-04:
https://github.com/MarlinFirmware/Marlin/pull/15771

ansemjo, 2019-11-05:
Weird, this wasn't closed when the PR was merged.

I believe this is fixed now, thanks.

PR #15771 (2019-11-04): Use prior endstop pin-to-interrupt macros
Revert recent AVR interrupt PRs and fix #15770 

Compiled ANET A8 and mega2560 with arduino 1.8.10 without any problem
closed 2019-11-05 by ansemjo