MarlinFirmware/Marlin #15772: [BUG] Driver Error: 8 Printer Halted TMC5160 SKR 1.3

thezak48, 2019-11-02:
### Description

Just installed an SKR 1.3 with TMC5160 on my Ender 3 and uploaded Marlin 2.0 and now I get this error everytime I try to turn on the printer "Driver Error: 8 Printer Halted" 

I can't seem to find out what is erroring out

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/3800782/Configuration.zip)

RudyBenner, 2019-11-02:
where is that 
Marlin/_Statusscreen.h

thezak48, 2019-11-02:
[Screen.zip](https://github.com/MarlinFirmware/Marlin/files/3800832/Screen.zip)
Here

RudyBenner, 2019-11-02:
C:\Users\RudolfBenner\Documents\GitHub\Marlin\Marlin\src\sd\usb_flashdrive\lib-uhs3\UHS_host\UHS_UsbCore.h

thezak48, 2019-11-02:
[UHS_UsbCore.zip](https://github.com/MarlinFirmware/Marlin/files/3800861/UHS_UsbCore.zip)
I assume you would like this

p3p, 2019-11-02:
There is a communication issue with a driver, you may find it easier to find the cause of the problem if you disable `MONITOR_DRIVER_STATUS` or its sub option `STOP_ON_ERROR`

This problem is usually down to wiring or pin config for the SPI or SERIAL connection to the drivers.

thisiskeithb, 2019-11-03:
No issues here with [my Ender-3 SKR 1.3/TMC5160 build](https://github.com/thisiskeithb/Marlin/tree/myskr) on the latest (and many before) commit reprinting the same gcode I've used in the past.

@p3p: So that means those LPC176* framework changes seem to be working fine as I haven't noticed anything different 🙂

boelle, 2019-11-03:
@thezak48 had a chance to check what @p3p said above?

mtowara, 2019-11-03:
@thezak48 I've had the same issue with this combination before. Where did you get your drivers? Directly from bigtree?

thezak48, 2019-11-03:
> @thezak48 had a chance to check what @p3p said above?

Yes the Driver Error has disappeared now however the drivers still seem to not communicate.

> @thezak48 I've had the same issue with this combination before. Where did you get your drivers? Directly from bigtree?

Yes I did

thezak48, 2019-11-03:
Thanks for everyone's help I seem to have resolved the issue, Only just found this https://github.com/MarlinFirmware/Marlin/issues/15296

closed 2019-11-03 by thezak48