MarlinFirmware/Marlin #15777: Marlin bugfix-2.0.x for SKR v1.1 PRO - Part cooling fan PWM scaling / Control limited to 71% ~ 76%

NickMilton, 2019-11-03:
#15515 # Description - Part cooling fan speed problem - severly restricted range of control

At indicated 70% fan starts turning
At indicated 71% fan spinning about 1/4 speed
At indicated 73% fan spinning about 1/2 speed
At 75% indicated and anything above fan runs at full speed

Below 70% fan does not spin. Above 75% there is no increase in speed

Obvious settings have been checked and tested with no reolution:
(1) FAN_MIN_PWM and FAN_MAX_PWM ranges played with & set to 0~255. No difference.
(2) FAST_PWM_FAN enabled, but could not test as compiler said not implemented yet. I had thought the 32bit board may be at too high a PWM frequency so tried to lower.
(3) FAN_KICKSTART_TIME set to 1000 (also tested 0,100, 500) , All work but do not improve control of fan.
(4) Have tested on 2 different BIQU SKR PRO V1.1 boards and with 4 different fans. All fans tested work perfectly well on an old AnetA8 printer board - so it is not a hardware issue. All fans 12V on 12V supply.

### Steps to Reproduce
Try to set cooling part fan speed in GCODE and manually on LCD
Below 70% fan does not spin. Above 75% there is no increase in speed (eg at 100% indicted the fan spins same speed as at 75% indicated)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3801298/Marlin.zip)

**Expected behavior:**
Fan should spin up over a much wider range (not just 70%~75%)

As above - control limited to 70% ~ 75% indicated

#### Additional Information - see config files attached

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3801299/Marlin.zip)

tpruvot, 2019-11-03:
i guess that depend on the fan model, boards are not shipped with fans...

NickMilton, 2019-11-03:
Appreciated

What surprises me is the old AnetA8 board runs the very same fans ober6a
wide range of rpms with no problem.

Cheers

thisiskeithb, 2019-11-03:
Have you tried enabling `FAN_SOFT_PWM`?

NickMilton, 2019-11-03:
Just tried on your advice and...

WooHoo - problem solved. Fan behaves very well - over a very wide range of indicated % settings. I had not tried before because I had not understood what if was for!!

GREATLY APPRECIATED. Hope I can return the favour

Nick

closed 2019-11-03 by NickMilton

thisiskeithb, 2019-11-03:
I have to enable that setting to get my Mechatronics 5015 blower to work on my SKR 1.3. It ticks a bit still, especially at lower RPM, but it's functional.

kongyuss, 2019-11-27:
> 只是尝试了您的建议，然后...
> 
> WooHoo-问题已解决。风扇的性能非常好-在很大范围的指示％设置中。我以前没有尝试过，因为我不知道该怎么做！
> 
> 不胜感激。希望我能回馈
> 
> 缺口

How to solve?

NickMilton, 2019-11-28:
> > 只是尝试了您的建议，然后...
> > WooHoo-问题已解决。风扇的性能非常好-在很大范围的指示％设置中。我以前没有尝试过，因为我不知道该怎么做！
> > 不胜感激。希望我能回馈
> > 缺口
> 
> How to solve?

Have you tried enabling FAN_SOFT_PWM?