MarlinFirmware/Marlin #15779: [BUG] (short description)

motion-au, 2019-11-03:
### Description

When I uncomment **#define Z_DUAL_STEPPER_DRIVERS** in configuration_adv.h to enable the additional Z stepper motor via the spare E1 output pins, I get smashed with compile errors.

Using latest marlin 2.0 bugfix, BIGTREE SKR 1.3 board. Also refer to pins.h file attached along with config files, I have previously needed to change E1 pinout to Z2 in that file after the Z2 stepper didnt respond because of a conflict with the E1 output not being at operating temp - but changing this does not appear to alter the compile errors I am experiencing now. That issue was fixed at the time but a number of changes may have been made in the firmware since and I am unsure if it remains relevant, but again I do not think it is affecting the compile errors as I toggle #define Z_DUAL_STEPPER_DRIVERS and it compiles fine regardless of what is in the pins.h file.

### Steps to Reproduce

Commenting #define Z_DUAL_STEPPER_DRIVERS will allow the firmware to compile without errors.

**Expected behavior:**
A successful compile and operation of the additional Z2 stepper via E1 output

**Actual behavior:**
**Compiling etc... then**
In file included from Marlin\src\gcode\config\M43.cpp:29:
Marlin\src\gcode\config\../../inc/../pins/pins.h:974:21: error: 'E1_MS1_PIN' was not declared in this scope
 #define __EPIN(p,q) E##p##_##q##_PIN
                     ^
Marlin\src\gcode\config\../../pins/pinsDebug.h:62:64: note: in definition of macro '_ADD_PIN_2'
 #define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
                                                                ^~~~
Marlin\src\gcode\config\../../pins/pinsDebug.h:64:44: note: in expansion of macro '_ADD_PIN'
 #define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
                                            ^~~~~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1270:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS1_PIN)
   ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:975:20: note: in expansion of macro '__EPIN'
 #define _EPIN(p,q) __EPIN(p,q)
                    ^~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:1087:27: note: in expansion of macro '_EPIN'
     #define Z2_MS1_PIN    _EPIN(Z2_E_INDEX, MS1)
                           ^~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1270:33: note: in expansion of macro 'Z2_MS1_PIN'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS1_PIN)
                                 ^~~~~~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:974:21: note: suggested alternative: 'Y2_MS1_PIN'
 #define __EPIN(p,q) E##p##_##q##_PIN
                     ^
Marlin\src\gcode\config\../../pins/pinsDebug.h:62:64: note: in definition of macro '_ADD_PIN_2'
 #define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
                                                                ^~~~
Marlin\src\gcode\config\../../pins/pinsDebug.h:64:44: note: in expansion of macro '_ADD_PIN'
 #define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
                                            ^~~~~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1270:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS1_PIN)
   ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:975:20: note: in expansion of macro '__EPIN'
 #define _EPIN(p,q) __EPIN(p,q)
                    ^~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:1087:27: note: in expansion of macro '_EPIN'
     #define Z2_MS1_PIN    _EPIN(Z2_E_INDEX, MS1)
                           ^~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1270:33: note: in expansion of macro 'Z2_MS1_PIN'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS1_PIN)
                                 ^~~~~~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:974:21: error: 'E1_MS2_PIN' was not declared in this scope
 #define __EPIN(p,q) E##p##_##q##_PIN
                     ^
Marlin\src\gcode\config\../../pins/pinsDebug.h:62:64: note: in definition of macro '_ADD_PIN_2'
 #define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
                                                                ^~~~
Marlin\src\gcode\config\../../pins/pinsDebug.h:64:44: note: in expansion of macro '_ADD_PIN'
 #define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
                                            ^~~~~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1273:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS2_PIN)
   ^~~~~~~~~~~~~~~~~~~
[... 51 lines cut ...]
Marlin\src\gcode\config\../../inc/../pins/pins.h:974:21: note: suggested alternative: 'X2_MS3_PIN'
 #define __EPIN(p,q) E##p##_##q##_PIN
                     ^
Marlin\src\gcode\config\../../pins/pinsDebug.h:62:64: note: in definition of macro '_ADD_PIN_2'
 #define _ADD_PIN_2(ENTRY_NAME, NAME, IS_DIGITAL) { ENTRY_NAME, NAME, IS_DIGITAL },
                                                                ^~~~
Marlin\src\gcode\config\../../pins/pinsDebug.h:64:44: note: in expansion of macro '_ADD_PIN'
 #define REPORT_NAME_DIGITAL(COUNTER, NAME) _ADD_PIN(NAME, COUNTER, true)
                                            ^~~~~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1276:3: note: in expansion of macro 'REPORT_NAME_DIGITAL'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS3_PIN)
   ^~~~~~~~~~~~~~~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:975:20: note: in expansion of macro '__EPIN'
 #define _EPIN(p,q) __EPIN(p,q)
                    ^~~~~~
Marlin\src\gcode\config\../../inc/../pins/pins.h:1093:27: note: in expansion of macro '_EPIN'
     #define Z2_MS3_PIN    _EPIN(Z2_E_INDEX, MS3)
                           ^~~~~
Marlin\src\gcode\config\../../pins/pinsDebug_list.h:1276:33: note: in expansion of macro 'Z2_MS3_PIN'
   REPORT_NAME_DIGITAL(__LINE__, Z2_MS3_PIN)
                                 ^~~~~~~~~~
*** [.pio\build\LPC1768\src\src\gcode\config\M43.cpp.o] Error 1

#### Additional Information
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/3801454/config.files.zip)

teemuatlut, 2019-11-03:
It's a known you issue. You can disable pins debugging for now or define MS1/2/3 pins as `-1` in your pins file.

motion-au, 2019-11-03:
Noted. Thank you

closed 2019-11-03 by motion-au