MarlinFirmware/Marlin #15785: Compile issue Bugfix2.0

sthgardner, 2019-11-03:
### Description

Compile errors

### Steps to Reproduce

1. Change Board to BOARD_BIGTREE_SKR_V1_3
2. default envs = LPC1768
3. 

**Expected behavior:** [What you expect to happen] Compiles

**Actual behavior:** [What actually happens] Errors

#### Additional Information> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\sthga\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.0.zip; framework: arduino; board: nxp_lpc1768)      
------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x 0.1.0 > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: toolchain-gccarmnoneeabi 1.80201.190214 (8.2.1), framework-arduino-lpc176x 0.2.0
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.0
|-- <U8glib-HAL> 0.4.1
|-- <TMCStepper> 0.5.2
|-- <Adafruit NeoPixel> 1.3.0
|-- <SailfishLCD>

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or 
copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\DebugMonitor.cpp: Permission denied
compilation terminated.
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\HAL.cpp: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\HAL_SPI.cpp: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\MarlinSerial.cpp: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\fast_pwm.cpp: Permission denied
compilation terminated.
cc1.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c: Permission denied
compilation terminated.
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\HAL_SPI.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\fast_pwm.cpp.o] Error 1
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o] Error 1
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\main.cpp: Permission denied
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o] Error 1
cc1plus.exe: fatal error: Marlin\src\HAL\HAL_LPC1768\persistent_store_flash.cpp: Permission denied
compilation terminated.
*** [.pio\build\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o] Error 1
========================================================== [FAILED] Took 6.35 seconds ==========================================================

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  FAILED    00:00:06.349
LPC1769                  IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      IGNORED
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
==================================================== 1 failed, 0 succeeded in 00:00:06.349 ==================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/3801761/configs.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

p3p, 2019-11-03:
This appears to a PlatformIO issue rather than Marlin, 

You may be able to fix it by removing the `.pio` folder that is created inside the Marlin folder to clear the cache, if that fail removing the `.platformio` folder from your user directory should fix it.

sthgardner, 2019-11-03:
Tried both. Still the same errors.

p3p, 2019-11-03:
I'm not sure then, its getting a `Permission denied` error trying to access files, you could try moving the Marlin folder somewhere else, or another drive and see what that does.. but you may want to report the problem to PlatformIO.

boelle, 2019-11-03:
also add your config files as the issue template asked you to

will temp close this one until they have been added

closed 2019-11-03 by boelle